keirf/flashfloppy #317: HFE with "unformatted" segment not working... v2.14

drdpj, 2020-02-15:
Hello - it's feasible that it's a badly encoded HFE, but it *does* work converted to SCP then written to a real disk. Track 39, last sector has a few unformatted 
[HFEs - v1 and 3.zip](https://github.com/keirf/FlashFloppy/files/4208577/HFEs.-.v1.and.3.zip)

The v1 is the original, the v3 is just saved out as a v3 (and the extra tracks removed)

keirf, 2020-02-15:
So do both of these work converted to SCP and then written to disk with Greaseweazle, but neither works with FlashFloppy?

keirf, 2020-02-15:
And the v3 is generated from the v1?

keirf, 2020-02-15:
There really doesn't seem to be anything that unusual here and FF should deal with HFEv1 just fine, it's a really simple format. I don't see what's unformatted. Do you mean extra tracks or something?

keirf, 2020-02-15:
And what machine is this for, a BBC? I wonder if it's a protection which is fussy about timings...

drdpj, 2020-02-15:
It's BBC Micro, yes. If you look at sector 9 on track 39 you'll see there's a gap with no fluxes between approx 3.9 and 4.5ms?

drdpj, 2020-02-15:
Something very odd - HxC is doing an appalling job of converting that HFE to SCP - the HFE loads fine, but this is what drops out...
[HFEv1_hfe.zip](https://github.com/keirf/FlashFloppy/files/4208653/HFEv1_hfe.zip)

keirf, 2020-02-15:
Ah I see, a No Flux Area. Flashfloppy will definitely get confused if it's longer than max timer setting (~900us). Maybe it can in other cases too, not sure it's been tested.

So even hxc's mangled scp works when written to disk?

keirf, 2020-02-15:
Or do you think its supposed to be a fuzzy bits area? Since you say the protection expects randomness rather than pure zeroes.

drdpj, 2020-02-15:
Seems to - I think all the data is on the earlier tracks!

drdpj, 2020-02-15:
Possibly fuzzy bits, but we have no way of knowing as the original is lost
to the mists of time... But NFA should yield random too? On a real disk it
just cranks up the gain...

keirf, 2020-02-15:
If you write no flux, you get randomness on read. If you write very rapid fluxes, you get no flux on read. There are protections out there which depend on either one of these. So: Which does No Flux in an HFE mean? It's not defined, right.

keirf, 2020-02-15:
Okay, had a closer look: Looks like a ~500us NFA on the HFEv1, HFEv3, and the generated SCP too. 

By the way the SCP is I think fine, as if you look in track-view mode all sectors are present and green, it's just HxC has decided to spray red all over certain tracks. It has a habit of doing that to SCP images, it's likely a bug so just something to watch out for.

Anyway, that 516us NFA in the SCP will be detected and written as NFA by Greaseweazle. So I'm pretty sure it is *intended* to be No Flux. As you wrote that very SCP image to disk via GW, and it works, right?

That size of NFA should be handled correctly by FlashFloppy in the HFE images too. Curious.

keirf, 2020-02-15:
Could you try dumping the HFE from Flashfloppy via greaseweazle and supply that as scp? Use the HFEv1 if the scp generated from it worked on disk. Yes I could do it myself 😉

drdpj, 2020-02-15:
[tbp-all.zip](https://github.com/keirf/FlashFloppy/files/4208936/tbp-all.zip)
Here's the original HFE (truncated to 40 tracks), the original converted SCP that was written, and the SCP that was read back :)

keirf, 2020-02-15:
So that is the scp read from gotek with that hfe mounted?

drdpj, 2020-02-15:
Oooh, I see what you mean... Hang on a mo :D

drdpj, 2020-02-15:
[tbp-ff-read.zip](https://github.com/keirf/FlashFloppy/files/4209022/tbp-ff-read.zip)
Here you go - read off the gotek with gw :)

keirf, 2020-02-15:
The NFA is still there, and the right size: 516us.

keirf, 2020-02-15:
So it's all a bit weird really. Is the Gotek tested on the same BBC as the real disk written by GW?

keirf, 2020-02-15:
Another thought: is the scp redump from real disk from the BBC drive? If not, is that possible to set up? Wonder if the old BBC drive might respond differently to that no flux area. Bit of a long shot. That flux area is starting to feel like a red herring.

drdpj, 2020-02-15:
So, tbp-read.scp is a dump from the floppy that works on the beeb from a 5.25" 40 track floppy drive. The HFE in the floppy emulator's failing as it seeks to track 39 during the initial loading (which must be where it's doing the check on that sector)...  I wonder... Could it be a speed thing?

drdpj, 2020-02-15:
(everything has been tested on the same computer!)

keirf, 2020-02-16:
Are you running on an 8271 FDC and/or with `index-suppression = no`?

keirf, 2020-02-16:
Also, to verify the bits coming off the Gotek are basically fine: Take `tbp-ff-read.scp` (the dump from Gotek) and write it back to disk. And test that disk. It will probably work.

drdpj, 2020-02-16:
Running with a 1770 with index-suppression=no.  The final test (dump from gotek back to disk) works fine.

keirf, 2020-02-16:
Please try changing to `index-suppression=yes`. The 1770 generally works okay like that right?

drdpj, 2020-02-16:
Yes, 1770 is fine like that - but changing that to a yes, still no dice.

keirf, 2020-02-16:
Tough cookie. I think it still must be timing related if it works Gotek -> disk -> BBC but not Gotek -> BBC.

keirf, 2020-02-16:
No other non default drive settings in FF.CFG?

drdpj, 2020-02-16:
host = acorn
index-suppression = yes
track-change = realtime
display-type = oled-128x32
oled-font = 8x16

keirf, 2020-02-16:
Try `track-change = instant` as well

drdpj, 2020-02-16:
No dices :)

keirf, 2020-02-16:
Final conclusion after some external discussions: this should be a flaky region (expressable in HFEv3 but not HFEv1).

closed 2020-02-16 by keirf