keirf/flashfloppy #318: Can't read 180K disks on Amstrad PCW9512

thejpster, 2020-02-15:
I have an Amstrad PCW9512 with a single 3" drive. I have completed the motor sense mod and built a 26-pin to 34-pin adaptor. The Gotek appears as Drive B: and I can read/write from 720k DSK files without issue so I know the drive is working. I can't, however, read 180K disk images.

If I create some test images with the following commands...

```console
$ dskform -type dsk -format pcw180 /media/user/GOTEK/000.DSK
$ dskform -type dsk -format pcw200 /media/user/GOTEK/001.DSK
$ dskform -type dsk -format pcw720 /media/user/GOTEK/002.DSK
```

Disk 002 works fine, but 000 and 001 give the following error:

```
CP/M Error on B: Disk I/O
BDOS Function = 14
```
Here's a video of the LEDs while running - in this case it's a 180K image downloaded from the Internet (so it should be good...):

https://photos.app.goo.gl/KsjZLxr8AGQtMAAp8

The Gotek is running FlashFloppy 2.14. Is this a DSK format parsing issue? Do I need to tune some configuration? I'm running without a config file, so it's all default.

Any suggestions gratefully received.

keirf, 2020-02-16:
Are these pure sector images (exactly 180k, 720k etc) or CPC DSK files (these are different sizes and have a text signature at their very start)?

thejpster, 2020-02-16:
They are DSK files, and start with the signature expected in your DSK driver file.

thejpster, 2020-02-16:
[disks.zip](https://github.com/keirf/FlashFloppy/files/4209662/disks.zip)

keirf, 2020-02-16:
Those DSKs should work as is with no configuration. Just to see if it is a DSK parsing issue, here are the two images converted using HxC software into HFE format. Please see if these work any differently:

[hfe_conv.zip](https://github.com/keirf/FlashFloppy/files/4209683/hfe_conv.zip)

thejpster, 2020-02-16:
pcw180_dsk.hfe gave a BDOS error. pcw720_dsk.hfe worked OK (doing a DIR reports 'No File').

keirf, 2020-02-16:
I believe then that there is some other reason for your machine giving the error.

A quick Google brings this up, worth a look, it could be baked into CP/M that Drive B is 720k. There may be workarounds (I didn't read too closely): http://www.cpcwiki.eu/forum/nc100-nc200-pcw-pda600/pcw8256-with-hxc-floppy-emulator/

keirf, 2020-02-16:
180k disks are indeed 40-track single-sided. The drive does not report the size of disk that is inserted. The filesystem, boot record, or some other on disk structure (probably in sector 0) must be used to detect that. It must be that which is going wrong. Either that section of the disk image data is not present or wrong, or your CP/M version is ignoring/overriding it at least for Drive B.

keirf, 2020-02-16:
Or.... https://www.cpcwiki.eu/forum/nc100-nc200-pcw-pda600/pcw8256-with-hxc-floppy-emulator/msg151389/#msg151389

It may be that CP/M has decided it needs to double-step a 180k disk. That wouldn't work unless tracks are doubled up in the disk image....

keirf, 2020-02-16:
I think either your 180k DSK needs to be doubled up on tracks (so that double-step commands go to the right track) or we need to convert to HFE with the double-step flag **and** then FlashFloppy needs fixing to obey the double-step flag. Sounds sensible? Are you wedded to DSK or happy to convert to HFE?

thejpster, 2020-02-16:
Double-stepping makes sense. I'm happy to convert to HFE, if there's a tool that runs on Linux.

It's not a drive B: issue as I've pulled drive A: and set the jumpers to S0, and I get the same effect.

keirf, 2020-02-16:
The HxC tools have to be built yourself on Linux. For starters, here's your PCW180 DSK image with doubled tracks. See how that goes:

[pcw180_double.zip](https://github.com/keirf/FlashFloppy/files/4209829/pcw180_double.zip)

thejpster, 2020-02-16:
`DISCKIT.COM` identifies that disk as 180K and if you run verify, you can see the Gotek display going up in 2s (0, 2, 4, 6, etc), while the on-screen track count goes up in ones. You can also do a `DIR` (and it reports `No Files`). Excellent!

So the only question remaining is, I guess - should we fix the flashfloppy firmware to divide all track counts by two if a certain option if a 40 track PCW image is detected, and/or is there a tool I can use to up-convert my 40 track DSK images to 80 tracks (where half the tracks are padding)?

keirf, 2020-02-16:
Right there indeed is the question. The only image format that allows to specify double step is HFE. Weird double-step-on-some-condition options to FF probably aren't tenable.

The morally best option here is probably to use HFE but uou may need to see if you can build hxc software for Linux. Or I can provide a script for doubling up your dsk images. It's a bit more gross as track data gets doubled up, but probably works OK. Do you need to he able to undouble (or convert back from HFE) to be able to use your DSK files elsewhere (other tools or emulators)?

thejpster, 2020-02-16:
I have no immediate need to transfer files to anyone else, and if I do, I'm not against knocking up a bit of Rust or Python to chop out every other track from a DSK file.

Given that anyone who uses FlashFloppy with an Amstrad PCW9512 (or an 8512 if you're replacing driving B:) is going to have this issue reading 180K disks, a `host=pcw` option which understands the difference between a 180K disk and a 720K disk (the only two disks that a PCW can read), doesn't seem too left field. Otherwise anyone who wants to download a copy of Batman, or Steve Davis Snooker, for the PCW will go through this same rigmarole.

keirf, 2020-02-16:
Maybe. For now then I will upload the track doubling code to the source repo. It's just a scrap of python.

keirf, 2020-02-16:
I don't know enough to be sure how PCW systems determine that double-step is required. It's a new area so I will tread lightly. I probably will implement the HFE double-step option however, as most people are happy to convert to HFE via the Windows/Mac GUI tool (also buildable on Linux) and I can then at least document that in the PCW section of the wiki.

I will leave this ticket open until I have done that (might be a week or two).

keirf, 2020-02-16:
Do try the script on downloaded DSK images. It may not work (there are two flavours of DSK). Attach one to this ticket if that happens and I'll fix the script.

thejpster, 2020-02-16:
It didn't work on these two: [disks.zip](https://github.com/keirf/FlashFloppy/files/4210896/disks.zip)

Linux `file` reports them as `Amstrad/Spectrum Extended .DSK data`. All the ones that worked were reported as `Amstrad/Spectrum .DSK data`.

keirf, 2020-02-16:
Okay try the script now.

thejpster, 2020-02-16:
Bingo - thank you very much.

keirf, 2020-03-13:
Both methods (convert to HFE, and double up tracks via Python script) are now documented in the wiki.

closed 2020-03-13 by keirf

P5ychoFox, 2020-12-06:
I've been reading the above with interest as I am currently trying (without success) to load 180k games on a 9512 using a Gotek. Has the double step issue been solved? My Flash Floppy firmware dates from 2018 so would using an up to date Flash Floppy file enable me to run these games or is there some code I'd have to input somehow.

tenuke, 2020-12-06:
Check this out
https://github.com/keirf/FlashFloppy/wiki/Host-Platforms#180k-disk-images

And I would also update the firmware, even though I dont know now if it is
a must

https://github.com/keirf/FlashFloppy/wiki/Firmware-Update

P5ychoFox, 2020-12-07:
Thanks. Here's how far I have got: I have copied the raw code from https://github.com/keirf/FlashFloppy/blame/master/scripts/edsk_double_step.py into a notepad++ document named: 'edsk_double_step') and it is in a folder named 'python scripts' on my desktop along with the 180k game snooker.dsk. In the command prompt after typing: **C:\Users\david>desktop>python scripts_edsk_double_step.py snooker.dsk fixed_snooker.dsk** I just get an error stating desktop is not a recognized command. Please could someone tell me what I'm doing wrong? I got the Hello World simple python script to run so I know I'm not totally useless. Thanks.

keirf, 2020-12-07:
You should type that line starting from **python**.

P5ychoFox, 2020-12-07:
Got it, it's running fine now. Many thanks! Apologies for all the noob questions.