keirf/flashfloppy #319: Akai S1100 and FlashFloppy E34 (error)

belzrebuth, 2020-02-26:
I can't get my gotek recognised. It's stuck on E34 no matter what I try on the USB stick side.
On my USB stick there's the FF.CFG file I got from here
https://drive.google.com/file/d/1JWmoVbSuA3u0KTOzmThWpH47l-FKbY6Y/edit
and a blank .img.
What am I doing wrong?!
Thanks!

keirf, 2020-02-26:
What is the img filename? What is the img file size?

belzrebuth, 2020-02-26:
DSKA0001.IMG and size = 800KB
I converter an .HFE image from HXC's QuickInstallImages AKAI S1000/S1100 to IMG and used that since I also got a e34 error using the .hfe file.

keirf, 2020-02-26:
So do you always get E34 regardless of IMG or HFE?

belzrebuth, 2020-02-26:
Yes, sadly..
My flash floppy firmware seems correctly installed.
I've reflashed it in order to be sure but I think there's something wrong with my USB stick files.

belzrebuth, 2020-02-26:
I held the two button with the USB drive disconnected and rST came up.
So after a reset I think it works!
I don't know if it's my bad or something else happened but it seems to be running now.
I really appreciate the quick suppport.
Thanks:)

closed 2020-02-26 by keirf