keirf/flashfloppy #321: OLED display on activity doesn't wake unless drive steps.

aerobaticant, 2020-03-02:
If the drive is accessed without stepping the heads the OLED doesn't wake up.

Keir:

> This is a tricky one, as some systems select a drive periodically without actually accessing it. Hence the choice to require actual head movement, or a write to disk, to trigger display-on.

Keir's suggestion (on FB) was:

> However I can extend display-on-activity (in FF.CFG) to allow other behaviour. For example "display-on-activity=yes,select"

keirf, 2020-03-03:
Here is a firmware to try. Put `display-on-activity=sel` in FF.CFG.

[ff_321_1.zip](https://github.com/keirf/FlashFloppy/files/4280755/ff_321_1.zip)

aerobaticant, 2020-03-03:
Hi Keir, that looks to be working just as expected.
Nice one!

keirf, 2020-03-03:
Okay this will be in the next 3.x release. For now you can use the interim firmware I posted here.

closed 2020-03-03 by keirf

aerobaticant, 2020-03-03:
Excellent, thank you!