keirf/flashfloppy #322: 128x32 Display Smokes

bensen-666, 2020-03-06:
Hello,

I flashed 2 Gotek SFR1M44-U100K with FlashFLoppy using USB cable method. This was no problem. The drives works well with original display. Then I bought 2 OLED 128x32 displays and changed wireing for the displays like on the hardware mod page. I connected it to power and the display smoked. I thought i had made a misstake. I checked wirering and tried the second gotek with the second display. Same problem it was smoking.
The i connected the displays to RP4 and they worked not. They seem broken. What happend there?

penfold42, 2020-03-06:
Either:
defective displays;
You swapped power and ground;
Faulty gotek that’s put 5v onto the 3v3 pin

bensen-666, 2020-03-06:
Holy Sh*t. I will test the pins next time manually. It'seems that the mainboards of the gotec devices are broken now. I ordered new displays from anothe manufacturer and put it in the gotek but it wasn smoking. After i tried it in the gotek the display works on Rpi but on the gotek no screen. 
I think the boards are broken. When i connect it to power no LED is blinking. But i can flash the FW again with USB cable

keirf, 2020-03-07:
Do the Gotek original 3-digit 7-segment LED displays still work?

bensen-666, 2020-03-07:
So i have figured the problem out. The first OLED Displays are incompatible, broken or from poor quality.
The first smoking displays were from GeeekPi from Amazon. Without any review, very strange... I tell it Amazon and they will analyze the problem and stop selling this displays.

The new display work out of the box. They are from AZDelivery. Very nice service they sent me an e-mail with datasheet.
The boards aren't broken. What a luck.

Thank you for your help. I never thought that the displays are sh*t.

closed 2020-03-07 by bensen-666