keirf/flashfloppy #324: FlashFloppy OSD

ramparts, 2020-03-28:
Hi there why use an external arduino for the OSD.
Could you implement like hxc just 2 lines for it ?
Most of the gotek users don't need to use the keyboard for image switching.
And you tell we can use an sd card on the gotek could you provide image with conections to the gotek drive ?

Kind regards 

JOSÉ GONÇALVES

keirf, 2020-03-29:
I may integrate OSD on a future design but it will not be a Gotek. I have no plan whatsoever to integrate OSD  into the gotek mcu.

Pin connections for SD card are in the Hardware Mods section of the wiki. It is not straightforward on a gotek pcb (you need reasonable soldering skills). Hence I don't hold anyone's hand by giving diagrams etc.

closed 2020-03-29 by keirf

ramparts, 2020-03-29:
Well it's a pitty since you have made a excellent work on the gotek till now

Best regards 

JOSÉ GONÇALVES