keirf/flashfloppy #325: Akai S1100 No samples on power up test program

belzrebuth, 2020-03-29:
Hi,
I have a weird issue with my S1100 and flash floppy.,.
My S1100 is running os 4.3.
It seems that when booting with an newly formatted emty disk loaded (001) the sampler starts with the default test program but contains no samples instead of the default sine, square, pulse.
If I power on the sampler without a USB drive it boots normally with the samples loaded.
The disk drive otherwise works okay.
I can load/format and save normally.
When testing I've only a blank 1600kB .img file and the FF.CFG file on my USB drive.
Any ideas what might be causing this?

keirf, 2020-03-29:
Are you sure that's not some normal behaviour?

belzrebuth, 2020-03-29:
Yes,
I should have noticed it before.

belzrebuth, 2020-03-29:
Ι'll try with a regular floppy and report back to ensure this is floppy related.
I can also place the original eproms back.
These are the 2 things I did recently on it.
I've upgraded the S1100's OS and replaced the floppy with a floppy emulator.

keirf, 2020-03-30:
I think that would be interesting. This would be a weird issue to be Gotek related.

closed 2020-05-29 by keirf