keirf/flashfloppy #326: Dsk ZX Spectrum Cpcdisk 2.5 782KB fail

retrocant, 2020-03-29:
Greetings, I have got 782KB dsk files for ZX Spectrum +3, in emulators it works perfectly, but I can't get it to work properly in flashfloppy. attached file.
[VOL0001.DSK.zip](https://github.com/keirf/FlashFloppy/files/4399182/VOL0001.DSK.zip)

keirf, 2020-03-30:
What doesn't work?

I see it is a double sided image. Is it intended for a double sided drive, on a real Spectrum, or should this image be split into A and B sides?

retrocant, 2020-03-30:
regards, it is meant to be able to work on a double sided unit

keirf, 2020-03-30:
Just tested, using FlashFloppy firmware version v3.13a. I loaded 1942 and Flying Shark successfully. Check your firmware version?

retrocant, 2020-03-30:
version 2.13

keirf, 2020-03-30:
Try v3.13a, as I have confirmed that works. If it's not working for you with 3.13a... something else is wrong ;)

retrocant, 2020-03-30:
I already have it functional, in the end what I have done is the following:
I have the flashfloppy connected to external drive B:
The A: internal unit had it connected, these files what it does is that when it sees that there is another floppy disk drive being the internal one although it disables by bridging the external unit 25 and 26, said unit A continues to see it.

solution, remove power from the internal disk drive and force A on the external one and it already works correctly.

thanks and greetings

closed 2020-03-30 by keirf