keirf/flashfloppy #327: Akai MPC 2000

GIMROD, 2020-04-07:
Hi is there some flash floppy software or process i can put the operating system of my mpc into a hfe file.....at the moment i have 30 image hfe files, 0000 having the operating system on, the FF UPD file and FFCFG file......at the moment the gotek is reading the file 0000 but the mpc states a disk error, i have the jumper in MO.....apologies i am new to this sort of thing....i am thinking the operating system may not of copied properly to the hfe file?  thanks

GIMROD, 2020-04-08:
My FF.CFG is host = akai  is this correct for the MPC2000 many thanks?

keirf, 2020-04-14:
How did you copy the operating system to the HFE file?

kempstonjoystick, 2020-05-15:
I'm hitting the same problem with the MPC 2000. The OS partially loads (the LCD displays the OS version), then at the same point every time, reports "Disk read error".

I'm using the HxCFloppyEmulator tool, v2.2.2.1 to produce the HFE image from the two OS files, which are named MPC2000.EXE and MPC2000.sys.

chrispoacher, 2020-05-15:
Hi, the files have to be written from the MPC I believe for the MPC2000, you can’t just copy the files on the PC. It works on the MPC2000xl this way, but sadly not the original MPC2000.

Chris

kempstonjoystick, 2020-05-15:
Hi Chris, that's not the case with conventional 3.5" floppy disks. With the MPC2000 you can simply copy the two OS files onto a formatted floppy and you're good to go. So I can't figure out why an HFE floppy image with the same content would behave any differently.

keirf, 2020-05-16:
Try creating a 1.44M IMG file and boot that rather than using an HFE file? In fact try using IMG files in general. Possibly the MPC uses PC formatted disks always but if it uses special Akai format for data disks then your data IMG files would need to be 1.6M size.

I suggest this because its a squeeze to get HD rate out of HFE on gotek sometimes. It's just worth sidestepping that possible issue with a more efficient image type.

Also are you sure the gotek is configured right? Is MPC using a Shugart interface? In which case the FF firmware should need no setup for that.

closed 2020-07-12 by keirf