keirf/flashfloppy #328: Atari 8-bit issues

Technoid6502, 2020-04-12:
I put a Gotek in an Atari XF551 drive with the HyperXF rom installed.

The drive will read a double-density image as such, but will not format one.  It will format in single and enhanced densities.

The XF551 is a double-sided drive with 360k capacity.  MFM 256bytes@sector, 40 tracks, 2 heads.  Flashfloppy will not format any double-sided 'media' at all.

The Atari is something of a babylon of formats:  Any format is possible in the 250kbit realm.  40 track, 77 track, 80 track, with all the permutations those drives produce with Single, enhanced, and double densities.

There is also the issue of interleaving.  The Ultraspeed interleave is 3:1.  Standard interleave is supported in single density, as is ultraspeed interleaving, but only partially.  A sector copy of an entire disk will produce an appropriately fast read, the interleave is functioning, but individual sector/file access does not, works at the standard interleave.  Here's a list of common interleaves by density.

									Reading order best for...
SD	1, 11, 4, 14, 5, 15, 8, 18, 9, 2, 12, 3, 13, 6, 16, 7, 17, 10	Disk formatted by Stock 810 or 1050
	1, 10, 17, 7, 16, 6, 13, 3, 12, 2, 9, 18, 8, 15, 5, 14, 4, 11	Disk formatted by USD, Speedy, Archiver (Chicago layout)
	1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18	Disk formatted in USD UltraSpeed Skew

DD	1, 12, 5, 16, 9, 2, 13, 6, 17, 10, 3, 14, 7, 18, 11, 4, 15, 8	DD Disk formatted by USD
	1, 11, 15, 6, 10, 14, 5, 9, 13, 4, 8, 18, 3, 7, 17, 2, 12, 16	DD Disk formatted by Happy, Speedy or Percom
	1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18	DD Disk formatted in UltraSpeed/Turbo 1050 Skew mode
The other utility that I know of that reads sectors 'out of order' for speed using a US Doubler is the SpartaDOS SCOPY command. (Specifically "SCOPY 9/16/86 ©1994 FTe") You manually tell it whether the source or destination disk is 'normal' by not specifying the /U (UltraSpeed) command line switch, which is supposed to read 'regular' disks in a fast way, or by specifying /U it will read sequentially... or so I thought:

SD	9, 2, 12, 3, 13, 6, 16, 7, 17, 10, 1, 11, 4, 14, 5, 15, 8, 18	Normal: ~1/2 rotation offset, slow reading Chicago skew disks (Those formatted by USD, Archiver, Speedy)
	5, 6, 7, 12, 13, 14, 15, 1, 2, 3, 8, 9, 10, 11, 16, 17, 18, 4	/U: ~1/2 rotation offset, [no SD turbo 1050 formatted disk handy to test]

DD	8, 15, 4, 11, 18, 7, 14, 3, 10, 17, 6, 13, 2, 9, 16, 5, 12, 1	Normal: Only fast with DD disks formatted by USD, slow reading Happy/Speedy/Percom
	2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 1	/U:  reads 1st sector last, slow reading Turbo 1050 std formatted disk

ED	15, 4, 18, 5, 19, 8, 22, 9, 23, 12, 26, 13, 2, 16, 3, 17, 6, 20, 7, 21, 10, 24, 11, 25, 14, 1	Normal: Stock 1050 or USD std ED disk fast, Turbo 1050 ED slow
	5, 6, 7, 12, 13, 14, 15, 20, 21, 22, 23, 1, 2, 3, 8, 9, 10, 11, 16, 17, 18, 19, 24, 25, 26, 4	/U: odd order maybe reads a USD skew disk faster?

I'm curious as to why the drive doesn't just accept and offer sectors in the order requested, as laid out.  The formatter sure doesn't take the controller literally as it should.

I love this product.  It works beautifully on my Amiga, older PeaSeas...  But the Atari 8-bit parts are partly broken.

I have an XF551 controller to test with, a Percom At88-SPD controller, and a Percom At88 with doubler.  All three of these controllers will control a 40 track double-sided drive in Single and DOuble densities.  The XF adds Enhanced, a middle density that is 26 sectors per track of 128byte sectors, an MFM mode.  The Percoms can control any 'low-density' drive on the market, 40 77 and 80 track, single or double-sided, MFM double density.  A 77 Track 8" double-sided drive will yeild 1mb, an 80 track ds/dd drive will yeild 720k on the Percoms.  I'd be happy to test any code you make.

Thanks!

Jeff

keirf, 2020-04-12:
There are other solutions on Atari 8-bit so basically the FlashFloppy support has hardly been user tested. Your best bet for arbitrary formatting/reading/writing is to use a DD HFE image such as this: https://github.com/keirf/FF_Images/raw/master/Amiga/Blank_AmigaDOS.hfe

Raw IMG and DSK files are essentially "hard formatted": the geometry and sector layout is either inferred from the file size, or via metadata in the image file, and cannot be changed by host formatting commands.

closed 2020-05-29 by keirf

manterola, 2025-10-28:
Regarding support for other other ATR images, what do you think about this modified code for flashfloppy/src/image/img.c ? (it adds support for 3.5" common ATR formats of 360KB (80 tracks, single side) and 720KB (80 tracks, double side).
`
/* To replace lines 642-660*/

/* 40-1-18, 256b/s, MFM, 180KB SS/DD image */

    nr_sectors = 18;
    im->nr_cyls = 40;
    im->nr_sides = 1;
    is_fm = FALSE;
    rate = ATR_RATE(250);
    if (no == 0) {
        is_fm = (sz < (130*1024));
        if (is_fm) {
            /* 40-1-18, 128b/s, FM, 90KB SS/SD image */
            rate = ATR_RATE(125);
        } else {
            /* 40-1-26, 128b/s, MFM, 130KB SS/ED image */
            nr_sectors = 26;
        }
	}
	else {
		if (sz == 256*18*40*2)-(3*128)) {
			/* 40-2-18 256b/s, MFM, 360KB DS/DD image */
			im->nr_sides = 2;
		}
		else if (sz == 256*18*80*1)-(3*128) {
			/* 80-1-18 256b/s, MFM 360KB SS/DD (typical 3.5 image single side) */
			im->nr_cyls = 80;
		}
		else if (sz == 256*18*80*2)-(3*128)	{
			/* 80-2-18 256b/s, MFM 720KB DS/DD (typical 3.5 image double side) */
			im->nr_sides = 2;
			im->nr_cyls = 80;
		}
		else {
			/* not supported, yet or invalid size*/
			return FALSE;
		}
    }`