keirf/flashfloppy #329: TRD Format issue

rexnz, 2020-04-19:
Hi all! I found that if I do a Format command of inserted image of 80 track and double side type (most common type of disks), the disk will be formatted as 80 tracks but only one side will be formatted! The size of the image would be the same as before format. After this operation I found some strange behavior of emulator, such as normal TRD images could not be read from TR-DOS, but can be inserted. If I try to open one side formatted image it in any emulator it will give me an error. Eventually I restarted emulator and eliminated reading issue, but format problem is persisting. Please keep in mind, that this issue might be related to my specific case, I can not connect read a disk on a physical real FDD, it always says no disk inserted, no lights on, but disk is spinning. Thanks.

keirf, 2020-04-19:
Geometries are detected based on format of trdos descriptor block at time the image is mounted. It can't then be changed by formatting. If you want that kind of fully general purpose image, format a blank HFE image file.

rexnz, 2020-04-19:
I mean the physical size will be the same 640k on the flash disk, but twice less free sectors will be after formating. Lets say I take a usual disk from any ftp resource, insert it, see all the 640 k and 2k+ sectors, format it and 320k only will be available (1k+ sectors). I did not put any commands saying it is one side disk, but it will format only one side always (other side even not appears on led not changing from 0 to 1 regarding which side is in process).

rexnz, 2020-04-19:
I can post an image (what was before and after format)

keirf, 2020-04-20:
Are you explicitly formatting as double sided ('$' character at start of disk title)?

rexnz, 2020-04-20:
Hi, I used both options, and gained the same result. After formating the image is working but with twice less amount sectors. I attached the image I used before format and what happened after and also a screenshot of actual steps. I also found that the magic-button stopped working properly. After loading a game and pressing the button I see recording is going into both sides, after finishing recording the computer is resetting and floppy emulator moves heads into 00 track. After this the image is unreadable. I also attached this file. ( I used a magic button on a fresh empty working disk same as before_format.trd). Is it possible that this linked to installed AY-8910 music coprocessor? At least the magic button worked properly before. I doubt this connected to this upgrade. The samples fo trd images and dialog can be found on this link https://drive.google.com/open?id=1OUYqDM-ypZKf9_5MvFkCTRzGzJHHd8ej   Thanks. Michael.

rexnz, 2020-04-20:
I found a few things. After any of format command with $ or without it the disk is readable even if I select another one and return back to new formated one. But once I reinsert this flash disk with these images into floppy emulator all these images became unreadable and shows an error *ERR*31* TR-DOS doesn't work and emulator keeps reading (green light is always on).

keirf, 2020-04-20:
I suggest to try formatting a blank DSDD HFE image such as this one which you can download: https://github.com/keirf/FF_Images/blob/master/Amiga/Blank_AmigaDOS.hfe

This will discount any possibility that this has to do with FlashFloppy's TRD handling, or any bad interaction of that with the host's FORMAT command.

closed 2020-05-29 by keirf