keirf/flashfloppy #330: Keyboard problem

Clifhanger3, 2020-04-21:
I am having trouble using the keyboard on Atari ST when i press F1 to search for files.

When I type:
Q it comes out as A
W comes out as Z
A comes out as Q
Z comes out as W
M key doesn't work at all

This is only an issue withing the flash floppy. The keyboard works fine in any text editing program I load.

Any ideas?

keirf, 2020-04-21:
This is with the AUTOBOOT program, right?

Clifhanger3, 2020-04-21:
> This is with the AUTOBOOT program, right?

yeah correct!

Squirrel61, 2020-04-21:
It looks like the QWERTY keyboard layout is hardcoded in the Autoboot
program, while you are using a keyboard with AZERTY layout.

Clifhanger3, 2020-04-21:
So i am using a QWERTY keyboard as that is the atari ST key layout.

Is there a way of changing this?

Squirrel61, 2020-04-21:
I don't know. I never had an ST but found on the internet that the AZERTY
keyboard seems to exist so it seemed the logical explanation for the
behaviour you found.

keirf, 2020-04-21:
Looks like @Squirrel61 is right but wrong way round: The AZERTY keyboard layout is hardcoded. Because the original author is French, of course :)

keirf, 2020-04-21:
It is hardcoded but could be fixed: The codes are all in atari_st/keymap.h. If someone wants to do a patch to QWERTY-fy them, I'll take it.

Clifhanger3, 2020-04-25:
> It is hardcoded but could be fixed: The codes are all in atari_st/keymap.h. If someone wants to do a patch to QWERTY-fy them, I'll take it.

This is where it's above my head. So i can't be of any assistance doing that haha. hopefully those smarter than me will be able to resolve :)

bfollowell, 2020-04-25:
OK, I have no idea how to compile or create a new AUTOBOOT file, but I was able to modify the keymap.h file to the correct values for a QWERTY style ST keyboard, at least the U.S & U.K. versions. Would someone be able to use this to create a QWERTY version of AUTOBOOT?

Thanks.
[keymap.zip](https://github.com/keirf/FlashFloppy/files/4533475/keymap.zip)

keirf, 2020-05-29:
Sorry for the delay. Here you go: Please test. If it works I will happily make this the keymap for next release.

[ST_AB_330_1.zip](https://github.com/keirf/FlashFloppy/files/4700299/ST_AB_330_1.zip)

bfollowell, 2020-05-29:
I will. Thank you!

Clifhanger3, 2020-06-06:
Yep, tested and it works as QWERTY.

Thank you very much!

keirf, 2020-07-12:
Pushed to the repository and will feature in the next release.

closed 2020-07-12 by keirf