keirf/flashfloppy #331: Support for the UFA1M44?

kkwannyc, 2020-04-22:
Any chance support can be added in the future?  I have the DU26 (26 pin) version which was successfully flashed and operates just great, but I would like to see a USB version of the Gotek perform similar to the DU26 version.  (no need for multiple 1.44MB partitions of ridiculousness, just blank or filled floppy disk images formatted to 1.44MB, OS/2 XDF or whatnot).

[image]

From looking at the logic board used (SFRD18), it looks like there are 2 STM32 chips - the Gotek portion looks like the normal STM32F105 (on top), while the USB controller looks to be an STM32F103 (below).  There looks to be 2 sets of programming/power headers (similar to the usual Gotek), and preliminary contact/pin traces look like they correspond to the usual conventions.

I am more than willing to donate hardware, take photos or do firmware dumps to make this happen.

keirf, 2020-05-29:
No immediate plan but I might pick one up, it could be interesting!

keirf, 2020-05-29:
By the way did you know that USB floppy drive protocol supports only 720kB and 1.44MB disks? It does make it less exciting :)

keirf, 2020-09-19:
Realistically never going to do this.

closed 2020-09-19 by keirf

nanosonde, 2020-10-04:
Can somebody tell me how the two STM32 are connected to each other?
One acts as the USB host and one as the USB device I assume.

keirf, 2020-10-04:
@nanosonde That has to be the case as each STM32 has only one USB port.

I haven't traced it out but they probably communicate by SPI.