keirf/flashfloppy #332: Support for Famicom Disk System?

philpem, 2020-04-29:
I notice that Quickdisk support has been added to FlashFloppy.
The Nintendo Famicom Disk System is based on QD and has a similar I/O interface.

What would be required to add .FDS image support to FlashFloppy?

keirf, 2020-04-29:
You will need to add an image format codecs to src/image folder and link it into the QD build.

philpem, 2020-05-22:
Adding some useful references for this...

  * FlashFloppy Quickdisk details -- https://github.com/keirf/FlashFloppy/wiki/Quick-Disk
  * #282 has some useful additional information
  * [NESDev FDS page](https://wiki.nesdev.com/w/index.php/Family_Computer_Disk_System)
    * [FDS Technical Reference](https://nesdev.com/FDS%20technical%20reference.txt) -- FDS ROM BIOS, RAM Adapter functionality, API and some info on the low-level disk format
    * [FDS Disk Format](https://wiki.nesdev.com/w/index.php/FDS_disk_format) -- high level
    * [.FDS image file format](https://wiki.nesdev.com/w/index.php/FDS_file_format)
  * Related OSS projects
    * [FDSEmu](https://github.com/holodnak/stm32-fdsemu)
    * [FDSPatch](https://github.com/vrai/fdspatch) -- fixes FDS format images to include the header
  * Hardware details
    * [Connector pinout](https://www.famicomdisksystem.com/disk-copy/) -- page 11
    * [Dubbing cable tutorial](https://www.famicomdisksystem.com/tutorials/cables/dubbing-cable-creation-tutorial/) -- how to make the connectors
    * [DIY FDSLoader dump cable](https://www.famicomdisksystem.com/tutorials/cables/how-to-build-a-fdsloader-dump-cable/)
    * [DIY FDSLoader load cable](https://famicomworld.com/workshop/tech/build-your-own-fdsloadr-cable/) -- using a Nintendo AV Multi Out port
    * [FDSLoader load cable](https://vrai.net/retro/?p=83) using a PCB
      * [FDSStick](http://www.tototek.com/store/index.php?main_page=product_info&products_id=196) and

keirf, 2020-05-29:
I can't see that this would be too super tricky. The fact that FDS contains multiple disks/sides which should presumably be user-selectable presents a possible need for UI modification. 

The actual framing up of blocks into the QD track looks pretty simple. None of the referenced OSS is very complex, in doing this job.

It needs someone motivated and with hardware to test against :)

chris-jh, 2020-12-06:
I have a spare Sharp Twin Famicom if you want to borrow it @keirf.

keirf, 2022-12-19:
Pretty sure this is fixed now by the longstanding QD support added to FlashFloppy.

philpem, 2023-01-02:
> Pretty sure this is fixed now by the longstanding QD support added to FlashFloppy.

Going by the documentation, only the QD image format is supported. So a tool would be needed to convert from FDS to QD -- and possibly multiple QD files, as a single FDS file may contain multiple disks.

There's an image converter for various FDS formats here: https://forums.nesdev.org/viewtopic.php?f=2&t=15895&start=15

I haven't been able to find a spec for the QD file format, but the loader code is here: https://github.com/keirf/flashfloppy/blob/master/src/image/qd.c

It looks fairly simple, but I'm not sure what the read/write window start and end are intended for, or what they'd need to be set to.

keirf, 2023-01-02:
QD is a raw container format like HFE. The whole raw bitstream gets played out, but READY is only asserted in the window between start and end. You would want to place the FDS file data within that window. I don't know whether FDS or the binaries produced by the conversion software you linked to also encode the bitstream to MFM (or FM, or whatever Famicom uses). If it does, you could basically plonk the binary into the middle of a QD container.

What do you think?

keirf, 2023-01-02:
By the way a blank QD is shipped with FlashFloppy and this is generated by a Python script in the FlashFloppy source tree. So you can see a typical construction with typical window bounds.