keirf/flashfloppy #335: Support for D40/D80 format (for ZX Spectrum)

ikonko, 2020-05-07:
Hello,

would it be possible to include support for Didaktik D40/D80 drive image formats? These drives used to be part of Didaktik eco-system and are ZX Spectrum/Timex etc. compatible.

It's not deviating from standard PC/MS-DOS floppy layout for 5.25" or 3.5" respectively, it's DD, 40 or 80 tracks, 9 sectors/track, 512 byte a sector.

For more details please see - https://translate.google.com/translate?sl=cs&tl=en&u=https%3A%2F%2Fcygnus.speccy.cz%2Fpopis_mdos-format.php

Thanks for excellent work, I enjoy your Gotek firmware the most!

keirf, 2020-05-08:
It will work fine already with double density HFE image files which you can format on the host. Also if the sector numbering is same as PC, you can use for example 720k IMG files (but these cannot be reformatted with different number of sectors per track).

omikron88, 2020-05-08:
The physical format is the same as PC DD exactly. But the earlier versions
of MDOS contain an error in FORMAT routine: It numbers both sides as 0, in
sector address field. It causes problems while reading such floppies at
i8272 based systems.

The very first sector contains a MDOS signature, disk name, random number
generated at format time and info about the geometry.

The file system itself is based on FAT12,
but it uses own different directory entries layout.

Sometimes uses custom format, up to 83 tracks and / or 10 sectors.

keirf, 2020-05-08:
If you want to use custom format, or want both sides numbered 0, then use HFE. Otherwise try using IMG files exactly 720k in size. Either way, format them via the Didaktik interface. See how you get on.

ikonko, 2020-05-11:
Thanks for your quick reply. I can confirm that recent firmware works well with following configuration in ff.cfg:
interface = ibmpc
host = pc-dos

And of course 720kB images must be renamed from .D80 extension to .IMG.

Tested on Didaktik Kompakt 1992 (MDOS1, WD2797).

All the best!

closed 2020-05-11 by ikonko

keirf, 2020-05-12:
What's the jumper configuration on the Gotek? With that info I will add Didaktik to the wiki.

ikonko, 2020-05-18:
Jumpers for Didaktik Kompakt internal cable as following: 

- S0 - closed (vertical)
- J5+JA - closed (horizontal upper position)
- all others - open

keirf, 2020-05-18:
J5 and JA are nonsense, you can physically remove those.

ikonko, 2020-05-18:
Thanks for the hint. I was suspicious it's just a spare for e.g. MO once needed.