keirf/flashfloppy #336: Unable to select image

Dan018, 2020-05-10:
OK, this may be a silly question, but how do I select an image to load from the USB stick?

I've flashed a Gotek to use in my A500 and have attached an OLED screen and rotary encoder. When I turn the Amiga on in native mode I can scroll through the images on the USB stick but when I click the encoder it takes me to a menu with a couple of options (clone image/eject and reload etc).

If I use the buttons I can again scroll through but pressing both buttons together brings up the same menu.

In Compatibility mode I can see Autoboot at position 000, but again can't select it and just get the menu options.

This could be a dodgy Gotek and I presume normally I would scroll to the image and press the encoder button to load it but I'd like to check before I write off the Gotek drive.

keirf, 2020-05-11:
By default an image is auto selected after two seconds of inactivity. Once an image is selected the select button becomes the eject button and takes you to the menu you've been seeing.

To disable auto select create an FF.CFG file containing `autoselect-file-secs = 0`.

Dan018, 2020-05-11:
That's great, thanks for your help.

I'm not sure what was wrong so I reflashed the drive and checked all the connections I'd made for the rotary and it now works.

closed 2020-05-11 by Dan018