keirf/flashfloppy #338: Amiga 2000, FF Selector won't load other images, just reboots into selector

kerobaros, 2020-05-19:
Sorry if this has been mentioned elsewhere, I tried searching and couldn't find it.

I have my v3.13a flashed Gotek in my newly-acquired Amiga 2000, hooked up as DF0. I don't currently have an OLED attached, just using the stock hardware, other than a piezo buzzer.

In native mode, with no FF.CFG file, the FlashFloppy works perfectly. I can switch between disk images with the front buttons, and the Amiga reads them fine.

When I put the Selector v8-FF AUTOBOOT.HFE and HXCSDFE.CFG on the flash drive I'm using, the Amiga boots into the Selector interface. I can assign disk images to slots. When I push F10 to save and reboot, the Amiga reboots right back into the Selector interface.

I'm not sure what I'm doing wrong here, but I'd be happy to help troubleshoot this however I can.

keirf, 2020-05-19:
Run Amiga Test Kit (https://github.com/keirf/Amiga-Stuff/releases/download/testkit-v1.7/AmigaTestKit-v1.7.zip) and do the floppy-drive write test on your Gotek.

kerobaros, 2020-05-19:
Tried to attach a screenshot of the test, but Github wasn't having it. 
Anyway, I'll type it in:
READY too fast (199us): Gotek or hacked PC drive?

Track 158 written:
 - Erase To Index Pulse: 201 ms (OK)
 - Write To Index Pulse: 227 ms (Late)
Track 159 written:
 - Erase To Index Pulse: 201 ms (OK)
 - Write To Index Pulse: 203 ms (OK)

Tracks 158 & 159 written okay
(Late Index Pulses may be due to drive emulation)

keirf, 2020-05-20:
Okay, writes work. Actually I suppose the selector wouldn't have got as far as it did if writes didn't work.

Do your slot assignments actually get saved? Ie. can you switch to them with buttons after reboot?

Have you tried F7 key which will take the currently selected image in the image-selection screen, save it to slot 1 (I think) and reboot you?

kerobaros, 2020-05-20:
Okay, here's what I just tried.
-added eight different images to the first eight slots
-pressed F10 to save and reboot
-Amiga reboots back into Selector interface
-go to slot list; all eight images still in previously assigned slots
-three-finger reset; Selector interface, images still in slots
-press F10 to save and reboot again
-insert the Zip cartridge I'm currently using as a system drive, so the Amiga boots into Workbench 1.3
-FFManager disk is shown in Workbench, with FFManager icon inside it
-pressing either button on Gotek causes the FFManager disk to be removed and immediately reinserted, front display only lists drive 000

¯\_(ツ)_/¯

kerobaros, 2020-05-20:
whoops, forgot about F7. tried that, wound up back in Selector.

keirf, 2020-05-20:
I wonder if HXCSDFE.CFG has ended up corrupt. I suggest back to basics, freshly format a stick, put fresh AUTOBOOT.HFE and HXCSDFE.CFG plus one other ADF, on it. Then see if you can assign that ADF.

keirf, 2020-05-20:
Also what is the stick (make, model)?  And could you try another?

kerobaros, 2020-05-22:
I have tried what you described with two separate sticks: an old PNY 512MB (the one I was already using) and a newer PNY USB3.0 64GB. Both work fine in native mode.

A thought: I have two floppy drives in the system, but the IDC connector on the ribbon cable to DF1: crumbled and broke. I need to order another ribbon cable, but for the meantime, I haven't changed the jumper on the board to tell the Amiga that there's no DF1: attached. Consequently, the drive shows as DF1: BAD on the Workbench. Can that cause any sort of issue with the Gotek attached to DF0:?

keirf, 2020-05-22:
Can you attach an hxcsdfe.cfg here and we can see if it's got modified at all by the file selector?

kerobaros, 2020-05-22:
Sure, here you go. Had to zip it for Github to accept the file.
[HXCSDFE.ZIP](https://github.com/keirf/FlashFloppy/files/4670783/HXCSDFE.ZIP)
-reformatted the aforementioned 64GB stick
-copied over config and autoboot from flashfloppy 3.14a zip file, and disk images
-booted 2000, selected disks for first three slots, saved and rebooted with F10
-rebooted to Selector, first three slots still filled in Selector interface

keirf, 2020-05-23:
The slots are saved correctly in the HXCSDFE.CFG file and I can select between slots 0-3 when I use that config file on two of my v3.13a Goteks (one with OLED display, the other with the original 3-digit display).

Are you definitely using v3.13a? You mention 3.14a in your previous message which doesn't exist.

Do you have any other config files at all (eg. FF.CFG)? Remove them if so. And you can clear any cached config in your Gotek by holding both buttons for a few seconds when no USB stick is inserted (you will see "Reset Flash Configuration" appear on the OLED display).

kerobaros, 2020-05-24:
I am definitely using 3.13a, yes. I had AmigaOS 3.1.4 on the brain, haha.

SO. I moved all my adfs to the root of the USB stick, and did the reset shortcut you mentioned. 'rst' showed up on my little three character display. Put the stick back in, rebooted back to the selector, set up my images, and /everything worked perfectly/.

I put the ADFs back in a folder, and tried again, and once again, everything worked! So I guess the glitch was resetting the config with that shortcut you mentioned. Did I miss that somewhere else in the documentation?

Regardless. Not a bug, just a PEBKAC. Thank you!

keirf, 2020-05-24:
It's in the docs somewhere. I should call it out in the Trouble shooting guide though, or perhaps as a step to try in the process of considering and filing a bug report. I'll keep this open while I consider a wiki update.

keirf, 2020-05-24:
More explicit now in the FF.CFG wiki page, and added a mention in the FAQ too.

closed 2020-05-24 by keirf