keirf/flashfloppy #339: Korg DSM-1 IMG vs. HFE-Format

Roddy61, 2020-05-20:
Hi,
I recently bought a Gotek FD-EMulator with Flash Floppy 2.14 installed.
After installing it into my Korg DSM-1 Sampler it works correctly (reading and writing) with IMG-Files (FF.CFG: Host Platform = akai)
but it refuses to read or write HFE-files in DSM-1 format (Host Platform akai or unspecified). it only reads Korg DSS-1 files in HFE. (host Platform: unspecified)

Can anyone help me out where the problem lies?
I thought HFE would be the native file type.

keirf, 2020-05-20:
What are you using for the DSM-1 HFE image file? Are they downloaded or created by you?

keirf, 2020-05-20:
Can you zip and attach one here?

It could also be a USB stick speed issue. Tried another?

Roddy61, 2020-05-21:
Thank you for helping! 
I've put 3 Files in:

1. a working DSS-1 HFE-File, 
2. a working IMG-File and 
3. the corresponding HFE-File (not working).

Strange, the DSM-HFE-File has only 5KB compressed size, while the IMG has 1.325 KB.

I've tried half a dozen USB-Sticks of different brands, but all showed the same problem.

Could the problem perhaps lie in the conversion tool copyQM? Is there an alternative?
[Files.zip](https://github.com/keirf/FlashFloppy/files/4661806/Files.zip)

keirf, 2020-05-24:
Yeah that DSM-1 HFE file is bogus: It's entirely empty (no sectors formatted on it at all).

However, did you know FlashFloppy should support the IMG files directly? That is actually the point of specifying `host=akai`! So just get rid of the bogus HFE and try the IMG direct instead. And let me know how you get on!

Roddy61, 2020-05-24:
Thanks for the info! 

I thought the HFE-Files were the native format for the Floppy emulator and as such would perhaps provide quicker data transfer, better handling or something like that.

But when doesn't make any difference, I'm fine with the IMG-Files.

Is there any real reason for preferring any format to the other?
And by the way ist there any parameter to speed up the data-transfer to the floppy controller a little bit still?

Strange, that my conversion tool does not work correctly. It's flopimage_convert.exe with hxcfe.exe and a Batch file to get from DSK to IMG and (not correctly) HFE.

keirf, 2020-05-24:
IMG is actually more efficient to access as it's way smaller. It should always be preferred unless there's some other compelling reason to use HFE.

closed 2020-05-24 by keirf