keirf/flashfloppy #340: Question:  Increase volume/include volume pot for speaker mod?

omega9380, 2020-05-24:
I'm running a Gotek device using FlashFloppy.  I've modded to use a 0.9" OLED, rotary encoder, and the speaker.

I'm using the circuit that was shown on the Hardware Mods wiki page so I can use a standard speaker (I have a PC speaker wired in).  My question is, can this circuit be modified to boost the volume?  Or even include a volume control pot?

Incidentally, I've designed a small (less than an inch square) board for that very circuit.  I've uploaded it to my Github if anyone wants to take a gander, and yes, I know I spelled circuit wrong:

Design with 4 pin header to use a standard PC speaker: https://github.com/omega9380/Adventures-in-EAGLE/tree/master/GOTEK%20Speaker%20Curcuit

Design with a thru-hole speaker on-board: https://github.com/omega9380/Adventures-in-EAGLE/tree/master/GOTEK%20Speaker%20Circuit%20-%20On%20Board

Please also let it be known that I am not an electrical engineer, just a hobbyist, and this is actually my very first board design using Eagle.  Criticism is welcome (and expected!).

Thanks!

keirf, 2020-05-24:
Is GND actually hooked up on the Board PDF? It's not obvious. All three pins of the transistor should definitely be hooked up.

You could try putting a control pot in series with the base of the transistor and see how that works. Of course FlashFloppy also has a volume control in FF.CFG.

For volume boost, perhaps hook up something like an LM386 (you can get ready-made boards with handy terminals) to a bigger speaker: You could do this instead of the simple BJT buffer circuit.

Or perhaps use the pulse from pin JB to trigger one of those tiny MP3 players to play a more realistic step sound? (Look up DFplayer on Ebay for example).

penfold42, 2020-05-24:
Once you’ve hooked up ground, you could also power the speaker vcc from 12v instead of 5v if your transistor is upgraded.

I’d also recommend a back emf clamping diode across the speaker

omega9380, 2020-05-24:
Thanks for the replies, now you are getting into electrical engineering :).  I'll dig around and play with it, I already have the volume at max in FF.CFG but the sound is still kinda quiet.

There's a ground flood on the 2nd layer, it's not obvious on the PDF but it's in the Eagle files.

Thanks!

closed 2020-05-29 by keirf