keirf/flashfloppy #341: Having problems with FlashFloppy i get the basics but i need help!

Bellstr, 2020-05-31:
Installed all correctly to the usb stick and the games etc have been working fine. Ive been manually selecting each game etc that i wanted with the gotek buttons. 

Well i found out today about FlashFloppy and thought this would be a miles easier and quicker way.

Ive installed it but the program hangs when its reading the HXCSDFE.CFG i wonder if someone else has come across this and wondered if they or you have a solution to my problem.

Thanks in advance.

keirf, 2020-06-01:
Clean formatted stick, put fresh HXCSDFE.CFG and AUTOBOOT.HFE on it from the FlashFloppy distribution zip. See if that works. I assume you are running on an Amiga?

DynaMic-Ensoniq, 2020-06-13:
Hi, Sorry for butting in, I'm new here and I am looking for some advice.
I recently bought a Gotec and on following your page and a few bits and pieces from elsewhere,   I now have it running great in my Ensoniq SD1 synth.  Thanks a million.

Can you advise on this?
I had previously changed my FDD for an older style which wrote to a 512mb Sandisk.  This unit stopped working after a little accident caused it to short against the metal chassis.  I t seems to be formatted in FAT 12 and I can't read it on Windows 10 and can't seem to find any info on this.  Is there any way I can load it into HxC via BIOS reads?

DynaMic-Ensoniq, 2020-06-13:
I should have mentioned, I downloaded a freeware "Read Write anything" and I can read the byte-swapped data so the media seems intact but I can't see a way of running a sequential copy.

DynaMic-Ensoniq, 2020-06-13:
Ah, I just saw the new issue button.  Shall I ask there or is this cool?  Sorry, new to all this.

keirf, 2020-06-13:
So what was the previous unit that shorted out? A gotek? If it was running factory firmware then I think that gets partitioned up with custom software. You might need to go find the factory gotek software to download in that case.

DynaMic-Ensoniq, 2020-06-13:
Hi Keir, thanks for your prompt response.  I've since tried various freeware packages but they just see an empty card.
The only information I can find on the device is off the circuit board.  HDMA22.PCB made by 2AVcom, www.2av.com.au<http://www.2av.com.au>.
After a second look at the byte swapped data, I found a string saying VFX-SD1.  It may be in the Ensoniq format.  I'll have a look at the company mentioned to see how it goes but I recall purchasing from a company in Russia who were far from helpful when the initial device they sent me didn't work.

DynaMic-Ensoniq, 2020-06-14:
Hi, I'm no further forward after checking the company out online.  What I have noticed though is a DOS 5 label in the byte swapped data.

________________________________
From: john boyle <johnboyle@hotmail.co.uk>
Sent: 13 June 2020 20:04
To: keirf/FlashFloppy <reply@reply.github.com>
Subject: Re: [keirf/FlashFloppy] Having problems with FlashFloppy i get the basics but i need help! (#341)

Hi Keir, thanks for your prompt response.  I've since tried various freeware packages but they just see an empty card.
The only information I can find on the device is off the circuit board.  HDMA22.PCB made by 2AVcom, www.2av.com.au<http://www.2av.com.au>.
After a second look at the byte swapped data, I found a string saying VFX-SD1.  It may be in the Ensoniq format.  I'll have a look at the company mentioned to see how it goes but I recall purchasing from a company in Russia who were far from helpful when the initial device they sent me didn't work.

DynaMic-Ensoniq, 2020-06-14:
After reading through the byte swap hex file, I found this string (swapped): VFX-SD Disk Formatted by EDM.EXE Verion 1.05.
I believe this is the Rubber Chicken Software but as far as I can find there is no way of reading this from a USB device.   This means it must be in the Ensoniq format.  Assuming we can write a disk using a disk image, it must be possible to reverse the process.  Do you know where I could start?
My thinking is if I could sequentially read the USB through BIOS and get the raw data onto my W10, it should just be reformatting for FAT 32.

________________________________
From: john boyle <johnboyle@hotmail.co.uk>
Sent: 14 June 2020 18:36
To: keirf/FlashFloppy <reply@reply.github.com>
Subject: Re: [keirf/FlashFloppy] Having problems with FlashFloppy i get the basics but i need help! (#341)

Hi, I'm no further forward after checking the company out online.  What I have noticed though is a DOS 5 label in the byte swapped data.

________________________________
From: john boyle <johnboyle@hotmail.co.uk>
Sent: 13 June 2020 20:04
To: keirf/FlashFloppy <reply@reply.github.com>
Subject: Re: [keirf/FlashFloppy] Having problems with FlashFloppy i get the basics but i need help! (#341)

Hi Keir, thanks for your prompt response.  I've since tried various freeware packages but they just see an empty card.
The only information I can find on the device is off the circuit board.  HDMA22.PCB made by 2AVcom, www.2av.com.au<http://www.2av.com.au>.
After a second look at the byte swapped data, I found a string saying VFX-SD1.  It may be in the Ensoniq format.  I'll have a look at the company mentioned to see how it goes but I recall purchasing from a company in Russia who were far from helpful when the initial device they sent me didn't work.

keirf, 2020-06-14:
You can sequentially read a USB stick on Mac or Linux using the command line 'dd' tool.

DynaMic-Ensoniq, 2020-06-14:
Hi, I finally made some headway but I am now stuck at the next hurdle.

I found a program able to read the drive - HxD.

I can read, select, export and save the data but do not know which format it needs to be saved in.  I tried just saving the defaults and tried loading as a raw file and saved it as a disk image without error, however it is unreadable in my synth.

DynaMic-Ensoniq, 2020-06-17:
Hi,

I finally got it sorted.

The trouble was my original FDD Emulator had packed in after shorting in my Synth (Ensoniq SD1).  All of my data was stored on a 512MB Sandisk Compact Flash card.  I had replaced the FDD Emulator with a Gotec drive, following your instructions to remap the pin 1 on the header to get the disk change from the logic.   Incidentally, what I done may be an easier way rather than trying to cut pins on the PCB.  I snipped pin 0 on the ribbon cable then wired via a wire add to PA13.

I used a program called HxD which helped me get the data from the non-Windows formatted card into my PC.  I then had to byte-swap the data and eventually found a trial utility online called 010 Editor.  Once the data was byte-swapped I managed to go through the Hex data and locate the disk images.  It was all stored as one single Hex file so it was a bit of messing around.  I used HxC to write the new FDD images to my new USB stick for use in the new Gotec emulator and all is well.

Thanks and regards,

John

________________________________
From: john boyle <johnboyle@hotmail.co.uk>
Sent: 14 June 2020 23:25
To: keirf/FlashFloppy <reply@reply.github.com>
Subject: Re: [keirf/FlashFloppy] Having problems with FlashFloppy i get the basics but i need help! (#341)

Hi, I finally made some headway but I am now stuck at the next hurdle.

I found a program able to read the drive - HxD.

I can read, select, export and save the data but do not know which format it needs to be saved in.  I tried just saving the defaults and tried loading as a raw file and saved it as a disk image without error, however it is unreadable in my synth.

closed 2020-07-12 by keirf