keirf/flashfloppy #343: 2DD Data Transfer Rate 300 kbits/s (PC-98)

sfish2020, 2020-06-15:
FDC Transfer Rate were added 300 kbits/s mode, in 199x.

Data Transfer Rate at 2DD mode.
| PC-98 (1982 - 1993) : 250 kbits/s
| PC-98 (1993 - 2001) : 300 kbits/s

And, AT clone and PC-98 file format are almost same.
I hope to merge raw file format.

Here is my patch file.
 2DD Data Transfer Rate : Add to FF.CFG.
 pc98 raw img : merged to img_type.
http://junk.s21.xrea.com/files/FF-pc98.diff

keirf, 2020-06-15:
You can already get this effect via IMG.CFG. Give it a try. One thing I'd like to add there is an ability to distinguish image types by file size, rather than solely by a tagged file name.

Another option is to add a data rate field to 'struct raw_type' and perhaps a pc98-300 (or whatever) host= type which would select a bunch of formats at your 300kbit data rate.

You can get the effect of interface=jpc with host=ibmpc and pin34=rdy

sfish2020, 2020-06-16:
The 1232KB format is not original by NEC. And, isn't by CASIO, too.

Here are IBM PC and NEC PC-98 formats list.
https://en.wikipedia.org/wiki/List_of_floppy_disk_formats

The original 1232KB format is used IBM PC < 8-inch > drive.
And, NEC was supported the compatible format at < 8-inch > drive.

At 5.25-inch drive, the compatible format was supported by NTT and Y-E DATA.
And, Japanese 16-bit PC were supported the compatible format.
Japanese 16-bit PC maker : NEC, Fujitsu, Toshibam, Hitachi, SHARP

At 3.5-inch drive, the same maker was kept supporting the compatible format.
And, AT clone maker were supported the compatible format for Japan Market
 from about 1995. So, IBM sold PC with 4 mode FDD (+1232KB, +2ED) in Japan.
The 3 mode FDD (+1232KB) for AT clone were popular parts in Japan
 from 1995 to 2010. 

IBM PC < 8-inch > drive : 1232KB format supported
IBM PC < 3.5-inch > drive for Japan Market : 1232KB format supported

And, casio_type (CASIO FZ-1) is the < 8-inch > drive compatible format.

I don't know NEC and CASIO original format.

And, jpc i/F was used.
: Japanes 8bit, 16bit PC
: MSX for Japan Market
: AKAI S900, S1000, S1100
: Ensoniq Mirage 
: KAWAI Q-80
: KORG DSS-1, DSM-1
: Roland S-750

Why akai-s950 was selected ?

keirf, 2020-06-16:
Regarding the akai-s950 choice, that's pure ignorance on my part at the time. I would be happy to add `interface=jppc` and `=jppc-hdout` if you like (P34=RDY, and P2=HD_OUT for the latter). Would that be agreeable?

Regarding a litany of extra formats, investigate IMG.CFG. I would be happy to incorporate example CFGs into the examples/ directory of the FlashFloppy distribution. Also I will be happy to allow IMG.CFG layouts to be tagged by file size instead of file name if you like (I mentioned this before). I'm not particularly engaged with extending the `host=` option further.

closed 2020-07-14 by keirf