keirf/flashfloppy #344: One Gotek drive behaves different to another on same cable

jms2pop, 2020-06-27:
Thanks for writing FlashFloppy - I find it is excellent and I have never had any problems with it until now. This might not be an actual problem with the software but maybe a new variant of Gotek drive, but I thought you would want to know about it.#

I have 2 Goteks, one bought this year and an older one. Both work fine on their own. I tried to set up the new one as a dual drive setup alongside a real 3.5" floppy and I experienced problems. Some of these might have been down to cable issues so to eliminate  possible causes I set up 2x Goteks on the same (new) cable and jumpered one as DS0 and the other as DS1. This is an Acorn machine so FF.CFG has interface=shugart and host=acorn for both drives. The cable is flat (no twist).

Both drives work, but subject to this quirk: the newer drive will not respond unless I have accessed the older drive first. After I have accessed the other drive, it works fine. Flipping which drive is 0 and which is 1 does not change this. Also the new drive's speaker doesn't work, but this could be a poor connection so can probably be disregarded.

Has anyone else reported this kind of problem before?

keirf, 2020-06-28:
Are both Goteks configured identically? Reset flash config if you are unsure (press both buttons for a few seconds while no USB stick is inserted). Don't disregard non-working speaker: Is it connected at JB as usual? If so, and configuration is not unusual (messing with speaker volume) then something electrical is wrong. Could be the speaker, but maybe not.

Not an issue I've heard of before. So there are zero problems if the 'bad' Gotek is connected alone, nor after a second drive has been accessed at least once?

keirf, 2020-06-28:
Also are both running up to date firmware? And have you tried swapping their order on the cable?

jms2pop, 2020-06-28:
Hello Keir, and thanks for getting back to me about this apparently strange fault!

 

To answer all your questions:

 

Are both Goteks configured identically? Reset flash config if you are unsure (press both buttons for a few seconds while no USB stick is inserted). 

Yes as far as I know – both USB sticks contain the same FF.CFG file. I have not tried pressing both buttons as you describe, but I have updated to the latest firmware so I would imagine that having new firmware plus the same FF.CFG should mean that both are set up identically.

Don't disregard non-working speaker: Is it connected at JB as usual? 

Yes it is. I can’t remember if it worked when I first installed it, but my recollection is that did, but quieter than the other drive.

If so, and configuration is not unusual (messing with speaker volume) then something electrical is wrong. Could be the speaker, but maybe not.

I’m going to look into this once I have extracted the Gotek from where it is fitted – I can’t get the lid off at the moment.

So there are zero problems if the 'bad' Gotek is connected alone, nor after a second drive has been accessed at least once?

That’s exactly it. I have solved the problem for the machine I am building by simply removing the “bad” Gotek to use it elsewhere on its own. I have now fitted my older Gotek alongside a real floppy drive in this machine, and they both seem to work together perfectly.

Also are both running up to date firmware? 

Yes, v3.13 (installed it yesterday). Previously I had v2.13 on one and v.2.14 on the other, which I thought might be a problem, but when I updated both drives it made no difference.

And have you tried swapping their order on the cable?

 

I think so, although I wasn’t absolutely forensic about this. Back when I was trying to get the “bad” Gotek working with a floppy drive (which is the intended setup) I tried many combinations of different floppy drives, positions on the cable, etc. When I switched to 2x Goteks I certainly switched drive IDs but I don’t think I swapped position on the cable. The “bad” one was always the first one on the cable. Currently my (working) setup has a real drive as the first on the cable and the “good” Gotek at the end of the cable.

 

As I write this it dawns on me that I need to swap them round to see what happens – I’ll let you know.

 

By the way, the “bad” Gotek has “Gotek system” silkscreened onto the board but the older one doesn’t.

 

John

keirf, 2020-06-28:
Don't assume anything about cached config. Make sure both goteks are reset as I have had weird issues reported before and solved by this.

Swapping on cable is an easy check so definitely do that.

Goteks are schematically simple and I wouldn't expect any real surprises between old and new.

jms2pop, 2020-06-28:
I have swapped them on the cable; this didn’t make any difference.

 

I also reset both of them. This didn’t make any difference either, but I want to check I did it correctly because a couple of things happened which I wasn’t expecting.

 

How I did it was to remove the USB stick and power up with both buttons held down. The Goteks responded with a message about USB updating, which is the same as what happened when I updated the firmware. Is that right? Also I noticed that the one which previously had v2.14 installed also displayed “v2.14” after the reset process. However, when I powered off then on again it displayed v3.13a as expected.

jms2pop, 2020-06-29:
I have now looked inside the new Gotek. As you suspected, it is identical to the old one except for not having the internal red "power" LED. The only other difference is that the speaker that I fitted is quite a bit bigger in diameter to the old one. I disconnected it altogether to see if this made any difference, but it doesn't. 

I also re-checked the message I get when resetting the drive. It is "FF Update Flash v2.14". However the firmware is actually v3.13a and this is what gets displayed normally.

keirf, 2020-06-29:
If you get that message, you are not resetting the drive. You need to press the buttons *after* power on but *before* inserting USB drive. You must be pressing them during power on.

jms2pop, 2020-06-29:
I was doing it wrong - glad I checked. When I hold down both buttons /after/ powering up, it says that it has been reset.

This has not changed the behaviour of the Gotek though.

I have discovered that I don't even need to have a USB stick in the "good" drive when I access it, in order to make the "bad" drive work. Just cataloguing an empty "drive 0" is sufficient to make "drive 1" work. But drive 1 doesn't work until I have done it.

keirf, 2020-06-29:
So one of these Goteks, same firmware, same configuration, doesn't work from power on until the other is accessed? Regardless of whether it is drive 0 or 1, and regardless of its position on the cable?

jms2pop, 2020-06-29:
Yes, your summary is correct. It's definitely the same FF.CFG because I copied it from one to the other. Both are on firmware 3.13a.

Regarding resetting, I can hold down both buttons for a while and get it to report "Reset Flash Configuration". I can also select Factory Reset from the main menu, which seems to have the same effect. But on the problematic Gotek, if I hold the buttons down during power on (as if updating), it still reports "FF update flash v2.14". So I'm still not convinced that I am resetting it correctly.

keirf, 2020-06-29:
The "FF update flash v2.14" is the bootloader version. It doesn't matter. It has no effect on normal operation.

Seems also that the bad drive is the bad drive? Ie it's not the good drive misbehaving and making the other seem bad?

Only other easy thing to suggest is seeing if it's a firmware regression by updating both Goteks back to something older like v1.2. Bit of a long shot. Without getting a logic analyser on these drives this is unlikely to get resolved further.

jms2pop, 2020-06-29:
That's fine - I can work around it by simply removing the problematic one and using it on its own. I just thought you might want to be aware of it in case anyone else find the same thing. Let's hope nobody does! Thanks - John

jms2pop, 2020-07-03:
Hello Keir, 

 

Sorry to bother you again, but I think I may have spotted the source of this problem.

 

I reverted to my intended setup of Gotek plus floppy drive, but on testing it I found the problem still existed – it was intermittent, less common with this particular Gotek for some reason, but still present. One totally repeatable test was this:

 

<Switch machine on>

*CAT 1

 

….this would always produce nothing, rather than catalogue drive 1.

 

If I then did:

*CAT 0

*CAT 1

 

… I would get the intended results (two catalogues), and both drives would generally work OK after that. I found that this behaviour didn’t change if I tried a different floppy drive or a different BBC micro. 

 

However I think I have just identified what is going on. As I understand it, when cataloguing the disc the Track 00 signal on pin 26 should see a negative edge and then go back high again. At least, this is what I see on my meter when everything is working. However when I boot up the Gotek, pin 26 is constantly low, and it remains low until after I have catalogued the Gotek drive, after which its steady state is high.

 

Have I found something significant here? Shouldn’t pin 26 always be high unless track 0 is being accessed?

 

John

keirf, 2020-07-04:
Trk0 should be LOW when the drive is on track 0 and the drive is selected. Flashfloppy powers on at track 0 so that part makes sense. If you believe it asserts trk0 while the gotek is not selected (ie its drive select line is not low) then that would be a bug!

keirf, 2020-07-04:
Here is a firmware which starts the Gotek drive at cyl #2 (and hence with TRK0 signal high). It really shouldn't change behaviour of anything. Note that DSKCHG (pin 2) will still be LOW for this drive even after an image is mounted, until a head-step command is received. That's normal Shugart behaviour.

[ff_344_1.zip](https://github.com/keirf/FlashFloppy/files/4872228/ff_344_1.zip)

jms2pop, 2020-07-04:
> From Keir Fraser <notifications@github.com>
> To keirf/FlashFloppy <FlashFloppy@noreply.github.com>
> Cc jms2pop <helenandjohn.simpson@ntlworld.com>,Author <author@noreply.github.com>

Ok, thanks for confirming how it's meant to work. What I'm saying relates to edges rather than levels.

 Attempting to catalogue drive 1 asserts DS1 correctly, and I would imagine that the drive is already at Track 0 anyway, so the combination of ds1 and trk0 being asserted at this point is correct. However, because Trk0 starts off in the low condition there is no negative edge in this situation and it seems this prevents the catalogue from being read.

Using a meter on pin 26, there is a distinct difference between what happens when accessing drive 1 immediately after the machine starts up, and subsequent accesses. The first time, trk0 is low and stays low, and I don't get any response from drive 1. During subsequent accesses (after I have first accessed drive 0) the signal always starts high, then goes low momentarily, then goes high again - and a catalogue is produced. It is necessary not just to select drive 0, but also to read it, in order to make drive 1 usable.

This applies for all accesses to drive 1 not just cataloguing, its just the easiest test to do.

Not having a scope I can't see what happens when I access drive 0 when it is in this initial state. I guess it blips trk0 high and then pulls it low again so there is a negative edge?

To prove that its the Gotek holding one of the bus lines at the wrong value, I have tried pulling the ribbon cable out of the back of it while Drive 1 is in the "not responding" state. When I do this, it responds immediately.

 I wonder if this is the cause or just a symptom, but it does illustrate clearly the problem that I'm having.

jms2pop, 2020-07-04:
Thanks - ill try it

Date: 04/07/2020  08:12  (GMT+00:00)

keirf, 2020-07-04:
TRK0 is not a level signal not an edge signal. From a TEAC data sheet for example (just one it's easy to find, and is a basic Shugart drive): "Level signal to indicate that the head is on track 00". Electrically, it could be as simple as a sensor bolted to the chassis at the track-0 location, coupled to a pick-up on the head assembly: This would then feed directly to the TRK0 output pin, gated only on the drive-select signal (via a 74LS38 gate for example).

Just to be clear: drive 0 is your Gotek, drive 1 is your real drive?

jms2pop, 2020-07-04:
Yes that’s correct. Not tested it yet but got as far as getting it onto a stick!

jms2pop, 2020-07-04:
I flashed the .upd file successfully to the drive, but the behaviour hasn't changed. Trk0 is still low on startup.

When I issue the command to catalogue drive 0 (the Gotek) this is what happens at pin 26:

- Initially it is low
- It goes high (maybe this is your modification?)
- It goes low again as the catalogue is output
- It reverts to high.

Its that initial low that is different to all subsequent accesses. Asserting Drive 1 doesn't make it go away.

Date: 04/07/2020  08:44  (GMT+00:00)

keirf, 2020-07-04:
Is this the original troublesome Gotek by the way, or the one you thought was okay?

I just did a quick startup check on my test Gotek, no ribbon cable (so the drive cannot be selected or signalled in any way). Powered on and checked pin 26. High (5v). Perhaps check your pin 26 after power on with the ribbon detached?

If you have an OLED screen on your Gotek, what track number is displayed at power on (while TRK0 is low)?

jms2pop, 2020-07-04:
This is the “good” one.

If I power it up without a cable, I still get 0v on pin 26. The OLED says we are on track 2, whereas on my other Gotek it starts up on track 0.

Incidentally your trial firmware from this morning doesn’t let me access anything else on drive 0 apart from the catalogue, and drive 1 doesn’t work at all now. It does pull pin 26 low, but reports an error rather than hanging.

I feel sure we are into something here.

By the way I fixed the non working buzzer on the other Gotek. The glue had seized up the vibrating part, so this is definitely not relevant.

keirf, 2020-07-04:
Can you attach your FF.CFG here? What jumpers do you have at the rear of the Gotek?

jms2pop, 2020-07-04:
See attached. I added “pin34=nc” yesterday as an experiment – I know this line isn’t connected at the computer end, but I wondered whether the two drives were maybe influencing each other in some way so I thought I would disconnect them in software.

 

There is only one jumper fitted, which is S0. There is a buzzer attached to JB and the rotary controller on the usual pins. I did try fitting jumpers at other positions as an experiment, but it did not have any effect.

 

John

keirf, 2020-07-04:
I don't think you can email it as an attachment. You either need to upload it somewhere and link to it, or you can do attachments in the Github issue tracker UI proper.

jms2pop, 2020-07-04:
Try this:
[FF.zip](https://github.com/keirf/FlashFloppy/files/4872871/FF.zip)

keirf, 2020-07-04:
Ok, nothing too exceptional in that FF.CFG. I also tried it on my test Gotek and could not reproduce the problem. I also ran the following test firmware, which displays on the second row, before "T:", a 0 if pin 26 is being driven low, else 1 if being driven/pulled high. On my test Gotek it displayed 1, as we would expect. Give it a go.

[ff_344_2.zip](https://github.com/keirf/FlashFloppy/files/4872909/ff_344_2.zip)

jms2pop, 2020-07-04:
It displays the "1" ... but the multimeter shows that the actual pin voltage is low (ie, the display is initially not correct). When I catalogue the disc, the voltage goes high, low, high and the display faithfully reports this. But the initial "1" is not right.

As a side-effect (which I'm hoping isn't a surprise to you!) loading from the disc still doesn't work, in fact trying to boot the game that is on there also causes activity on drive 1.

keirf, 2020-07-04:
Well it is surprising, as the changes I made to the firmware shouldn't break anything, but then again something is weirdly going wrong here so anything could happen really.

You need to take some voltage measurements on chip U1 on the Gotek PCB at power on (when TRK0 is the wrong value). Voltage at pins 3, 4, and 14 please. Reference to ground (which you can pick up at U1 pin 7, or loads of other places of course).

jms2pop, 2020-07-04:
OK:
First I checked that pin 26 was low (it was). These are all relative to U1pin 7:
pin 3: 3.3V
pin 4: 4.3V
pin 14: 5.1V

keirf, 2020-07-04:
Okay, so this is the problem: U1 is a hex inverter, and pin 4 should be the logical complement of pin 3. But both are HIGH. How can this be?

1. First note that use of HC04 in the Gotek is actually something of a bug. The voltage threshold for recognising a HIGH input on HC logic is 70% VCC. VCC here is 5.1v and so the threshold is around 3.5v. And 3.3v is less than that.

2. But obviously this usually works, in hundreds of thousands of Goteks. The threshold in practice would typically be less than 3.0v. So why is yours misbehaving, and possibly across more than one Gotek?

3. I would conclude perhaps the Acorn is kicking out more than 5v at power on? Just a theory. This would raise the input threshold, and may end up latching the HIGH incorrectly as a LOW, until the input has a large voltage change (ie. when firmware next drives TRK0 LOW properly, then HIGH again).

To test this, is it possible to power the Gotek outside your Acorn from another voltage source? For example your USB-TTL serial adapter would do. Just need to power on the Gotek with nothing but power connected, and see voltage on pin 26.

keirf, 2020-07-04:
Another possibility is that excessive voltage at some other time has damaged the 74HC04 and caused this erratic persistent behaviour.

jms2pop, 2020-07-04:
The PSU is making 5.2v so I’d say that is a sound theory!

keirf, 2020-07-04:
The thing is it may also overshoot at power on. You might consider looking at the psu. As for fixing the gotek, if you're handy at soldering then replacing U1 with 74HCT04 is a good call as that has correct voltage thresholds for driving with 3.3v logic.

jms2pop, 2020-07-04:
Yes I think that could be easier than adjusting the PSU voltage. I can do the SMD soldering reasonably well. I'll seek advice...
I tested the "bad" Gotek with an external power supply and yes, pin 26 goes to 5V when switching on.

keirf, 2020-07-05:
Yeah I'd ask around on Stardot, that seems to have some good people on it. I'll close this again for now but feel free to follow up with your conclusions, or more questions.

closed 2020-07-05 by keirf

jms2pop, 2020-07-08:
Hello Keir,

 

I replaced the 74HC04 with a 74HCT04 and I am very pleased to report that it seems to have cured the problem!

 

Thank you so much for all your patient help with this.

 

John

keirf, 2020-07-08:
Brilliant! I'll have to bear this issue in mind as a possibility for future bug reports. Thanks for the update!