keirf/flashfloppy #345: General Music (GEM) WS1

ojaksch, 2020-07-07:
Not an issue, but a success story: Just repaired my wife's GEM WS1 (leaked condensators and backup battery) and built in a gotek w/ flashfloppy. After testing some settings it turns out that WS1 is working perfectly with
**interface = ibmpc** and setting the drive-select jumper to position **S0**. Sadly 720kb disks only ;)

Maybe this information could take it's place to Wiki's Host-Platforms.

keirf, 2020-07-12:
Can the WS1 do 1.44M disks? I know some GEMs can even do 1.6M, which requires `host=gem` in FF.CFG, but I don't think that includes WS1.

One thing you could try is `interface=ibmpc-hdout`.

ojaksch, 2020-07-12:
Thanks for the suggestion. I've tried both parameters and combinations of them; the WS1 is still working with it's 720KB images then, but neither with 1.44MB nor with 1.6MB ones...still complaining about "corrupted data on disk". I suppose that WS1's floppy controller isn't capable of these formats.
But hey, no matter, there're is space for thousands of 720KB disk images now.

keirf, 2020-07-14:
Added to the wiki, thanks!

closed 2020-07-14 by keirf