keirf/flashfloppy #348: Access slightly larger floppy disk images?

matthew65536, 2020-07-11:
I'm using this device on the PC and it works well, but it has problems with slightly larger disk images about 1.68 MB, and those images can be put on a floppy disk, but the gotek just ignores the images.

keirf, 2020-07-11:
Images exactly 1.68MB size (1720320 bytes) should not be ignored. They should be recognised. Have you changed `host=` option in FF.CFG?

matthew65536, 2020-07-11:
For some reason, it will recognize the image, but the program on the PC will say the files inside are corrupted, and i'm not sure why because .img file works just fine on a virtual machine, and yes the "host= " is set to unspecified because it on a IBM PC compatible

keirf, 2020-07-12:
It may depend on PC OS version and BIOS. I'm unclear on what the general support is on older PCs for reading DMF (1.68M) floppies. You could try converting to HFE v2 format using HxC software (https://hxc2001.com/download/floppy_drive_emulator/HxCFloppyEmulator_soft_beta.zip) and see if that image format works better.

Or you could try this updated firmware which lays out the 1.68M tracks a little differently:

[ff_348_1.zip](https://github.com/keirf/FlashFloppy/files/4908124/ff_348_1.zip)

My suspicion is they are being handled correctly, as I'm pretty sure I'd have had other support tickets if they weren't. Plenty of people using Goteks on PCs to install Microsoft products...

keirf, 2020-09-02:
Dead ticket. Closing.

Leo-PL, 2020-10-12:
@keirf I had exact same issue, trying to install Win95 to a '90 386SX laptop, and the build you posted here worked for me. Could you share the patch or push a branch containing it?

Leo-PL, 2020-10-12:
Ah, it was too soon to speak. I got the files to list properly, but copying them to HDD fails, either by "Not ready" or "Sector not found" error. Still, I'd like to help out solving this. I'll try converting the images to HxC format too.

matthew65536, 2020-10-12:
Yeah. I meant to reply. I know my disk images are good, they work fine
under VM, but DOS acts like it can't find the sectors or the disk is
corrupt.

keirf, 2020-10-13:
@Leo-PL and @matthew65536 are you both talking specifically about 1.68MB (DMF) images? And in IMG file format?

I'm going to attach the latest firmware build for you to try. If it's still problematic, we can tweak the track geometry using IMG.CFG which is more flexible in recent builds.

[ff_348_2.zip](https://github.com/keirf/FlashFloppy/files/5370100/ff_348_2.zip)

Leo-PL, 2020-10-13:
I got the same results with this build. The copy process seems to fail just after switching to side 1. I'll try booting Linux on that machine and testing with `dd`. 
I'm using  "Windows 95 OSR2 (OEM) (3.5-DMF)" available over at WinWorldPc, specifically Disk02.img with SHA256 991b7d929bd83413abc8b2c1f9e5dfa9ad10f55d8221acddd4dac197d724cba3.

Maybe gathering of some logs could help?

keirf, 2020-10-13:
Try unzipping this IMG.CFG to the root of your USB stick. It will redefine the track geometry for 1.68MB IMG files. Specifically it will remove the sector interleave used in the default layout built into the firmware.

[IMG.CFG.zip](https://github.com/keirf/FlashFloppy/files/5373468/IMG.CFG.zip)

Leo-PL, 2020-10-13:
Still nothing. I guess it's the right time to test in a newer machine.

matthew65536, 2020-10-13:
I am talking about the 1.68 MB img formatted disks, and it may be a day or
two before i can get back to you about the new firmware.

keirf, 2020-10-14:
If someone has a real 1.68mb disk it would be really interesting to get a dump of it (eg with Greaseweazle)

Leo-PL, 2020-10-14:
I might try to format such a disk using 2M on a real FDD, or to write one of those Win95 DMF disks, and then have a go at it with Greaseweazle. I just need to pick up a Blue Pill from the hackerspace.

Leo-PL, 2020-11-12:
I flashed my Gotek with v3.21 and managed to copy over contents of one DMF disk successfully under MS-DOS 6.22. MD5 Checksums did match. So this is a huge step forward!

Writing did work partially - I could successfully write small files to the disk, but got "Sector not found" error trying to write anything bigger, than what would fit in track 0. However, what was written successfully, wasn't corrupted after opening an image on my "modern" PC.

Debian 1.3.1 copied over a truncated image - I tried to do full image to verify checksum on it, but I'm pretty sure, it isn't related to FlashFloppy itself, but probable lack of support for 21-sector-tracked FDDs in Linux kernel of that era.

keirf, 2020-11-12:
You cannot traditionally write to DMF disks anyway as the sector gap is too short. It's a DISTRIBUTION format.

This doesn't really apply to Flashfloppy running an IMG file though, where the gap doesn't really matter...

Leo-PL, 2020-11-12:
Yeah, true that.

I should have tried writing with 2M or similar tool, without that DOS in fact _should_ complain. For me the issue is resolved, though I'll probably do a test with 2M anyway.

@matthew65536 how about you?

closed 2020-12-02 by keirf