keirf/flashfloppy #349: Problem with 720DD in pc compatible

malloc32, 2020-07-12:
Hello, I have a Philips NMS9100 (PC Compatible, XT 8088-2, with one real drive and a gotek).
My problems are that I can't format a 720kb image, always the computer formats it to 360KB, even I have tried to make a backup from a real diskette 3.5" 720kb using diskcopy command and I have an error "drive types or diskette types not compatible". Using MS DOS 3.21 and 5.0. If I run command "chkdsk" it say that is an 720KB floppy, but when format it, always 360KB. I have try for force size using "format a:/f:720 in 5.0 dos version, but it doesn't work, say that is incompatible.

My gotek is a SFRC9220 flashed with flashfloppy v2.14, I can read floppies with it, select, play games...... all seems to work, but when I copy or format the size seems to be 360kb.

I have play with jumper JC(close- open), select S1. In ff.cgf file "interface = ibmpc", even I have tried to set manually the image parameters in img.cfg

cyls = 80
heads = 2
secs = 9
bps = 512

but when I format a 720 image, always when cyls arrived to 40 then finish and print floppy parameters (360KB).

Nothing works. I made empty images with winimage (720KB) or using a downloaded image from internet (dos 3.5" 720kb).

Thank you.

keirf, 2020-07-12:
Do you have an example img file, and also perhaps your FF.CFG you can zip and attach here?

I do wonder if this is simply BIOS expecting the drive to be 360k? But if you were to swap the drives can you format the real drive 720k no problem?

malloc32, 2020-07-12:
Sorry... I change drivers and real drive formatr only 360KB.... but if I boot with official Philips floppy MS DOS 3.21 then work at 720KB real drive and gotek with flashfloppy, so it is a "drivers" problem, I don't know that in old DOS this can append... Sorry about disturb you.
Thank you very much. Porblem solved.

closed 2020-07-12 by keirf