keirf/flashfloppy #352: IBM PS/2 (in this case Model 30-286) Support

drdpj, 2020-07-24:
The Model 30 drives are 1.44MB,  34 pin with 5V and 12V on pins 3 and 6 of the ribbon cable respectively. People have successfully had goteks with the stock firmware running on these systems in place of the original drive, I am unfortunately unable to get it working with FlashFloppy - DOS keeps reporting a general failure, the images are good, I've messed around with pin 2 and pin 34 in various permutations (pin 34 low will result in a "drive not ready" error rather than general failure).

Save for the power, the pinout is standard IBM 34 pin. Some modern drives require 1k-10k(ish) pullups to 5V on 8, 26, 28, 30 and 34 to get them to work. The Gotek already has pullups here and it's know to work with the stock firmware, so it would seem Flashfloppy isn't behaving in the same way and consequently not getting on with the PS/2.  The question is what? 

[FFLOG.TXT](https://github.com/keirf/FlashFloppy/files/4974255/FFLOG.TXT)

 I have tested the gotek on another machine, with the same SD card, so I know that it's working properly. Logfile attached but there's nothing particularly illuminating in there that I can see?

keirf, 2020-07-24:
Have you had it working with stock firmware?

drdpj, 2020-07-24:
I haven't, but there's a youtube video of it running with stock and a
number of accounts of it running with stock on various websites. I wouldn't
know where to find the stock firmware to get it back on to test it!

keirf, 2020-07-25:
Have you tried `interface=ibmpc-hdout`?

keirf, 2020-07-25:
There must also be the possibility the problem is not the Gotek itself.

drdpj, 2020-07-25:
The original floppy drive works (mostly! It's very grumpy about
formatting). I did try the ibmhd setting, it didn't work but will give it
another try. Have also had pin2=dens/ndens with no joy.

drdpj, 2020-07-25:
ibmpc-hdout didn't help :(

OK - another data point. Just tried a 720kb sector image which did work (up
until now it had been 1.44mb sector images). This reminds me of a thread I
came across yesterday that I can't find again where someone was adapting a
standard drive and could only ever get it to read 720k disks, never 1.44,
and never managed to work out why.  I can dump the reference disk with
greaseweazle if that would be useful? I think it more likely that it's
something that's not being signalled somewhere though?

d.

drdpj, 2020-07-25:
Cracked it, I think - 
[image]

Pin 4 is NC, but actually does connect to the drive type line: 

[image]

Putting the jumper on that pin, which pulls it down seems to cause it to start behaving properly (I've also got pin02=ndens, as PS/2 density select when it is used is inverted, but I did that at the same time as the jumper, so not sure which is having the impact!)

That identity thing isn't, so far as I can tell, meant to be used on the Model 30-286, but clearly it is!

d.

keirf, 2020-07-25:
So with the jumper applied can you read 720k and 1.44m disks? Does it just identify a HD-capable drive? Or should pin 4 indicate what density is currently in use?

drdpj, 2020-07-25:
Yes, both image types work. It looks like it identifies drive type, but the
circuit diagram labels it NC and reading around suggests it wasn't used on
the early PS/2s...looks like it is needed in this instance so should
probably go in the wiki?

drdpj, 2020-07-25:
(along with making sure you've snipped off pins 3 and 6 unless you like
carbonised ribbon cable!)

keirf, 2020-07-25:
Is the 5v/12v thing not indicated on the schematic? I thought I'd see it in the detail for J12!

drdpj, 2020-07-25:
No, none of the power seems to be in the PS/2 schematic.  If you forget to
swap to the special cable you made, all sorts of things can go wrong...
[image: image.png]

>

keirf, 2020-07-25:
https://github.com/keirf/FlashFloppy/wiki/Host-Platforms#ibm-ps2

closed 2020-07-25 by keirf