keirf/flashfloppy #353: Support for 360K IBM images in 720K drives

qoq, 2020-08-06:
I have a PC XT clone that that works with 360K 5.25 floppies (40trk/2sides/9SPT), but its internal floppy drive actually has 80 tracks/2sides. So, in BIOS INT13 skips every other track and pretends it has a 40-track drive. Cannot find a way to feed it 360K images through FF, can this be supported?

keirf, 2020-08-06:
Convert to HFE using HxC software (https://hxc2001.com/download/floppy_drive_emulator/HxCFloppyEmulator_soft_beta.zip).
1. 'Load' your floppy image
2. 'SD HxC Floppy Emulator settings' -> click the 'Double Step' tickbox at top right
3. 'Export' as HFE file (should be the default export type).

Use the HFE on FlashFloppy and it will be automatically instructed to double step.

Another possibility I could add to FF.CFG in future is 'force-80cyl = yes' and automatically double step for 40cyl images. But this option does not exist at the moment.

qoq, 2020-08-06:
> Convert to HFE using HxC software 

It helped. I can read disks, mostly. For whatever reason, with HFE images, when reading some files, it goes beyond cyl 42 and errors out. Direct support in FF would be much appreciated. Should the option not go into IMG.CFG, so it could be selectively applied to tagged images?

keirf, 2020-08-06:
Which FlashFloppy version are you using?

qoq, 2020-08-07:
3.15

qoq, 2020-08-07:
OK "fixed" everything for now by patching BIOS to not double step. Patiently waiting for the new config option to appear in FF.

keirf, 2020-08-09:
The HFE issue is actually a bug. I can fix that for you. Separately you still want direct IMG support?

keirf, 2020-08-09:
Here is a firmware to test which should fix double-step HFE images. Please test.

[ff_353_1.zip](https://github.com/keirf/FlashFloppy/files/5047451/ff_353_1.zip)

keirf, 2020-08-17:
I'll assume it works.

closed 2020-08-17 by keirf

qoq, 2020-08-18:
Just tested... HFE is fixed, thanks! Would still love to have double-step IMG option for 40-track images.

commit 58614c27 (2020-08-17): hfe: Fix double-step cylinder limit
Fixes #353