keirf/flashfloppy #355: Autorecognize ADF imagas

kipper2k, 2020-08-13:
Hi Keir,

 I notice that flashfloppy recognises the HFE images without the need for adding the 4 character suffix (.HFE) If you try to start an adf file without the .ADF then we get an error 31 message, is there any way that adf images can be recognised by the "DOS" file header so that flashfloppy knows what format the file is in and saves 4 characters so that the game can use a full16 characters in its name. This can cut down greatly on the name scrolling in order to see the entire name

Thanks

Brian

keirf, 2020-08-13:
Is this specifically for the autoboot.hfe file selector? I strip off the suffix on the lcd/oled. 

The problem is that not all adf will have a valid amiga filesystem on them and/or bootblock. Then the DOS header may not appear.

kipper2k, 2020-08-13:
Hi, thx for reply, this is when viewing oled /osd  display while cycling through listing, i am assuming that NDOS disks have a different header, is it possible to use the file size to indicate format as they all are the same size with mabbe a settings option to enable/disable the check ?

kipper2k, 2020-08-13:
as a sidenote, if i convert the adfs to the larger physical size of HFE then i can leave off the suffix and it will be fine but i am not sure if there is any loss of functionality etc as regards writing to the image for game saves etc

keirf, 2020-08-14:
First of all, default OLED/OSD setup should be 21-char rows not 16? Also it's unusual to use the file selector in concert with OLED/OSD -- it's usually faster to do direct navigation especially if you have a rotary encoder too.

Anyhow, there does seem to be an issue where I'm not stripping the file extension from the filename row. Which is what I do in direct navigation mode. So if the file selector mode is important to you we could have a go at fixing that.

kipper2k, 2020-08-14:
Hi Keir,

 Here is a little vid of my OSD display. http://www.kipper2k.com/oled.mp4 you can see Battle squadron display in HFE format that doesnt need the extension and is 15 characters and no scrolling, and then i toggle it to the adf version which is 19 characters long and is forced to scroll to show the full name.  There are a lot of games that could fit without the need for scrolling. If this is too much hassle for you to do i can always do a bulk convert to HFE. 

Thanks for looking at this

keirf, 2020-08-14:
Do you have OLED as well or OSD only?

kipper2k, 2020-08-14:
here is what is on the a1200... both display at the same time... http://www.kipper2k.com/a1200display.jpg

keirf, 2020-08-14:
Okay yes that makes sense, because it is a 1602 LCD, and OSD is (still, currently) tied to the row width of the LCD/OLED.

Right, yes I will have a look at doing you a patch. Out of interest could you attach here an example HXCSDFE.CFG with offending trailing .adf entries? Just so I know exactly what config version etc you have. Probably latest same as me but worth being sure.

Might be a few days. I have a backlog of 100 Greaseweazles to pack and ship.

kipper2k, 2020-08-14:
[HXCSDFE.ZIP](https://github.com/keirf/FlashFloppy/files/5074891/HXCSDFE.ZIP)

Thanks :+1:

keirf, 2020-08-17:
Please could you give this firmware a quick test and see how you get on? If you could try with some mixed-case file suffixes (ADF, adf, Adf, etc) as well, that'd be good!

[ff_355_1.zip](https://github.com/keirf/FlashFloppy/files/5085325/ff_355_1.zip)

kipper2k, 2020-08-18:
thanks, i will test it later this afternoon :+1:

kipper2k, 2020-08-18:
k, tested a few things... 
first, with no LED/OLED attached. The OSD will display up to 40 characters on the screen without scrolling, it does not show the .adf suffix (perfect). (tried .adf and .ADF) 
- with the small OLED attached the OSD will allow 21 characters to be displayed on OSD without scrolling,  the suffix is not displayed so can use max 21 letters for the game name without a scroll
- with the 1602 LED attached, the OSD will allow 16 characters to be displayed on OSD without scrolling, the suffix is not displayed so the max 16 characters can be used for the game name.

Thanks for the update Keir, very happy :+1:

commit d244c30a (2020-08-19): Remove file-type extension from HXCSDFE.CFG filenames. Matches direct nav behaviour.
Fixes #355
closed 2020-08-19 by keirf