keirf/flashfloppy #356: ERROR: Media Attach failed! on Amiga 500 Plus when used as external

Wetmackeral, 2020-08-14:
I can't work out why I keep getting this error message, I've followed the setup instructions to the letter.

Is this due to the fact it's plugged in as an external drive? How can I get around this?

keirf, 2020-08-14:
Have you tried reformatting the USB stick, or trying another stick?

keirf, 2020-08-15:
And do you get this error every time, or intermittently?

Wetmackeral, 2020-08-15:
Hi, thanks for this, using a different memory stick worked first-time - I was getting it constantly on every attempt with the previous stick. The non-working stick was a SanDisk Ultrafit 64GB USB3.1 (formatted to FAT32 from exFAT); the working stick is a SanDisk Ultra Flair 128GB USB 3.0 (natively FAT32)

closed 2020-08-15 by Wetmackeral