keirf/flashfloppy #357: UPD FF 3.15 my Rotary Encoder no longer works

retr0bill, 2020-08-18:
I had flash floppy 2.14 firmware on my Gotek and my rotary encoder worked great.

used the  3.15 upd file and updated to 3.15 and now my rotary encoder no longer works. flashing back to 2.14 restores rotary encoder functionality.

Any ideas?

Thanks!

keirf, 2020-08-18:
Do you have any rotary settings configured in FF.CFG?

The rotary handling is quite different in 3.x and works better with a range of encoders. But I could make the old behaviour available with a rotary=old setting perhaps?

retr0bill, 2020-08-18:
Thanks Keirf

No I didnt know the settings needed to be added to FF.CFG, I thought the behavior wold be the same as 2.14

It would be great if there was a setting in like rotary=old, then if it doesnt work then i can mess with ff.cfg

Thanks

Bill

retr0bill, 2020-08-19:
Keir,

I created a FF.CFG and added:

rotary=full,buttons

and still doesnt work, am I missing something?

Thanks

Bill

keirf, 2020-08-19:
Hi Bill,

No there is no need to mess with FF.CFG (yet) I just wanted to check if you had any existing settings which affect rotary encoder. Please change your FF.CFG line to the default `rotary=full` and then could you try releases v3.8a and v3.9a which are both available on the Downloads page. Then let me know if either or both works okay with your rotary encoder. Thanks!

What I will probably then need to do is upload a modified firmware here for you to test, ahead of the next FlashFloppy release.

keirf, 2020-08-19:
I will also ask on the FlashFloppy FB group whether anyone else is having rotary encoder issues. Yours is the first report, which doesn't mean it's not legitimate, but I have no idea how common this issue could be.

retr0bill, 2020-08-19:
Hi Keir,

I tested 3.8 and the rotary encoder works and allows me to scroll through all the files. However, the rotary encoder button doesn't work. 

I tested 3.9 and the rotary encoder doesn't allow me to scroll through the files. What happens is it scrolls through the other choices that appear below the file name like, clone, delete, exit to file selector, etc. And also here the rotary encoder button doesn't work.

ff.cfg is set to:

rotary=full

Thanks

Bill

keirf, 2020-08-19:
If you revert to 2.14 does the encoder button work? It is weird that it has stopped working. It could be a hardware problem?

keirf, 2020-08-19:
Also, here is a modified version of 3.15 with fixes which I may release as part of 3.16. But I would confirm everything still works okay in 2.14 first!

[ff_357_1.zip](https://github.com/keirf/FlashFloppy/files/5097506/ff_357_1.zip)

retr0bill, 2020-08-19:
I am wiring up a new rotary encoder. The switch no longer works on the one I have been testing. I will let you know what happens.

keirf, 2020-08-20:
Here is a new v3.16 release candidate. Worth a try with your new encoder: https://drive.google.com/file/d/1NpBrq4M01p3lEL6Hdx6Y84F_D2Ap7sBw/view?usp=sharing

fabriziolussana, 2020-08-25:
Hi,
I'm facing what seems to be the same or a similar problem as retr0bill's with 3.16. Rotary encoder, which used to work in 2.x, has become almost unusable. I have to turn it a lot (a variable number of clicks in either direction) to scroll one position and it doesn't always work (it seems to depend on how fast I turn the knob, the faster the more chances it works). 
I had rotary=full in ff.cfg.
Thanks

keirf, 2020-08-26:
Can you try 3.15 and also 3.13a and see how they work?

I may put in a rotary=old option. There must be some old roraries out there that don't like the new approach. Finding one method that works for every crappy encoder seems impossible. I revised the code based on testing of recent crappy encoders!

fabriziolussana, 2020-08-26:
With both 3.15 and 3.13a the rotary encoder is completely unresponsive.
Thanks

keirf, 2020-08-27:
I will put together a fixed version and attach it here for you to test.

keirf, 2020-08-27:
Here is a version for you to try. Please put `rotary = full,v2` in FF.CFG.

[ff_357_1.zip](https://github.com/keirf/FlashFloppy/files/5137969/ff_357_1.zip)

retr0bill, 2020-08-30:
Hi Keir,

I installed a new rotary encoder. I downloaded the FF_357_1.zip but when you unzip it is 3.16, is that what its supposed to be?

I installed it, and changed FF.CFG to rotary = full,v2.  The issue I have now is that after I choose the file to load, I don't need to push the rotary encoder, if I wait a couple of seconds it selects the file automatically. 

Is this the expected behavior?

Thanks!

Bill

keirf, 2020-08-30:
Yes, 357_1 is not really plain 3.16, it has fixes (specifically the 'v2' addition to the rotary= config option). But I guess your new encoder works anyway, without the v2 option.

To stop auto-selecting files and folders you need to add lines to FF.CFG:
```
autoselect-file-secs = 0
autoselect-folder-secs = 0
```

retr0bill, 2020-08-30:
Do you want me to keep using this version ? Should I remove the V2 from my FF.CFG?

Thanks

fabriziolussana, 2020-08-31:
I tested ff_357_1.zip / rotary=full,v2 and it does solve the problem. With that said, I tried all the encoders I have, and only the one seems to be seriously affected, so it must be because it's crappier than average. The rest all worked to satisfaction on regular 3.x builds.
Thanks.

keirf, 2020-08-31:
@fabriziolussana Thanks for the test. I'll have to decide whether it is worth including in v3.17. It's a small fix so I may.

@retr0bill You can switch back to plain v3.16 from the Downloads page, and revert your rotary setting to `rotary=full`.

retr0bill, 2020-08-31:
Switched to 3.16 with rotary=full and my rotary encoders are now working

commit 2341619c (2020-10-11): FF.CFG: New option "rotary=v2" uses v2.x rotary encoder logic.

This is more permissive than v3.x logic and may work better with
some cheap encoders which skip Gray-code states altogether.

Fixes #357
closed 2020-10-11 by keirf