keirf/flashfloppy #358: V3.15 doesn't show F-F after flashing SFR1M44-FEL-DL

mscheo, 2020-08-20:
My First time doing this. But I am sure it was successfully flashed. The "verify after upload" said again it was successful.  I removed those flashing jumpers and connected 5V-Gnd to the unit. The 3 7-segment LEDs didn't even get lighted up. Reflashed again. Same result.

So I downloaded V2.14 and flashed the GOTEK again.  When I connected 5V-Gnd to the unit, the 3 7-segment LEDs show "F-F". Phew!

Guess I will stay with 2.14 for a while as I learn more about using this wonderful piece of firmware on my Yamaha EL-900. Thank You, Keir.

keirf, 2020-08-20:
From v2.14 you can safely try an update to v3.15 using USB stick. See https://github.com/keirf/FlashFloppy/wiki/Firmware-Update

You can update again to v2.14 by the same method. Ignore the stuff about "Updating the Bootloader".

mscheo, 2020-08-21:
Yeah, with the 3.15 .upd file, I successfully updated to V3.15.  Thanks, Keir.

closed 2020-08-21 by mscheo

mscheo, 2020-08-21:
Hi Keir,

Thanks for your follow-up and notes.

Yeah, I could update to 3.15 using the .upd file. When I released the 2 
buttons, the 7-segment LEDs switched from "uPd" to "uSb" and then after 
a short while, to "000".
If would be good to show the current release at power on when there is 
no USB stick inserted, switching between "F-F" and "214" or "3.15".

I understand the OLED 128x4 could display 4 lines, but as of now, F-F 
could only shows 3 lines on it, right?

Thanks again for your work

Kind Regards,
MS Cheo