keirf/flashfloppy #360: Broken Gotek

hikodaci, 2020-08-27:
I have some floppy emulators by Gotek. It was working on the machine something happened and now it is not working. It is not flashed, it worked with original firmware.

Now I want to update it with flashfloppy firmware but When I connect it TTL cable to PC. It gives me error as "Unrecognized device... Please, reset your device then try again.

I reset the gotek from on it RST pins, also plug out and plug-in cable many times. But there is nothing.

May I want you to help me, please?

Best Regards

keirf, 2020-08-27:
Maybe the Gotek is broken? Are you using serial or DFU/USB method? I can't really give any clearer guidance than the wiki, and various Youtube videos I'm afraid.

closed 2020-08-27 by keirf

hikodaci, 2020-08-28:
I updated the firmware as flashfloppy v3.16.

But I don't know How to use it exactly. There is a picture which has to be in floppy to operate the machine.
[image]
You can find the files in the zip file.
[Robot.zip](https://github.com/keirf/FlashFloppy/files/5140739/Robot.zip)

Please tell me what I have to operate USB on the Gotek with these files. And please find the video of the Gotek after flashed.
[gotek.zip](https://github.com/keirf/FlashFloppy/files/5140752/gotek.zip)
 
Best Regards

keirf, 2020-08-28:
In all likelihood you need to put those folders into a disk image (IMG). Perhaps a 1.44MB IMG file. There is software on Windows to do this (WinImage for example). I would try creating a 1.44MB IMG file, copying these folders into the image, and then FlashFloppy can mount the resulting IMG file when you put it on your USB stick.

hikodaci, 2020-08-28:
This is the read from machine operation, so write from machine to USB ?

I will try to make img file.