keirf/flashfloppy #361: Possibility of mention in the wiki or web link for official manufacturing partner of OpenFlops?

simulant-systems, 2020-08-28:
Hi keirf, we've been working with Sukko Pera in order to make his OpenFlops design available to those wanting to buy a pre-made drive with also a donation to charity for each drive sold.

I wondered if you may be interested in adding it to the Wiki and FlashFloppy github if possible please? Here is the link:

https://www.simulant.uk/openflops

Let me know if you would like a sample posting out to check the quality etc or for testing. Thanks again for any help :)

keirf, 2020-08-28:
Quite happy to link. I must point out that two drive emulation is not supported. The board may be ready for it with the leds behaving as you state, but the firmware does not yet support it. There is no timeline for it either.

simulant-systems, 2020-08-28:
Ah apologies, when speaking to Sukko in planning for manufacture he mentioned it should be ready in the 3.15+ release, I'll amend the product page description to say it will be coming later. Thanks again and hopefully these being available pre-made may help with further features too. Best wishes

SukkoPera, 2020-09-03:
I think there's been a slight misunderstanding here: all I did was quote a post on FB where @keirf was saying he *might* work on the 2-drive feature after releasing a stable 3.15 (or whatever that was) release. I also added that the release had just happened, so maybe we could see this feature soon afterwards, but I never meant that it would happen for certain. Sorry if I wasn't clear enough.

simulant-systems, 2020-09-10:
Hi no problem Sukko, that was my misunderstanding so I am sorry too. The main thing though I suppose is that the hardware does support it. So at least it can happen hopefully which is great. I am also happy to post out Keirf some OpenFlops boards too if it may help at all, even just as reference or such. Thanks to both of you guys for all the hardwork, it is so great to have these floppy emulators available and now having OpenFlops there just means we finally have the open hardware option too :)

simulant-systems, 2020-09-10:
ps also just to add that I found Atari ST seems to only work with a jumper on M0 with Openflops and no other jumpers set. Yet on Gotek the ST just has a jumper on S0. So it may be worth mentioning under Atari ST on the wiki about M0 for the Openflops board or if there is any way I can help at all to document different settings for different machines and such I'll be glad to help. I don't mind adding them to the product page or even a different wiki but if @keirf is happy adding the Openflops stuff gradually to the FlashFloppy wiki then that would be good too to keep it all in one place. Let me know if I can do anything to help etc so there is more information out there for the users.

keirf, 2020-09-10:
I'm surprised ST would need jumper on MO. I've not had any other reports about that.

simulant-systems, 2020-09-10:
> I'm surprised ST would need jumper on MO. I've not had any other reports about that.

Yes seems to work great with M0 on OpenFlops, I'll let you know if any other host platform settings come up :)

keirf, 2020-09-10:
What I mean is noone else has needed to jumper MO, and it's very unlikely indeed that the jumpering would be different on Openflops assuming the jumper block is electrically the same and labelled the same. That would make no sense!

simulant-systems, 2020-09-10:
@keirf Scratch that, I've removed those Atari ST comments above as on further testing I got it working with just S0. I realised with an external drive plugged in the M0 just confuses matters. But yes S0 is working now on unplugging and replugging in the usb stick while the system is powered then resetting it etc.

One thing on Atari ST I am noticing is that I have to choose the disk image and reset each time for TOS to see it. Is this normal on 3.15 firmware? I suppose I should probably branch that off into a different github issue rather than this one if you want me to delete all the comments etc?

keirf, 2020-09-10:
No this ticket is fine. Atari ST has a ridiculous disk-change-detect mechanism where it looks for changes in write-protect signal. But this means that if you switch between write-protected images, the line never changes, and disk-change not detected!

Solutions: 
1. Don't write protect your images (do not configure this in FF.CFG, and ensure that FAT file image permissions on USB stick are writable)
2. Press Esc in TOS disk contents window. This apparently manually forces re-read of disk contents.

SukkoPera, 2020-09-10:
> What I mean is noone else has needed to jumper MO, and it's very unlikely indeed that the jumpering would be different on Openflops assuming the jumper block is electrically the same and labelled the same. That would make no sense!

As far as my intention was, the M0 position should be the same as on the original Gotek. Maybe @keirf can have a quick look [at the schematics](https://github.com/SukkoPera/OpenFlops/blob/master/doc/schematics.pdf) and confirm this (or test it on the OpenFlops board I sent you some time ago).

keirf, 2020-09-10:
@SukkoPera your schematic looks fine. No problem there! It seems this MO confusion is now resolved.

simulant-systems, 2020-09-10:
@keirf and @SukkoPera ah yes confirmed here on the ST. I'm set up on Sel0 disk 0 with OpenFlops (or S0 as it is called on the Gotek) and pressing ESC in TOS is working to force the computer to re-read the drive and show the right disks etc just as you say. :)