keirf/flashfloppy #362: KORG Triton synthesizer: writing does not work...

0xSeb-as, 2020-08-30:
Hi,

first of all thanks a lot for this project!

From the service manual of the KORG Triton synthesizer, I've been able to determine that interface is **ibmpc** (P2=unused, P34=DSKCHG), and unit is **S1** (pin11 'D_SEL1' is connected, pin10 'D_SEL0' is not).
Board is Gotek SFRC922D, my jumper configuration is S1 and JC (I know JC is useless as I have interface set as ibmpc).

Reading works fine (albeit the feeling is that it's a bit slow), but writing always ends with error...
I tried putting a jumper on MO, but it had no effect (btw I'm interested to know what MO is used for).
Maybe this can help: I don't control written file size (it depends on the content I choose to be saved), but the only time writing seemed to work is for a file that ended being 940 bytes (so bigger than 1 sector, and smaller that 1 track)...

Any idea?
Thanks!

keirf, 2020-08-30:
Try messing with the `write-drain` option in FF.CFG. Put it back to default (`instant`) if other values don't help.

0xSeb-as, 2020-08-31:
Thank you for your answer, however all 3 possible values for `write-drain` lead the same write error :(

I don't konw if it can help, but on the HW-side: **FDC** is a `HD63266F` and original **FDD** was `DF354H912C`.

keirf, 2020-08-31:
What image file type are you using? eg, IMG, HFE. It's possible the original disks have sector interleave or skew and this turns out to be more critical for writes. I would need to see a dump of an original Korg disk though. Or perhaps you could format an HFE image on the synth?

Other possible approaches: Try another USB stick, as some can be slow for writes. Also there is a logfile firmware you could try running which will log info on writes: https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive

0xSeb-as, 2020-08-31:
I'm using HxCFloppyEmulator software for any disk image operations.

I'm exclusively using .IMG format, standard 1.44MB HD created with it, then I copy original content from .zip file to disk image, so everything happens at file level, not disk image level.
My writing tests on the synth are performed on empty disk images.

Regarding formatting on the synth, what you mean is that I create an empty .HFE image first, then I format it on the synth, right?
I already used the logfile firmware to help on initial configuration, I'll give it a try too...

I'm gonna try faster USB sticks first indeed; current one is a standard and a bit old already 16GB Verbatim, I have some even older Corsair and Patriot sticks that should be faster.

keirf, 2020-09-01:
Yes, if you format an hfe image on the synth we can inspect it to see if there is a special sector arrangement.

chrispoacher, 2020-09-01:
I had this same issue Keir remember, It turned out that some usb keys wrote fine but some didn’t work. Never figured out why really as it worked on std gotek fw ok. Best just try a goo quality officially branded key at no more than 16gb usb2 type.

Chris

keirf, 2020-09-01:
Thanks Chris. There has to be a reason for it all the same. Perhaps a formatted HFE will shed some light. Also there is the logfile firmware where logging may shed some light. But in the end yes perhaps a different stick will solve it. :)

0xSeb-as, 2020-09-02:
Ok, so using that good old 2GB Corsair Flash Voyager GT... worked perfectly!
I did both quick and full format on .hfe images, on both sticks, but for some reason github doesn't want to upload the files to this message, so you can grab them [here](http://dl.free.fr/sE7qhSFqQ) instead...

Given the size of an .hfe image I assume there's a ton of low-level metadata, so I was expecting that the resulting formatted images would be identical for both sticks, but it's not the case it seems so maybe there's something here?

The only thing I didn't have time to test is the logfile-fw, and unfortunately I won't be able to in the coming week...
I really want to help in finding the underlying issue here if possible, so expect me to come back when I can perform more tests, but I'll still follow this thread in the meantime (even though the issue is not the one describe in the title)!

keirf, 2020-09-02:
HFE is a raw bit-level representation of each disk track, that's why the files are so big. Since formatting will vary/jitter a little as to where each track starts relative to the index mark, it is totally expected that HFE images will not exactly match.

Anyway I had a look at the HFE images and there's nothing surprising at all. There is no sector interleave or skew, they are absolutely bog standard 1.44M images like would be used on a PC.

keirf, 2020-09-02:
So I think logfile-fw comes next. A log of a write on the 2G stick and one that fails on the 16G stick and we can have a look at timings.

0xSeb-as, 2020-09-03:
I managed to find some time before being unavailable, so here are my hopefully interesting results (logs [here](http://dl.free.fr/bYsmRwm9i)).

On both sticks I have 0001.hfe and 0002.img.
Here's what I did: "select 1, quick format; select 2, quick format; select 1, save; select 2, save; select 1;"
For the Corsair, I added "save (on already selected 1); select 2;" (explanation below).

On the Verbatim, every write fails, as expected.
On the Corsair, write succeeds but only for .img, it always fails for .hfe (that's why I did it 2 times).

Another unrelated thing that might be another issue: as the Corsair stick has an activity led, I noticed that it's blinking continuously when HFE image is selected (even though no "disk" access is performed, as indicated by the gotek display), that is not the case with .IMG (happened on 3.15 I was using before as well as 3.16-log).

What do you think @keirf ?

keirf, 2020-09-03:
Do the quick formats fail as well as the writes?

0xSeb-as, 2020-09-03:
I'm tempted to say that quick format works: I have no indication of the synthesizer that it failed, and when the disk is selected again after formatting, its label went from 'nolabel' to the default one assigned by the synth.

keirf, 2020-09-03:
Okay let's leave the quick formats aside as they appear to all work on both sticks. The writes are interesting, as all the failures (IMG & HFE on Verbatim, HFE on Corsair) happen at exactly the same point during the write. No randomness there. I can't compare timings with the good IMG write on Corsair as the write is so long that I have lost logging data for the start of the write :( Perhaps it makes sense to do this with a smaller file write?

keirf, 2020-09-03:
As for stick timings, the Verbatim is inserting some 650ms+ 512-byte writes during the IMG write. That's pretty long (but not unheard of). It's strange that the write doesn't barf there, but at the same consistent point (after 16 sector writes).

The Corsair has clearly better IMG 512-byte sector write times: mostly 1.8ms, with the occasional ~37ms latency. Perhaps it's getting lucky getting away with the latter, or perhaps the KORG can tolerate those once in a while.

The Corsair has pretty consistent HFE write times too, so it's a bit of a surprise that barfs after 16 sectors like on the Verbatim.

0xSeb-as, 2020-09-03:
Ok I'm gonna try that, it'll be trial and error to find a size that exhibit this behavior maybe (as I said earlier, I can define which type of data is saved, but I have no idea beforehand about the resulting file size).

Also, I probably won't be able to do it before at least a week from now...

Can you roughly estimate a 'maximum' file size that shouldn't be exceeded for the log to be meaningful?

keirf, 2020-09-03:
One thing I'm not logging is simple end-of-write to start-of-read latency: Time taken to start streaming read data to the host after the sector write completes (from the point of view of the KORG host). There are things I can maybe do here by tweaking FlashFloppy's caching strategy.

That caching strategy, by the way, is why IMG files don't flash the USB stick when idle: The entire track data fits in Gotek RAM and so we stop reading the stick. That isn't the case for HFE, so we are continually streaming from stick even when 'idle' sat on the same track.

keirf, 2020-09-03:
As for largest file size I'd say maybe 20kB?

0xSeb-as, 2020-09-03:
ok thanks :)

keirf, 2020-09-03:
I'll wait til you or Chris Poacher has time to do some testing.

I think I need to move the mass storage and caching layer into a separate process, or otherwise make it asynchronous from the main emulation logic. Then I can issue prefetches, write back dirty buffers, etc without disturbing emulation (at least in some cases where I can be sure I won't need to issue a read that could get stuck behind a write).

The alternative would be to make more of the emulation logic event/IRQ driven but that's probably harder. I'll have to think about it.

keirf, 2020-09-05:
Probably the right thing to do is an overhaul of IMG handling where, when all tracks of a cylinder fit in Gotek RAM, I prefetch both heads and buffer all dirty sectors until heads are stepped or all sectors are in RAM. Write latency can then be hidden behind seeks (probably not entirely so there's a disadvantage there, but offset by much more accurate behaviour on writes and head-select).

Possibly I could do this by making more IMG logic asynchronous. But I think putting a simple process scheduler in FlashFloppy might make sense.

closed 2022-06-22 by keirf

KpynuH, 2023-10-04:
I have the same problem with Korg KARMA. Jumper at S1, interface=ibmpc as required. Must be the same as Triton. It doesn't read attached HFE images at all and has an "Error writing to medium" when I try to save all my programs and combis to file. I used USB 2.0 8GB Trancend, 4GB Silicon Power and even USB 3.0 32GB SanDisk.
I've went through quite a few posts and comments here and there by Korg owners. Most of them don't use FlashFloppy and there is no info about it at all.
Is it possible to attach basic FF.CFG and IMG.CFG files for download for Triton users just to make it work with FF Gotek from the box? I'm totally new to this and it is really hard to get through all that options and adjustments.

KpynuH, 2023-10-04:
> Probably the right thing to do is an overhaul of IMG handling where, when all tracks of a cylinder fit in Gotek RAM, I prefetch both heads and buffer all dirty sectors until heads are stepped or all sectors are in RAM. Write latency can then be hidden behind seeks (probably not entirely so there's a disadvantage there, but offset by much more accurate behaviour on writes and head-select).
> 
> Possibly I could do this by making more IMG logic asynchronous. But I think putting a simple process scheduler in FlashFloppy might make sense. The other advantage is that other image handlers (DSK, HFE, ADF) could then more easily benefit....

I have the same problem with Korg KARMA. Jumper at S1, interface=ibmpc as required. Must be the same as Triton. It doesn't read attached HFE images at all and has an "Error writing to medium" when I try to save all my programs and combis to file. I used USB 2.0 8GB Trancend, 4GB Silicon Power and even USB 3.0 32GB SanDisk.
I've went through quite a few posts and comments here and there by Korg owners. Most of them don't use FlashFloppy and there is no info about it at all.
Is it possible to attach basic FF.CFG, IMG.CFG and empty image files to downloads for Triton users just to make it work with FF Gotek right out of the box? I'm totally new to this and hardly imagined that I couldn't fin a solution for regular use for a week after installation as it is really tough for me to get through all those options and adjustments.

kulichbulich, 2024-04-27:
I also had a problem with data recording on the KORG Triton (I also tried an experimental one with asynchronous I/P). I found that FlashFloppy is sensitive to the type of USB Flash drive. I have tried a this USB Flash: SanDisk Ultra Fit 16GB USB 3.0 Flash Drive is working for me.

keirf, 2024-04-27:
@kulichbulich does that stick work even with v3.42 firmware?

MikeyMcG, 2024-07-14:
I have just bought a Gotek with Artery AT32F435 chip, and has Flashfloppy+ v3.42 preinstalled. I will (hopefully) be using it with my Triton Rack. I have it in "Native" mode and the jumpers set to S1, which I am assuming is the header on the left of S0 as the first 2 headers aren't marked on the board, with the second jumper across J5 and JA pins (as it is not required?). 

I have my 64GB Sandisk Ultra usb 3.0 stick formatted to FAT32, which the Gotek recognises. I have all the installation files for my EXB-PCM cards in individual folders, but not as images. When I create img files using imgburn, I get flashfloppy error 32 But the Triton is showing the usb drive as being unformatted.

From reading the wiki, I do not need to use any FF or CFG files for the drive, as it has been setup in hardware.

Can any of you tell me where I'm going wrong with this?

keirf, 2024-07-14:
You can only get error 32 if you have HXCSDFE.CFG in the usb drive. You don't want that, so delete it.

keirf, 2024-07-14:
That's a bad IMG file size. Probably not good floppy img files, unsurprising since imgburn makes IMG files for optical media. Try using diskflashback which is a new tool by Rob Smith. You should be able to Google its webpage and tutorial video.

MikeyMcG, 2024-07-14:
I tried installing diskflashback but it doesn't run on my pc (Win 10 pro, v22H2). I double-click and nothing happens.

MikeyMcG, 2024-07-14:
In the FF.CFG file, I set the interface to "ibmpc", the host to "akai", and the display-type to "oled-128x64-rotate" which sorted out the flipped screen as well. Once I'm able to create proper img files, I'll try again.