keirf/flashfloppy #363: Kuka KRC1

hikodaci, 2020-09-01:
I have a machine which is made by KUKA Robot. There is Win95 inside it. And the floppy drive reads as 1.44 Standart floppy.

I can read and the write files in explorer on the computer so there is a no different mod as dos or etc. 

I can use the GOTEK with original firmware on this machine. And I upgraded it for Flashfloppy firmware V3.16

After flashing, The GOTEK digits shows F-F and When I push the right button it shows 3 and 16. I think there is no problem with flashing.

I have created two different IMG files (empty and my files) with a software of WinImage on the Windows. (Dolu is meaning Full, Bos is meaning Empty)

[image]
[image]
I put two IMG files, FF.CFG and IMG.CFG files in the flash memory which is formatted as Fat32.
[image]
Also, I changed the interface in FF.CFG file.
[image]

The GOTEK shows 000 and 001 on its digits, I think they are my IMG files. But when I operate the read function on machine, It is not working. Also, writing is not working too.

What is the problem? And what have I to do?

keirf, 2020-09-01:
What's happened to your IMG file sizes? In the last picture they are no longer 1,440KB in size. That is why it's not working.

keirf, 2020-09-01:
Also you do not need FF.CFG or IMG.CFG files. Remove them. Also clear cached config by holding both buttons *after* power on, with no USB stick inserted, for five seconds until the display shows 'RST'. This clears any cached config.

Now make sure you have jumpers at positions S1 and JC at the rear of the Gotek. And no other jumpers.

hikodaci, 2020-09-01:
> What's happened to your IMG file sizes? In the last picture, they are no longer 1,440KB in size. That is why it's not working.
> 
> EDIT: The IMG files must each be exactly 1,474,560 bytes long. Please check that.

The last picture is wrong I know. I have tried to shows only config file settings in this picture, not IMG files.

My IMG files are 1.440KB too, and I think there is no problem with them.

[image]

I will try it again to remove the config files on the flash memory.

keirf, 2020-09-01:
I think actually the problem is the example IMG.CFG you had installed. Just removing that should fix your problem. You could leave FF.CFG and not reset your flash config.

hikodaci, 2020-09-01:
This is the machine and GOTEK.

[image]

[image]

I reset the GOTEK with push two buttons and it shows RST on the display.
The config files are removed on the flash memory. There are only 2 IMG files on it.

When I put the flash memory it shows me that,
[image]
Then it shows 000 and I can change it to 001.
However, when I Load from GOTEK to Machine, Machine showed errors.
[image]
And I Tried save to Floppy it showed this error.
[image]

keirf, 2020-09-01:
Definitely only the two IMG files on the USB stick? Does the green LED on front of the Gotek ever light up?

keirf, 2020-09-01:
And show a picture of jumpers at the rear of the Gotek.

hikodaci, 2020-09-01:
This is the inside of USB stick.

[image]

And it is the jumpers at the rear of the Gotek.

[image]

hikodaci, 2020-09-01:
> Definitely only the two IMG files on the USB stick? Does the green LED on front of the Gotek ever light up?

The Green Led on the front of Gotek is never light up.

keirf, 2020-09-01:
The green LED is the drive select indicator so if it never lights up, it's no surprise you can't read/write: the drive is never selected! This would also explain why you see the track indicator (`0_`) for a short while but never again: The track number is never changed because the drive is never selected.

Try the S1 jumper at S0, and also at MO and see if you can get that green LED to light at all.

If you are at all handy with an oscilloscope or logic analyser you could also look for signs of drive- and motor-select signalling on pins 10, 12, 14, and/or 16.

keirf, 2020-09-01:
You can also try all the above with JC removed too. This could be a simple drive-select/interface-select configuration error at the jumpers.

hikodaci, 2020-09-01:
> The green LED is the drive select indicator so if it never lights up, it's no surprise you can't read/write: the drive is never selected! This would also explain why you see the track indicator [...]

I tried it M0, S1 and S0 with JC and without JC but there is still no light up of green led.

Also I can see again _0 when I push the right button and it changes from 000 to 001.

keirf, 2020-09-01:
Then I start to wonder if it is a hardware or cabling problem. You need to fault-find why there is no drive-select reaching the Gotek. 

Another option is to try permanently selecting the Gotek by moving the S1 jumper to be horizontal across S0-JB on the lower of the two rows. This should permanently light the green LED.

hikodaci, 2020-09-01:
The Gotek is working on the machine with original firmware and its jumper is only S1.

But I want to make it with Flash Floppy.

keirf, 2020-09-01:
This would not be the first "I only changed the firmware" bug report where the firmware was not to blame.

keirf, 2020-09-01:
I would revert to jumpers at S1 and JC, those are almost certainly correct. Another thing possibly to try is a single-line FF.CFG containing `chgrst=delay-3`. This changes disk-change reporting. If it doesn't help you can revert to default `chgrst=step`. But my guess now is that in reflashing the Gotek the cabling has been upset.

hikodaci, 2020-09-01:
Should I make flashing with new one GOTEK? Can this problem be from GOTEK?

And How can I sure the upgrade firmware with flashfloppy is completed as successfully?

I remember the green led is light up yesterday when I try to read the disk, but it wasn't working still on the machine. I don't understand why it is not lighting now?

keirf, 2020-09-01:
The problem is unlikely to be the Gotek: The LED is basically driven by the host asserting drive-select, there is no smartness on the Gotek side to decide whether to light that LED. Far more likely the drive cable has gone bad or is loose at one end or the other.

If you saw the green LED light yesterday then that is excellent news: The IMG.CFG you copied onto your stick would at that point have been the thing stopping your Gotek from working (the example IMG.CFG overrides the default IMG disk layout and forces it to be a 720k DD PC disk!).

keirf, 2020-09-01:
If you have a supply of Goteks perhaps you should swap in a factory-firmware one and confirm it still works?

hikodaci, 2020-09-01:
> If you have a supply of Goteks perhaps you should swap in a factory-firmware one and confirm it still works?

I have another GOTEK which has original firmware. And it is working right now.

keirf, 2020-09-01:
And green LED lights of course I'm sure.... How weird!

hikodaci, 2020-09-01:
> The problem is unlikely to be the Gotek: The LED is basically driven by the host asserting drive-select, there is no smartness on the Gotek side to decide whether to light that LED. Far more likely [...]

So, What I have to do? Should I upgrade my Gotek again with flash floppy from zero?

Yes, I have put the FF.CFG and IMG.CFG files in USB stick with IMG files at the same time. But now Green Led is not light up never. And I'm sure the cable is not broken.

hikodaci, 2020-09-01:
> And green LED lights of course I'm sure.... How weird!

Yes, original GOTEK's led is lighting when the machine read or write.

keirf, 2020-09-01:
Place a jumper horizontally across S0 and JB along lower row (closest to the S0, JB, etc white writing). The green LED should permanently light?

hikodaci, 2020-09-01:
> Place a jumper horizontally across S0 and JB along lower row (closest to the S0, JB, etc white writing). The green LED should permanently light?

I will try tomorrow and inform you. Thank you for everything and your efforts

keirf, 2020-09-01:
Yes come back to it tomorrow. Working out why that green LED no longer lights is your focus. I suppose it could be a bad Gotek or jumper or something. It would be very very rare though

hikodaci, 2020-09-02:
Hi dear,

I fixed the problem, it because of Jumper. My Jumper pin is broken. When I tried to place a jumper horizontally across S0 and JB, the green led flashing and I changed the jumper pin with a new one the green led light permanently. Then I place it S1 and JB my USB stick was operated on it smoothly. Thank you for all your support. You can add the KUKA KRC1 Robot on the list which is working Flash Floppy.

hikodaci, 2020-09-02:
Last question, Now I can not open the USB stick in windows10. What is your suggestion? I need the IMG files created by machine in the USB stick.

keirf, 2020-09-02:
What do you mean you can't open it? Stick not recognised?

keirf, 2020-09-02:
And to be clear: These are blank IMG files that you wrote files to from the KUKA, and now you want to look at these filled IMG files on Windows 10? That should work perfectly fine.

hikodaci, 2020-09-02:
"You need to format the disk in drive E: before you can use it."

hikodaci, 2020-09-02:
> And to be clear: These are blank IMG files that you wrote files to from the KUKA, and now you want to look at these filled IMG files on Windows 10? That should work perfectly fine.

There were 2 IMG files which are created by me on Windows10. One of them full and other one was empty.

I upload and download on the Gotek in Kuka, there was no problem. And when I want to search the files on Windows, It gives this error message about USB stick.

keirf, 2020-09-02:
Somehow the MBR has been lost I suspect. Repartition/reformat as MBR/FAT32 and try again. I don't know how this will have happened, the only reports I get like this are from using the AUTOBOOT file selector on Atari ST and Amiga computers. Of course you are not running an AUTOBOOT file selector.

keirf, 2020-09-02:
In this state I would assume the stick no longer works in the Gotek either?

hikodaci, 2020-09-02:
> In this state I would assume the stick no longer works in the Gotek either?

I will try again with the different USB stick for full steps.

hikodaci, 2020-09-02:
ok. it is working with a new USB stick non problem. I think it was about USB stick error.

closed 2020-09-02 by keirf