keirf/flashfloppy #364: Quickdisk firmware on Sharp MZ-800 issue

ysblokje, 2020-09-03:
I wired everything up according to the wiki and the machine is sure there is a drive there. I can format the drive (but does it?) and ask for a listing of files.

When I try to actually save information to disk I see the LED displays 

-  001
-  0= (both upper and lower bar on the 3rd segment light up for a fraction of a second)
-  0_ (top bar turns off)
-  0 

And repeat.

When it's done saving it comes back with the error message "QD: Unformat err". 
Again if I ask for a listing, no error message appears. 

I would love to see this work on my machine but I have no idea where to even start to debug this. I did enable the logging firmware.

ysblokje, 2020-09-03:
My config 

## FF.CFG: Example FlashFloppy Configuration File

# Place in the root folder or FF/ subfolder of your USB drive.
# NOTE: If FF/ exists, IMG.CFG must reside there, not the root folder.

# Uncommented lines below are the default settings.
# Uncommented options cannot be overridden by settings in other config files.

##
## DRIVE EMULATION

# Floppy-drive interface mode
# shugart: P2=DSKCHG, P34=RDY
# ibmpc: P2=unused, P34=DSKCHG
# ibmpc-hdout: P2=HD_OUT, P34=DSKCHG (not generally needed: prefer 'ibmpc')
# akai-s950: P2=HD_OUT, P34=RDY (Akai S950)
# amiga: P2=DSKCHG, P34=DRIVE_ID (not generally needed: prefer 'shugart')
# jc: JC closed: ibmpc, JC open: shugart
interface = jc

# Host platform: Improves image-format detection for generic types such as IMG
# acorn: Acorn ADFS
# akai: Akai (S01, S20, S950), Korg, SC Prophet 3000
# casio: Casio (FZ-1)
# dec: DEC (RX33, RX50)
# ensoniq: Ensoniq (ASR, TS, etc)
# fluke: Fluke 9100
# gem: General Music (S2, S3)
# ibm-3174: IBM 3174 Establishment Controller
# kaypro: Kaypro
# memotech: Memotech
# msx: MSX
# nascom: Nascom
# pc98: NEC PC-98
# pc-dos: PC DOS Format (geometry determined from Bios Parameter Block)
# tandy-coco: Tandy Color Computer (CoCo)
# ti99: TI-99/4A
# uknc: UKNC / DVK Soviet PDP-11
# unspecified: Common default geometries (including IBM PC)
host = unspecified

# Pins 2 & 34 output (drive->host) manual configuration
# auto: Auto-configure from interface= setting
# nc: Unused/No Connection [eg. if pin is an input (host->drive) on your host]
# low: Always 0 volts (0v)
# high: Always 5 volts (5v)
# rdy: Drive ready (Ready = 0v)
# nrdy: Logical complement of above
# dens: Density mode (High Density = 0v)
# ndens: Logical complement of above
# chg: Disk changed (Changed = 0v)
# nchg: Logical complement of above
# Values: auto, nc, low, high, rdy, nrdy, dens, ndens, chg, nchg
pin02 = auto
pin34 = auto

# Forcibly write-protect images, or respect the FAT read-only attribute?
# Values: yes | no
write-protect = no

# Filter glitches in the SIDE-select signal shorter than N microseconds
# Values: 0 <= N <= 255
side-select-glitch-filter = 0

# Rotational offset of disk after a track change
# instant: No rotation during track change
# realtime: Emulate rotation of disk while track is changing
# Values: instant | realtime
[... 172 lines cut ...]
display-scroll-pause = 2000

# LCD/OLED long filename scroll rate during navigation (ms per update)
# Values: 0 <= N <= 65535
nav-scroll-rate = 80

# LCD/OLED long filename pause before scroll, during navigation (milliseconds)
# Values: 0 <= N <= 65535
nav-scroll-pause = 300

##
## MISCELLANEOUS

# Speaker volume for head STEP
# Values: 0 <= N <= 20
step-volume = 10

# Report the specified version number to host software
# Values: <quoted-string> ("" means report real version)
# eg. da-report-version = "v3.0.0.0"
da-report-version = ""

# Automatically extend certain types of truncated image file (SSD,DSD,TRD)?
# Values: yes | no
extend-image = yes

ysblokje, 2020-09-03:
The created logfile 

** FlashFloppy v3.16 for Gotek
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FlashFloppy

Build: Aug 22 2020 14:26:14
Board: Standard
Config: Flash Slot 0 (ver 2, size 69)
Interface: QuickDisk, JC=On (Roland)
I2C: Invalid bus SCL=1 SDA=0
Display: 3-Digit LED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0951
 PID : 1666
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : Kingston
 Product : DataTraveler 3.0
 Serial Number : 0C9D92C19C3CE420791E0BF8
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 105 items
0:F: 'Blank.qd' 
Cache 105 items
Mode: Native
Current slot: 0/0
Name: 'Blank' Type: qd
Attr: 20 Clus: 0000000c Size: 204800
Fast Seek: 1 frags
Cache 29 items
Write 22016-26111 (4096)... 10969 us
Write 26112-30207 (4096)... 10357 us
Write 27648-31743 (4096)... 9750 us
Write 16384-20479 (4096)... 10653 us
Write 20480-24575 (4096)... 10252 us
Write 16384-20479 (4096)... 12147 us
Write 20480-24575 (4096)... 10222 us
Write 22016-26111 (4096)... 10734 us
Write 26112-30207 (4096)... 10281 us
Write 27648-31743 (4096)... 9759 us
Write 16384-20479 (4096)... 10731 us
Write 20480-24575 (4096)... 10290 us
Write 16384-20479 (4096)... 10763 us
Write 20480-24575 (4096)... 10243 us
Write 22016-26111 (4096)... 10685 us
Write 26112-30207 (4096)... 10277 us
Write 27648-31743 (4096)... 9765 us
Write 16384-20479 (4096)... 12179 us
Write 20480-24575 (4096)... 10252 us
Write 16384-20479 (4096)... 10700 us
Write 20480-24575 (4096)... 10288 us
Write 22016-26111 (4096)... 10613 us
Write 26112-30207 (4096)... 10255 us
Write 27648-31743 (4096)... 9792 us
Write 16384-20479 (4096)... 10665 us
Write 20480-24575 (4096)... 10231 us
Write 16384-20479 (4096)... 10615 us
Write 20480-24575 (4096)... 10210 us
> USBH_USR_DeviceDisconnected
> USBH_USR_DeInit
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0951
 PID : 1666
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : Kingston
 Product : DataTraveler 3.0
 Serial Number : 0C9D92C19C3CE420791E0BF8
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 105 items
0:F: 'disk1.qd' 
Cache 105 items
Mode: Native
Current slot: 0/1
Name: 'disk1' Type: qd
Attr: 20 Clus: 0000000c Size: 204800
Fast Seek: 1 frags
Cache 29 items
Cache 105 items
Before: After: Current slot: 1/1
Name: 'disk2' Type: qd
Attr: 20 Clus: 00000022 Size: 204800

keirf, 2020-09-05:
Let me have a think. I may need to send new logging firmware to see what the write stream timing looks like.

keirf, 2020-09-05:
Here is a new firmware to try which includes more write logging. So please gather another logfile. You can attach it, there's no need to paste it as text into the ticket.

[ff_364_1.zip](https://github.com/keirf/FlashFloppy/files/5177729/ff_364_1.zip)

keirf, 2020-09-05:
You can also attach here the failed QD file after format attempt.

ysblokje, 2020-09-05:
[results.zip](https://github.com/keirf/FlashFloppy/files/5177892/results.zip)

Ok here are my blank.qd / fflog.txt and image_a.cfg . I removed everything else and started fresh with the usb drive empty. 
What I do notice is that the timestamp of the blank.qd hasn't changed after the things I did.

What I did was : 

- QD (get directory)
- QF (format)  No error after format
- QS (save, 0x1200-0x1400 start 0x1200 filename tst)  when it was done it told me unformat error.
- QD no errors and no files.

ysblokje, 2020-09-05:
also here's an image that has only been formatted.

[onlyformatted.zip](https://github.com/keirf/FlashFloppy/files/5177911/onlyformatted.zip)

keirf, 2020-09-05:
Hmm well actually the data written to the QD image looks okay I think. The log file you gathered weirdly has no write info in it at all.... Perhaps you didn't flush the FFLOG.TXT to stick after your format/write tests?

Anyway, putting that aside, what jumpers have you tried? You should have one at MO for sure. Have you tried with and without an additional jumper at JC as well?

ysblokje, 2020-09-05:
Oh... I see that I might have made a mistake with the logging.  I will redo the things I did, but switch image before ejecting the usb stick.
Also I cleaned all the pins today and did some more checks. I did jumper M0 and tried the JC too. With JC jumpered it seems to work better in detecting things. 

[format_proper_log.zip](https://github.com/keirf/FlashFloppy/files/5178026/format_proper_log.zip)

ysblokje, 2020-09-05:
Now I have tried to write something to the image from my previous post and got the unformatted error again.
[failure_writing_file.zip](https://github.com/keirf/FlashFloppy/files/5178034/failure_writing_file.zip)

keirf, 2020-09-05:
Weird. Stuff has definitely got written to the QD image file. The extra logging tells me that the Sharp is using the usual QD bitrate. Maybe it's a timing thing in terms of spin-up time, or size of the 'valid data window', or something.

keirf, 2020-09-05:
Here's a firmware with a new Blank QD image which shrinks the 'data window' to 5.5 sec which I think is more usual for many QD drives. It also shrinks the motor spin-up time to 1 sec (previously 2 secs). So we can see if that helps. No need for logs.

[ff_364_2.zip](https://github.com/keirf/FlashFloppy/files/5178159/ff_364_2.zip)

ysblokje, 2020-09-05:
Alas, no change in behaviour.

There's is one thing else to consider too. This hardware is 30-35 years old at this point. There could be other issues at play.

I have no spare hardware to try this on either.

keirf, 2020-09-05:
No other ideas at the moment. It can't be anything very complex at the quickdisk drive side of things as quickdisk is dumb as a box of rocks. The electronics is very very simple. That was the attraction!

keirf, 2020-09-06:
I'm not sure whether to leave this open or not. I'll leave it for now.

ysblokje, 2020-09-12:
I can tell you that I fixed the belt in my quickdrive and that will in fact read and write disks.

keirf, 2020-09-12:
I'd really need to get my hands on the hardware, or have someone gather a logic analyser trace on the original quickdrive using something like https://www.ebay.co.uk/itm/254626862929 (plus probably logic probe clips to attach to interesting drive signal lines).

Perhaps the drive has a slightly different pin out. Perhaps one of the signals is inverted compared with usual. Must be something minor but significant like this I guess.

keirf, 2020-09-12:
I found the MZ-1F11 service manual. https://original.sharpmz.org/mz-700/download/mz1f11_sm_supplement_gb.pdf

Section 4 (circuit diagram) indicates the usual 10-pin QD connections at connector T-4. Which makes it all the weirder that it doesn't work.

Perhaps we should do some more logging of control signals. RESET and MOTOR are particularly interesting. I will make you a new logging firmware.

keirf, 2020-09-12:
With the previous firmware (ff_364_2) did you both update the firmware and use the included new blank QD image file?

Anyhow, here is another one to try, using the logfile firmware to gather a FFLOG.TXT file. This will log any time the RESET and MOTOR signals change. Which could be interesting.

[ff_364_3.zip](https://github.com/keirf/FlashFloppy/files/5212667/ff_364_3.zip)

ysblokje, 2020-09-12:
> I'd really need to get my hands on the hardware, or have someone gather a logic analyser trace on the original quickdrive using something like https://www.ebay.co.uk/itm/254626862929 (plus probably logic probe clips to attach to interesting drive signal lines).

There's something I want! I will get one of those. I've got plenty of breadboard and related materials to make this into a nice workable setup.

keirf, 2020-09-12:
With ff_643_3 above could you start with a fresh Blank QD, no FFLOG.TXT. Then: format, and then do the small-file write, and gather FFLOG.TXT that results?

One thing I wonder is, FlashFloppy does not start reading again immediately after a write: Reads always restart from the very start, *plus* we wait for the QD to travel to the start position from where the write ended. Which can take several seconds. Possibly real QDs don't behave like this. Most systems seem to tolerate it but perhaps not the Sharp?

ysblokje, 2020-09-15:
I got my little analyzer today, what I will try to do is record with a real drive and the new firmware attached to the controller.
First I have to look up some tutorials on how to use these things though. So don't expect them before the end of the day ;)

keirf, 2020-09-15:
The Saleae software is quite straightforward, fortunately. Sampling at a rate of say 2Msa/s (500ns resolution) should be good enough. You should be able to grab all signals as there's only eight of them.

ysblokje, 2020-09-15:
So here are the first 2 files on real hardware.

[realqd_captures.zip](https://github.com/keirf/FlashFloppy/files/5226746/realqd_captures.zip)

ysblokje, 2020-09-15:
And here are the captures of 364_1 and 364_3 with the FFlogs.

[ffcaps_log.zip](https://github.com/keirf/FlashFloppy/files/5226974/ffcaps_log.zip)

keirf, 2020-09-15:
Fascinating thank you. I think the most obvious difference is that the real QD drive the Read Data remains active most of the time, even during writes (although I'm not sure the Read Data actually looks well formed during a write). But certainly Read Data is active after a write, unlike on FlashFloppy. So probably that is something to fix.

ysblokje, 2020-09-16:
What surprised me was that the "read" continued when the prompt was back. So I am not sure what is going on there.

keirf, 2020-09-16:
That's the Quick drive automatically running on to end of track. It's probably too dumb to do otherwise.

keirf, 2020-09-16:
Okay here's a firmware which restarts reads immediately after end of a write. Let's see if it helps.

[ff_364_4.zip](https://github.com/keirf/FlashFloppy/files/5231536/ff_364_4.zip)

ysblokje, 2020-09-16:
My first test seems to be successful! Very nice indeed!

keirf, 2020-09-16:
If you see some more success I will send you one final tweaked firmware to try. You can then do a clean test with that, and the included Blank QD, and if that's all good I will put in master branch for inclusion in v3.17.

ysblokje, 2020-09-16:
I have just been dumping program after program on the one Image. is there a theoretical maximum? (Not that I am complaining)

keirf, 2020-09-16:
I would say 60kB of storage capacity, approximately.

Let me sort you a final firmware to test...

keirf, 2020-09-16:
Here you go. Not much different that 364_4.zip so should work fine. This is basically the release candidate for this fix.

[ff_364_5.zip](https://github.com/keirf/FlashFloppy/files/5233984/ff_364_5.zip)

ysblokje, 2020-09-16:
I loaded up your latest version. put a BASIC interpreter on a disk, wrote a small program, saved it, removed said program from memory and reloaded it from disk. 
This was all a success.

commit ed96ecd0 (2020-09-16): quickdisk: Restart read stream at the end of a write.
Fixes #364
closed 2020-09-16 by keirf