keirf/flashfloppy #365: Error when writing to ADF from Amiga

dppc, 2020-09-04:
I'm using a Gotek with the 3.15 firmware connected externally as df2 (chained from a real external floppy drive at df1) to an NTSC Amiga 1000 using Kickstart 1.3.

The drive works fine for reading but gets errors when writing to ADF files. Writing to an HFE file works (so I'm able to get around the problem by writing to HFE and then converting the HFE to ADF).

Attached is an example of what I get if I format an ADF file. You can see here (https://imgur.com/SFzyp7e) a comparison of it (on the right) to what a newly formatted AmigaDOS ADF should look like.

Any suggestions on how to get writing to an ADF to work?

[test.zip](https://github.com/keirf/FlashFloppy/files/5175842/test.zip)

keirf, 2020-09-04:
Can you try v2.14?

dppc, 2020-09-04:
Ok, I tried v2.14 and it appears to have the same problems.

keirf, 2020-09-05:
Okay, best return to v3.16 then.

Can't write to ADF but can to HFE is a new one on me. You may need to document exactly how you format the image, and it may make sense to gather logs using logfile firmware (https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive).

Does the problem occur if you don't daisy chain through DF1?

Does the problem occur if you format/overwrite the disk with other tools (eg. XCopy)?

You can also run Amiga Test Kit (https://github.com/keirf/Amiga-Stuff/blob/master/README.md) and run the floppy read/write tests on DF2.

These tools you may have to write to real disk to boot from DF0, but you should be able to DISKCOPY the ADF images from Gotek DF2 to your real drive.

dppc, 2020-09-12:
I haven't had time to try everything you suggested yet, but I installed v3.16 and confirmed that XCopy still gets verification errors when copying to an ADF but works fine copying to an HFE. (Doesn't matter if the source drive is df0 or df1).

The ADF and HFE images I'm attempting to write to are the blanks you provided in your FF_Images collection.

I also ran Amiga Test Kit and ran the floppy read/write tests on df2 (an ADF). Other than the "ready too fast," no errors were reported.

keirf, 2020-09-12:
Any FF.CFG file? Also try holding both buttons for a few seconds after power on and with no USB stick inserted, until the display says rst or reset. This will clear any cached config.

dppc, 2020-09-12:
No FF.CFG file. I did the reset as you suggested, but that doesn't appear to have made any difference. I guess the next step would be to try with the logging firmware?

keirf, 2020-09-12:
Yes I think so. Doing something like the amiga test kit write test and see if you can get that to fail, or an xcopy.

keirf, 2020-09-13:
Another thing is to try another USB stick. What are you using currently?

keirf, 2020-09-19:
Anything more? This is obviously working for most people so there's some weak link here on your setup.

dppc, 2020-09-19:
Well, I tried another USB stick (I was using a 32gb Verbatim branded one, and then tried a 64gb SanDisk), and saw the exact same problem.

I also tried attaching the Gotek to my Amiga 500 instead of my Amiga 1000. Same problem there, too.

Haven't tried the logging firmware yet.

keirf, 2020-09-19:
Just did an "X-Copy Pro '92" DOSCOPY from an ADF in DF1 (containing random garbage) to a blank ADF in DF0. Both drives are FlashFloppy. The copy worked correctly. I then did the same from DF0 to DF1, which also worked.

So you've discounted the Amiga, and the USB stick, and you've reset and Flash-resident config. Could it be the enclosure? I can't really see why copying to HFE would work in that case, however. Oh, and your ADF images are 901120 bytes, right?

Logfile could hopefully be useful, if the logging buffer is large enough that we don't lose too much logging. But we are surely diagnosing some kind of hardware problem here, there are too many Amiga FF users for this to go undiscovered by a wider audience.

dppc, 2020-09-25:
I installed the logging firmware and did the following:

- Booted X-Copy from df0:
- Started a copy from df1: to df2: (df2 is the Gotek)
- On the first track, X-Copy stopped and reported a verification error.
- I had it retry once and it repeated the error message.
- I clicked Stop in X-Copy.
- I switched the disk image and pulled the USB stick.

Logfile is attached.
[FFLOG.TXT](https://github.com/keirf/FlashFloppy/files/5285765/FFLOG.TXT)

keirf, 2020-09-26:
Your writes are taking ridiculous amounts of time to complete, possibly it is because the writes are small (512 byte) and the stick is not optimised for it. Anyway it's impossible to hide the write latency and so X-Copy and other tools will presumably time out. 

These big USB3 sticks are not best. Perhaps I should document that. Small slow sticks are best (eg. old ones, or these days I suppose USB2 16GB, like the cruzer blade 3 pack I can see on Amazon, or a Kingston DataTraveler).

In future I may change write strategy (buffer the writes until track change). This is tracked on #362. But for now go find a "crappy" stick and try again :)

keirf, 2020-09-26:
Ps What you're looking for in FFLOG.TXT is not extreme write times. eg in your log you see snippets like:
```
Write 0/6-10... 4401 us
Write 0/0-4... 4481 us
Write 0/5-5... 1045662 us
Write 0/6-10... 4434 us
Write 0/0-4... 4450 us
```

That 1,045,662 microsecond latency is over 1 second. The drive goes 'silent' for that time after the write. The Amiga probably doesn't like that.

That said, there are no extreme latencies reported until after X-Copy has tried to write the entire track 0 once. It only reports an extreme latency on the second attempt. So I don't know why it is reattempting at all... All sectors were correctly found and written to stick on the first attempt.

keirf, 2020-09-26:
You could even try with your 64GB SanDisk to be honest: Even that shouldn't have the extreme write latencies seen with the Verbatim. I note Verbatim is a particular offender on #362 as well.

You can copy from/to the supplied source/dest ADFs which should make it obvious which sectors were correctly copied. You can download them here: https://drive.google.com/file/d/1xAVtlY5QCgnKBH_l3rjaMnPgK-EmGudX/view?usp=sharing

Please attach FFLOG.TXT again and the destination ADF after the copy attempt.

keirf, 2020-09-26:
Finally if you let me know exact X-Copy version (eg download link) and exact settings you use for copy, I can try it again myself.

dppc, 2020-09-26:
I found an old Kingston DataTraveler 4GB stick and tried that. Same problem...

I'm using X-Copy Pro 5.21 NTSC version. Here's the link: http://jope.fi/xcopy/zip/xcopy_v5.21_usa_master_optimized.zip

For testing with the ADFs you provided, should I stop it after the first verification error? Or have it continue through the whole disk?

keirf, 2020-09-26:
Okay there clearly is something else up if it's across three different sticks. I will have a look at the X-Copy Pro. What exactly do you copy *from* (what's the source disk)?

keirf, 2020-09-26:
You could just go as far as the first verify error by the way. Any more would probably overflow the logging buffer (for FFLOG.TXT) anyway. I'm hoping our best chance here in any case is that I can reproduce this problem myself. Hence wondering about your source disk... though I understand you had problems formatting too? What tool did you use for that?

keirf, 2020-09-27:
Yes! Reproduced the issue. Ok now I can investigate if it can be fixed. It must be something to do with FlashFloppy serving up sectors in pre-defined order, and if XCopy writes them in a different order and doesn't read them back in that order it gets upset. In general this is unavoidable: ADF doesn't record the order sectors were written. But for merely satisfying XCopy copy/format, there are a couple of possible fixes I can try.

keirf, 2020-09-27:
Okay, I think this is fixed. Please try the below firmware (it will report as v3.100, so you can distinguish it from proper releases). You may well find the Verbatim stick still doesn't work because it has such slow response to some small writes. Worth a test to see. But you should see success on your other sticks, especially the Kingston DT 4GB.

Please report back how it goes!

[ff_365_1.zip](https://github.com/keirf/FlashFloppy/files/5288016/ff_365_1.zip)

dppc, 2020-09-27:
Yep, that's fixed it. I was able to successfully copy to ADF with X-Copy, and I formatted an ADF image which also now worked fine.

Thanks very much for your tenacity in hunting this down and getting it resolved!

closed 2020-09-28 by keirf

keirf, 2020-09-28:
Thanks for the testing and persistent bug reporting. Without it bugs don't get fixed!

commit e23c893f (2020-09-28): adf: Remember sector write order for current cylinder if it is modified.

Although we forget as soon as the heads are stepped or disk ejected, this
is sufficient to satisfy X-Copy in verify mode when copying a source disk
with out-of-order sectors (seems rare but does happen).

Fixes #365