keirf/flashfloppy #366: Akai s900

j0vius, 2020-09-08:
Hello
   I'm using a openflops gotec clone https://www.simulant.uk/shop/OpenFlops-USB-Floppy-Disk-Drive-Emulator
 im running flash floppy 3.6  . i, getting a variety of problems ranging from the Akai  displaying oooops drive not mounted , or  i can brows the disk but cant save format or load as it does nothing or hangs the akai 
ive changed the ff config in the host and interface settings . ive tried 3 different usb drives 
under the akai settings for the gotec it does say jumper in the SO but the openflops doesnt have that jumper but ive tried a few different options

has anyone had success with a s900 ? 
thanks

keirf, 2020-09-09:
I don't have an explicit mention of S900 in the wiki (https://github.com/keirf/FlashFloppy/wiki/Host-Platforms#akai) but I think it should straightforwardly work. Try jumper at S0 and following in FF.CFG:
```
host = akai
interface = shugart
pin02 = nc
```

j0vius, 2020-09-09:
Hi thanks for reply , the s900 is basically a s950  with out a couple of
minor differences memory and sampler rate but there basically the same .
settings you've mentioned sort of work then I get error on Akai and it
hangs , out just disk not mounted

I'm thinking the problem could be more the openflops hardwhere than your
software?

j0vius, 2020-09-09:
Hi thanks for reply , the s900 is basically a s950  with out a couple of minor differences memory and sampler rate but there basically the same . settings you've mentioned sort of work then I get error on Akai and it hangs while saving  , or just disk not mounted

keirf, 2020-09-09:
Well, for example, the S900 has a double-density drive, whereas the S950 can read/write high-density. That's quite a difference, floppy-wise!

It certainly means pin 2 on the floppy interface differs as that is double- vs high-density select on the S950. The pin is probably unused on the S900.

It also means that high-density (1600kB) Akai images will not work on the S900.

I did mess with Akai handling a bit during v3.x development. You could try updating to v2.14 from the Downloads page (https://github.com/keirf/FlashFloppy/wiki/Downloads): you can downgrade via the same method as for firmware updates.

It's probably unlikely to be the hardware... but can't state that with certainty.

j0vius, 2020-09-09:
your more of a expert than i am , now you mention it i do remember the disks wern't compatible ithas been 20 years from when i last used this sampler.
down graded to 214 , changed setting in ff .  hangs when saving and wont load os at boot . i think the images are ok they have worked on the od ocation

simulant-systems, 2020-09-09:
Hi just to try to help as j0vius mentioned not having S0 jumper on the OpenFlops board. This is called SEL0 on this drive. You can set SEL0 to either 0 or 1 drive select as OpenFlops supports emulating two disk drives, so make sure it is set on SEL0 drive 0 jumper. This is the default they're sent out with.

As keirf said it is probably more down to your synth model or firmware currently not supporting the S950 than the hardware but we are of course happy to replace or refund any drives that are found to be defective. If needed to you can post an issue here to ask: [](https://github.com/SukkoPera/OpenFlops/issues)

and the OpenFlops designer may also have some insights into your issue too. Let me know if I can also help at all :)

j0vius, 2020-09-09:
Hi
I've tride the 214 still not working ?

keirf, 2020-09-09:
There's someone on the FB group with a S900 who reports it working. He has `pin02 = auto`, not that it probably matters.

Does the OS disk load okay?

j0vius, 2020-09-09:
I'll have another go tomorrow

Yes the is 214 loaded fine I tride with the 214 bootloader as well no
difference

simulant-systems, 2020-09-10:
Hi just a thought that occurred to me, it may be worth also trying the OpenFlops board with the jumper just on M0 drive select 0 position.

The reason it may be worth trying is that I found on Amiga computers Sel0 drive select 0 is fine and for most computers that works great on the default setting, but on Atari ST (which is also Shugart FlashFloppy mode) the disk images start to read and then fail, but moving the jumper to M0 gets them to work great.

Perhaps this Akai synth will be similar and also need the M0 setting instead... I thought it may be worth a try too. If S900's just don't work with FlashFloppy at all then I am really sorry j0vius and happy to refund you for the drive and postage etc. Unfortunately I wrongly advised that it should work due to the Akai s950 working.

As keirf mentions too, there are gotek users on Facebook and forums that I have seen who have had trouble getting the s900 working but have found some fixes, so hopefully there may still be a way to get it working properly.

keirf, 2020-09-10:
Revert to v3.16 and wait and see if we get further info on the FB post. I have tagged another (successful) user on that post.

simulant-systems, 2020-09-16:
Just to post to @j0vius that it may be worth trying a reset like this user did who also uses an OpenFlops board:

https://github.com/keirf/FlashFloppy/issues/369

It fixed his issues, so that may be all you need too :) Thanks again

j0vius, 2020-09-16:
Hello 
  yes i reset flashed the flash Rom every time i load in a new ff config and when down grading to 2.14( i think )

keirf, 2020-09-17:
@j0vius I have sent you confirmed working images and config via Facebook PM.

closed 2020-09-19 by keirf