keirf/flashfloppy #367: Add Yamaha QX5FD MIDI sequencer to supported devices

chriswareham, 2020-09-11:
I have tested FlashFloppy firmware v3.16 on a Yamaha QX5FD MIDI sequencer with a Gotek drive. Formatting, saving and loading all work using a blank HFE file generated from the HxC floppy emulator software. Please add this to the supported devices list on the wiki.

I previously tried a HxC drive in the same machine without success. I will order another drive and test the more sophisticated Yamaha QX3 as well, since this exhibits the same issues with a HxC drive - I would test with the Gotek I already have but the QX5FD is a bit awkward to reassemble!

chrispoacher, 2020-09-12:
Does the QX5FD require an additional adapter or does it connect direct? I’ve heard some horror stories about this and the QX3. J

chriswareham, 2020-09-12:
I had to extend the power cables on both the QX5FD and QX3, but otherwise no modifications - the original 34 pin connectors are standard Shugart. I also modified my QX1 (yes, I like Yamaha sequencers) to have a HxC rev F, and that just needed a cheap floppy adapter for the cables.

keirf, 2020-09-12:
Added to the wiki. Thanks!

closed 2020-09-12 by keirf