keirf/flashfloppy #368: OSD alternatives

github-duran, 2020-09-12:
Hi Keirf, 

Some questions about OSD. Apologies if this is the wrong forum for that, but would appreciate if you have a minute.

I'm about to install FF on a clean Gotek for the first time. I'v tried neither FF or HxC before but I did read a lot about both and realised HxC does not require a Blue Pill where FF seems to do.

- In how far is FF compatible with HxC with regards to OSD?  In particular, can I do without the Blue Pill (as that's extra Hardware I need to put somewhere, eats energy and requires to wait a month on delivery), and connect FF the same as HxC for OSD, with just two wires and a resistor?
https://torlus.com/floppy/forum/viewtopic.php?p=20587#p20587

- If FF is not compatible like that, do you have plans to make it like that? I'm not sure what the design reason was for the Blue Pill. Would be interesting if you can say a few words about that.

- Would a Black Pill (or any other STM32 board) work?

keirf, 2020-09-12:
FF requires the Blue Pill (or equivalent board -- Black Pills can work too). It was an early design decision to put the functionality in a separate MCU: the timing for the OSD text needs to be tight, and that is easier guaranteed on a dedicated CPU and where the pin assignments can be made from scratch to target particular STM peripherals. 

The main downside I think is that it is harder to update the OSD firmware, because it's not part of FlashFloppy proper. That said, it is rarely changed or fixed. Perhaps a full screen mode would be nice to have, as currently it is limited to a max text size of 40x3. But I can't think of much else I would add that users would want to update to.

keirf, 2020-09-12:
Certainly I don't think the extra hardware is a problem per se. The Blue Pill is cheap, and easily physically fitted inside most machines (with care to avoid shorting anything!). Some clones, such as AmiGotek on SellMyRetro, integrates the Blue Pill and Gotek into a single miniaturised board.

closed 2020-09-12 by keirf

github-duran, 2020-09-12:
Thank you for your (Quick!!) reply. I do have a Black Pill lying around and will hook that one up in a couple of days. Might even be a nice test bed for trying to program that micro.
That 40x3 max text size limit you are talking about, is that a gotek firmware limit or is it limited in the OSD (Blue Pill) firmware?

keirf, 2020-09-12:
It's limited in the OSD firmware at the moment. Actually FlashFloppy itself can only do a maximum of 40x4. Both would need modifying to allow, for example, navigation page by page through folders on a USB stick.