keirf/flashfloppy #369: Issues when using Amstrad DSK files and HFE files

ixmeins, 2020-09-15:
Hi, I have a number of openflops drives, I have 3.16 installed but have tried older versions of FF. One unit happily works with both CPC DSK and HFE files whereas the rest will always return read errors from the same DSK files. Can't even CAT them. Is there a difference between the way those two files are parsed to the hardware? Weird issue so I'm clutching at straws.

simulant-systems, 2020-09-15:
Sorry to hear this. Have you tried those drives on any other computer platforms to see if they work ok or whether this is just an issue specific to the firmware with Amstrad CPC?

keirf, 2020-09-15:
Is this using the same USB stick across all the openflops drives that you test?

Try resetting flash config and make sure you don't have any possibly problematic FF.CFG file on your USB stick. Reset flash config: Hold both buttons after power on and with no USB stick inserted, for a few seconds until display shows RST or RESET.

ixmeins, 2020-09-15:
WINNER! The reset configuration worked. I must have just got lucky with the one that worked straight off the soldering iron so to speak. Thanks Keir.

keirf, 2020-09-15:
I've got a feeling a few openflops problems have been solved this way. Perhaps during test these devices are being used with a non-standard FF.CFG? This gets cached in the openflops hardware until the Reset operation. Something to be aware of @simulant-systems ?

simulant-systems, 2020-09-16:
Ah thanks that is a good tip for me to always try with OpenFlops users that buy one of our boards in the future. It looks like @ixmeins built his own drives and we didn't supply these ones but I'll suggest it to anyone else who has a problem. Also it may be worth a try for the Akai S900 user too. I'll post there also. Thanks again :)

keirf, 2020-09-16:
@simulant-systems I can't put my finger on it but yes the need to reset flash config seems to have become more common recently. Perhaps it's not the Simulant Systems builds specifically. As a first thing to try when troubleshooting it seems to be good guidancew though!

closed 2020-09-16 by keirf