keirf/flashfloppy #370: Reading issue with some combinations of USB keys, GEM S2 Synth and firmware >1.2

malkav30, 2020-09-17:
[GEM_img.zip](https://github.com/keirf/FlashFloppy/files/5237710/GEM_img.zip)
Hello,
As exchanged on Facebook, I have a problem on a GEM S2 with USB Keys not working correctly, depending on the version of FlashFloppy that I use:
- using FlashFloppy 1.2 works with every USB Key that I tested (except for Sandisk ultrafit, which is normal)
- using FlashFloppy 2.0a and onwards, when using any other key thant the Memup one, the Gotek behaves correctly, but when the GEM tries to access the image, nothing happens on the gotek side, and the image is reported as "not formatted" by the GEM. when using the Memup, everything works correctly. 
This behaviour is consistent with 2.0a, 2.13, 2.14 and 3.15 firmwares

Keys tested : 
- Sandisk Ultrafit 16gb
- Memup 8gb
- Storex 128mb
- BTC 128mb
- Kingston DataTraveller 32gb
- Emtec 4gb

Attached is the config that I use and an image that is working with the Memup key, regardless of the firmware.

Note : when using 1.2 firmware, Emtec access LED blinks steadily ; when using 2.0a firmware, Emtec LED stops blinking after a few seconds

keirf, 2020-09-17:
Delete the IMG.CFG and try again with latest v3.16. Apologies, the default IMG.CFG file (well actually it's not a default, it's just an example!) captures all IMG files and is giving you the wrong disk geometry. 

The Memup stick must be missing the IMG.CFG or it is not getting loaded, for some reason.

v3.17 is going to change the examples/IMG.CFG file to not have a [default] stanza. As this has bitten a few people!

closed 2020-09-19 by keirf

malkav30, 2020-09-21:
Indeed, that was it. Thanks a lot for the help. It appears the config file was corrupted on my "working key".