keirf/flashfloppy #371: SSD1306 oled display stuck

SamlerFun, 2020-09-19:
I have tried different versions of flashfloppy (v2.14, v3.12a, v3.17). And I tried all the options for setting up FF.CFG. But the display still freezes. Also I tried the option without FF.CFG
GOTEK sfr1m44-u100k bought on ebay
OLED SSD1306 0.91 inch 128x32 IIC I2C bought on ebay
I'm testing with the Roland S550 sampler. GOTEK works, S550 boots, but oled frozen.
What could be the problem?

penfold42, 2020-09-20:
Do you have pull-up resistors on the I2C pins ?

SamlerFun, 2020-09-20:
On the OLEDs pcb I see R2, R3, R4 = 4.7kΩ
They are connected to 3.3V and R3, R4 are connected to SCK and SDA by the second tap.

keirf, 2020-09-20:
The display is probably bad, have you another to try?

keirf, 2020-09-20:
You shouldn't need any FF.CFG to drive that display. You could delete it, and clear any cached config by pressing both buttons after power on, with no stick inserted, for a few seconds until display shows Reset/Rst (if your display works at all ;) ).

SamlerFun, 2020-09-21:
I have two displays and two GOTEKs with the same result.
I tried to make a reset, it didn't help.
Probably have to try other displays.
Many thanks for support!

keirf, 2020-09-21:
Sounds like you'd be confident you got SCK,SDA right way round? Got a link to the displays?

SamlerFun, 2020-09-21:
Yes, mine were bought by these:
https://www.ebay.com/itm/0-91-inch-128x32-IIC-I2C-Blue-OLED-Display-Module-DC3-3V-5V-128-32-Arduino/254325164538?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D20160908105057%26meid%3D6637fb53a46042a1912b949d824d399d%26pid%3D100675%26rk%3D3%26rkt%3D15%26mehot%3Dnone%26sd%3D313222980779%26itm%3D254325164538%26pmt%3D0%26noa%3D1%26pg%3D2380057%26brand%3DUnbranded&_trksid=p2380057.c100675.m4236&_trkparms=pageci%3A983c309f-fc17-11ea-a957-86fa1c53a7fb%7Cparentrq%3Ab116a5fc1740a9e42a466a8bfff3bd5a%7Ciid%3A1

SamlerFun, 2020-09-21:
I have double-checked the correctness of the SCK, SDA connection. It's hard to make a mistake, but when such a situation is, I double-checked it.

keirf, 2020-09-21:
I guess you'll have to try another seller. If they work you know these ones are definitely duff. I have had very few complaints about non working oled displays, problems are most often duff displays assuming display is compatible (which this appears to be).

SamlerFun, 2020-09-21:
Yes, I will try displays from another seller, especially since I see variants in my city.
I have a small report on the S550 sampler, maybe it will be useful. But I'll make it separate.
Keirf, many Thanks for your project!

closed 2020-09-26 by keirf

SamlerFun, 2020-11-10:
Maybe a little late, but ...
"OLED SSD1306 0.91 inch 128x32" purchased in another store - works well!