keirf/flashfloppy #372: "rotary" option - unable to set 2 values

baritonomarchetto, 2020-09-20:
Hello Keirf,

I am using your software in my AKAI S2000 with no major issues. Only thing is, if I define two values for the "rotary" option in my FF.cfg file, it only gets the first value.

In particular, if I set:

`rotary = full, reverse`

or

`rotary = full ,reverse`

the "reverse" option is ignored.

If instead I set

`rotary = reverse`

the reverse option is set correctly.

I suppose something like

`rotary = half, reverse` could still work, being "half" not the default option, but I did not test it.

I don't think it's relevant, but FF.cfg is under "FF" folder in my specific case.

(thanks for the hard work so far!)

keirf, 2020-09-20:
No whitespace between options please, only the "," is allowed. Whitespace is allowed around "=". That's just how it is, at the moment at least. The parser is pretty dumb!

keirf, 2020-09-20:
Ie. set `rotary = full,reverse` exactly like that

baritonomarchetto, 2020-09-20:
Ok, perfect. Please set the notes on FF accordingly: actually there's a space in the description that could lead some user (like me :)) to use the wrong format ;)

closed 2020-09-20 by baritonomarchetto

keirf, 2020-09-20:
I have made it even clearer in the wiki now :)