keirf/flashfloppy #373: Roland S550 corrupted sample data when saving

SamlerFun, 2020-09-21:
With different versions of flashfloppy (v2.14, v3.12a, v3.17). 
I noticed that when saving, one sample is always distorted. The nature of the distortion is a small but strong admixture of clicks (this is audible and visible on the monitor in the waveform viewing mode)
But, this takes place only when saving "Save chain" and "Save set"
When "Save BL" (save block) - everything is fine (checked many times)

keirf, 2020-09-21:
That's a weird bug. I wonder if some blocks get dropped. Have you tried an HFE image file? A double density one like this (reformatted on the Roland) would work: https://github.com/keirf/FF_Images/raw/master/Amiga/Blank_AmigaDOS.hfe

keirf, 2020-09-21:
You can also try downgrading to v1.2. I don't suppose it would make a difference though.

keirf, 2020-09-22:
I'm going to sniff around a possible IMG write issue. Just from eyeballing the code it looks like there's a subtle bug. I could be wrong, but I can probably rewrite that code cleaner and get you to test it.

Meanwhile an HFE test would be very interesting and that would sidestep any potential issue.

keirf, 2020-09-22:
Ok I can't say I've certainly fixed any bugs, but I gave the write code a good clean up and simplification. So you could also test that, with IMG/OUT files. That's assuming you were using IMG or OUT files of course. HFE tests also still interesting.

[ff_373_1.zip](https://github.com/keirf/FlashFloppy/files/5261622/ff_373_1.zip)

SamlerFun, 2020-09-22:
In the next couple of hours I will test HFE and the new ff_373_1

SamlerFun, 2020-09-22:
The saving process went well, but the download from hfe - independently loads three times in a row and writes the disk error. Most of the samples are distorted.
When flashing ff_373_1, the first time saving and reloading went fine, all samples were fine.
But on rechecking (saving and reloading), one sample was corrupted, as well as with flashfloppy-v3.17

SamlerFun, 2020-09-22:
Just in case, I will describe the procedure.
1. S550 is load OS from 000.IMG
2. Load a bank of samples from 001.IMG
Check that all the samples are loaded correctly.
3. Save "Save chain" to 002.IMG
4. Loading "Load chain" from 002.IMG

SamlerFun, 2020-09-22:
"The saving process went well, but the download from hfe....." - with Blank_AmigaDOS.hfe (reformatted in S550)

keirf, 2020-09-22:
OK so seems maybe a timing related thing then. Worse on hfe?

keirf, 2020-09-22:
Try another usb stick?

SamlerFun, 2020-09-22:
I'll try another usb stick tomorrow.
In general, it seems okay to save only as "Save BL".
"Worse on hfe?" - Yes, with Blank_AmigaDOS.hfe (reformatted in S550) loading is not normal, most samples are highly distorted.

keirf, 2020-09-23:
It does seem a bit like #362, except here the write doesn't immediately give an error. So perhaps ideas for ultimately solving the problem there would work here too. Also if it is a similar problem, different USB stick may help. Branded, smaller, "slower" sticks can actually sometimes work better.

SamlerFun, 2020-09-26:
Now I tried another usb stick and everything is fine! (checked twice).
Usb stick which had problems with saving - SANDISK Cruzer Fit 16Gb [sdcz33-016g-g35]
I have several of these and tried three of them, all three have the same result (when saving, one sample is distorted).
Usb stick with which everything is fine - Transcend ultra speed 8gb

And yet, during repeated checks, with SANDISK Cruzer Fit - "Save BL" - one sample is distorted as well as with "Save chain"

SamlerFun, 2020-09-26:
SANDISK Cruzer Fit, I don't even know, maybe it's a bad batch or "gray" goods, they were cheap.

keirf, 2020-09-26:
Did you get them from a generally good retailer such as Amazon (not marketplace)?

SamlerFun, 2020-09-26:
I bought a SanDisk at an electronics store in my city.

keirf, 2020-09-26:
Depends if you trust that store I guess. There are test programs you could subject it to, USB/Flash testers. Or it could simply be a speed thing on FlashFloppy same as #362. Anyway I guess it's resolved to your satisfaction.

closed 2020-09-26 by keirf