keirf/flashfloppy #374: Write and Read Error on Brother Bas311 Asewing machine

hikodaci, 2020-09-24:
I use flashfloppy v316 in my Gotek.
When I operate it on Brother Bas311 sewing machine, it gives the error as code 4. _(Code 4: Floppy disk no inserted, or cable not properly connected.)_

My IMG file is 1440KB and empty.

I have tried many different jumper settings but only S0 and M0 give a response from the machine.

Please check the videos which are recorded by me when I try on the machine.

[https://youtu.be/DSYJ1Z6apbg](url)

keirf, 2020-09-24:
There doesn't seem to be documentation on the disk format used? After all perhaps it is not PC format at all. Or maybe it's not high density.

I would suggest try a double density HFE image, and a high density HFE image, and see if the Brother wants to write a custom format onto them? I attach a couple of blank HFEs below:

[hfe.zip](https://github.com/keirf/FlashFloppy/files/5274457/hfe.zip)

Another thing to try is with and without jumper JC.

hikodaci, 2020-09-24:
How can I use it with HFE image? Have to I change something on config file or anything else?

Will be it a problem to use only put the HFE images into USB stick?

keirf, 2020-09-24:
No need to change config. You can put one HFE image file only on the USB stick, no other files on the stick, and that HFE will automatically be mounted as image 000.

hikodaci, 2020-09-24:
Yesterday, I tried a .hfe file which is enclosed below and it didn't work. Is it same with your suggestion or not? If it is we have to try another way.

[DSKA0000.zip](https://github.com/keirf/FlashFloppy/files/5275596/DSKA0000.zip)

keirf, 2020-09-24:
You tried writing to that HFE and this is the result?

keirf, 2020-09-24:
By the way I am unable to open the zip file you attached here.

hikodaci, 2020-09-24:
Yes I have tried it with Hfe file. This HFE file is working on my other gotek which is updated with hxc. So I'm sure the hfe file is working. I share with you a picture below I want to show you how it looks like.
[image]

keirf, 2020-09-24:
Ok yes that's a high-density HFE image I think. So this works with HxC in this BAS311A?

hikodaci, 2020-09-24:
>  So this works with HxC in this BAS311A?

No it is working with hxc in different machine but I have tried it with flashfloppy in this brother machine.

keirf, 2020-09-24:
Do you know if this machine uses high density or double density disks?

keirf, 2020-09-24:
Even info about the original drive (eg model nbumber).

hikodaci, 2020-09-24:
> Do you know if this machine uses high density or double density disks?

It is 2HD floppy.
and floppy drive is teac fd-235hf-c929

keirf, 2020-09-24:
I believe C929 has straps/jumpers for user configuration (see http://jope.fi/drives/HW_HF_C929_C940_C991_Rev-A.pdf Section 11). Can you take a picture of the 6x2 strap configuration block on the original drive?

hikodaci, 2020-09-24:
> I believe C929 has straps/jumpers for user configuration (see http://jope.fi/drives/HW_HF_C929_C940_C991_Rev-A.pdf Section 11). Can you take a picture of the 6x2 strap configuration block on the original drive?

yes of course

[image]
[image]

keirf, 2020-09-24:
So that would indicate the following in an FF.CFG file at the root of your USB stick:
```
interface=shugart
pin02=nc
```
And jumper at S0 only would be most correct. If a jumper at MO helps, then that's ok. You could also try permanently selecting the drive. You can do this by placing a jumper horizontally across S0 and JB (that's then the only jumper you need).

keirf, 2020-09-24:
Like this:

[image]

keirf, 2020-09-24:
(Ignore the jumper on the programming block at the right hand side!)

keirf, 2020-09-24:
And then continue using your high density (approx 4MB) HFE image file. Don't use an IMG file.

keirf, 2020-09-24:
Another interesting thing would be to save to a real disk using original drive, and then dump that disk (eg using a Greaseweazle device) to SCP or HFE for further inspection.

Another course of action with FlashFloppy is to gather a log file (https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive)

hikodaci, 2020-09-24:
It wrote the file in the .hfe image. But When I try again it gives error. I have to turn off and turn on the machine to write again the file to the USB stick.

hikodaci, 2020-09-24:
I use S0 jumper setting. and horizontal S0 and JB. Then S0 + JA, S0 +JB, S0+JC, S0+J5.

keirf, 2020-09-24:
Does the one file written to the hfe image actually work? Ie can you power cycle and then read it back in okay?

hikodaci, 2020-09-24:
When I turn off and turn on, it is reading and writing only 1 time. When I try one more time to read or write a different file it gives the error.

keirf, 2020-09-24:
Can it read/write any file that first time? Can you send the HFE image after the write? Can you also run logfile firmware and send FFLOG.TXT after a write as described here https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive

hikodaci, 2020-09-24:
[DSKA0000.zip](https://github.com/keirf/FlashFloppy/files/5278696/DSKA0000.zip)
It is the .hfe file.

hikodaci, 2020-09-24:
[FFLOG.TXT](https://github.com/keirf/FlashFloppy/files/5278719/FFLOG.TXT)
And it is the txt file

keirf, 2020-09-24:
That FFLOG.TXT contrains no evidence of writes. Perhaps you pulled the stick without ejecting the image first (which flushes the logging to the usb stick)?

keirf, 2020-09-24:
The DSKA0000.HFE is double density not high density (only 2MB size)? Inspecting it I see empty side 0 and side 1 filled with 16 * 128-byte FM sectors per track. Did you format the HFE on the Brother? Or did you take it from somewhere else?

hikodaci, 2020-09-25:
The log file is including the reading information. First pattern was read and loaded succesfully. But second was failed.

hikodaci, 2020-09-25:
> The DSKA0000.HFE is double density not high density (only 2MB size)? Inspecting it I see empty side 0 and side 1 filled with 16 * 128-byte FM sectors per track. Did you format the HFE on the Brother? Or did you take it from somewhere else?

This HFE file is coming from Hxc which is successfully worked on BAS311A. I wrote a pattern on it as successfully. Then turn off the machine and turn on again then I wrote another pattern on it as successfully. When I read the first pattern from the USB stick, it was Ok. Then tried to read the second pattern the machine was stopped and gave an error. But If I turn off and turn on then read again for the second pattern, it was successfully read.

keirf, 2020-09-25:
Ok that being the case, the disk format is 16 * 128-byte sectors, 80 cylinder, single sided.

So here is an IMG file and IMG.CFG file you can put on your USB stick. Remove all other HFE and IMG files (keep only your FF.CFG file!). The included `brother.img` file will then automatically mount as image 000. The included IMG.CFG file will ensure that the image is converted into the correct track format. IMG has generally been more reliable than HFE on flaky setups.

[brother_img.zip](https://github.com/keirf/FlashFloppy/files/5281202/brother_img.zip)

hikodaci, 2020-09-25:
Which jumper setting should I use on it?

keirf, 2020-09-25:
Same as you used on HxC I guess. S0 should work, but whatever you got working best.

keirf, 2020-09-28:
Did it work?

hikodaci, 2020-09-28:
> Did it work?

I haven't tried yet.

keirf, 2020-10-11:
Closed pending further info. Please reopen if issues continue.

closed 2020-10-11 by keirf