keirf/flashfloppy #376: disk change signal

sub205, 2020-09-30:
Regarding my ticket from 2019 (not working on newer china goteks #250 ) i found the root cause of the issue:

On my older drives, the disk change stayed high all time, on the newer drives (where flashfloppy does NOT work) it's getting pulled down while trying to read the image, resulting in an error.

When manually pulling the signal high, everything works fine.

Target is a regular IBM PC interface (generic 386, generic floppy controller).

Is there any configuration to do for this to work?
I was a bit puzzled because the older drives worked out of the box.

sub205, 2020-09-30:
follow up:

User to stupid error solved by closing JC.
Still not sure why it works on my older drives without this jumper...

closed 2020-09-30 by sub205

keirf, 2020-09-30:
Maybe you'd configured the older drives differently?

sub205, 2020-09-30:
No. Same exact config, i also compared the PCBs.
Additionally, for DD images to work, i had to create a FF.CFG:

interface = ibmpc
host = pc-dos

Otherwise, i was unable to boot DR-DOS 720k images (which also happens when trying to write them to 1.44M floppys!)

sub205, 2020-09-30:
Hm, now disk change isn't detected.
Will dig deeper into this the next days with my logic analyzer ...

keirf, 2020-09-30:
Mysterious. You can clear any possibly cached config in the old Goteks by pressing both buttons after power on, with no stick inserted, for a few seconds until the display changes. Then you can be sure you are on a level config playing field old vs new. Also running same up to date firmware on old Goteks?

If you are confident dskchg is on the correct pin (34?) then bear in mind the default is to assert it until a new image is mounted and the disk head is stepped. Pretty standard floppy interface. But some hosts work better with chgrst=delay-3 in FF.CFG. This automatically deasserts dskchg 1.5s after an image is mounted.

sub205, 2020-10-08:
ok, i was able to fix the last bug. Disk-Change-Signal wasn't getting trough so i was stuck with the first image the system loaded.
Turned out it was a defect mosfet (Q4) on the PCB.