keirf/flashfloppy #378: Mixed disk density

mikewbrantley, 2020-10-06:
Add new options to FlashFloppy img.cfg to support single density on track zero and double density on other tracks.  
This should provide a separate configuration for single and double density.  In other words, track zero would have configuration options and tracks 1 upward would have it's own configuration options.

mikewbrantley, 2020-10-06:
[hfe files.zip](https://github.com/keirf/FlashFloppy/files/5335177/hfe.files.zip)
These are the 3 files I emailed you.

keirf, 2020-10-06:
The FLEX formatted HFE has 10*256b sectors per track, cylinder 0, FM recording. Sectors are numbered 1-20 (across both sides) and have interleave 4:1, and that extends across to side 1 (sector id 11 is the second sector on side 1).

Cylinders > 0: 18*256b sectors per track, MFM recording. Sectors are numbered 1-36 (across both sides) and have interleave 6:1, and that extends across to side 1 (sector id 19 is the sixth sector on side 1).

The DSK looks to be all MFM, as it is 737,280 bytes. I would expect a DSK with FM track0 to be 733,184 bytes. Also intyeresting to note that some FLEX docs mention that track0 has no sector1 and has a sector0 instead. That isn't the case in the HFE image here! Perhaps there is variation in FLEX disk format across different systems running FLEX?

mikewbrantley, 2020-10-06:
The physical disk that is formatted with Flex has Track 0 starting with sector 1 and goes through sector 10 on side one.  Side two starts with sector 11 and goes through sector 20.  This is FM recording.  Tracks 1 through 79 are recorded with MFM and has 18 sectors per side. 

I will get in touch with Mike Evenson and ask about the dsk file formatting he does with the emulator tool.  Now that I think about it, the only time I ever use this tool is to build a dsk file for storage and to extract files from it.  The flashfloppy emulator is the first time I have ever tried extracting these files from anything other than Mike's utility.  I'm sorry for making this so confusing by expecting the emulator tool to perform just as the Flex formatter.

keirf, 2020-10-06:
Here is another DSK sent by @mikewbrantley. This one is formatted with FM track 0. Note that it is still 720k. Cylinder 0 is padded out to 36*256-byte size by 16 unused sectors (file offsets 0x1400-0x2400). Note that the space allocation and padding seems to be done per cylinder, not per track. ie. the DSK file logically comprises 80 9kB cylinders, rather than 160 4.5kB tracks.

This ought to be somehow made configurable in IMG.CFG...

[BB184.zip](https://github.com/keirf/FlashFloppy/files/5336724/BB184.zip)

keirf, 2020-10-06:
For our information, roughly speaking the IMG.CFG geometry should be something like the following. Actually the image and its geometry can't yet be precisely described. I will edit the following as I extend the IMG.CFG options.
```
[default]
cyls = 80
heads = 2
bps = 256
secs = 18
mode = mfm
interleave = 6
id = 1:19
tracks = 0.*
    secs = 10
    mode = fm
    interleave = 4
    id = 1:11
```

drdpj, 2020-10-06:
Here you go!

[480ZBOOT.IMD.zip](https://github.com/keirf/FlashFloppy/files/5337035/480ZBOOT.IMD.zip)

mikewbrantley, 2020-10-06:
Looks good.  It will give us something to experiment with.

From: Keir Fraser 
Sent: Tuesday, October 6, 2020 5:13 PM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

For our information, roughly speaking the IMG.CFG geometry should be something like the following. Actually the image and its geometry can't yet be precisely described. I will edit the following as I extend the IMG.CFG options.

[default]
cyls = 80
heads = 2
bps = 256
secs = 18
mode = mfm
interleave = 6
id = 1:19
tracks = 0.*
    secs = 10
    mode = fm
    interleave = 4
    id = 1:11

mikewbrantley, 2020-10-08:
Michael Evenson developed a program to create a dsk image with 10 sectors per side on track 0 and 18 per side on tracks 1 through 79.  I am attaching the file I created using this software.  Does it look correct now?

[TEST.zip](https://github.com/keirf/FlashFloppy/files/5351269/TEST.zip)

keirf, 2020-10-08:
Hi yes it does. However an earlier DSK you sent also looked formatted correctly: BB184.dsk. But that one had track 0 padded out to same size as other tracks (0x2400 bytes). Whereas this new TEST.DSK has track 0 taking only 0x1400 bytes. So.... to pad or not to pad? Does it matter to FLEX utilities either way (can any of them handle the mixed density)?

mikewbrantley, 2020-10-08:
The software that runs on the PC and creates the dsk file can handle either padded or not but the actual computer that will be accessing flashfloppy will not work with the padded track zero.  It expects track zero to be 10 sectors per side.  I have tried using a padded dsk file and the computer fails with a disk read error.  I will play around with it some more but so far it does not like the padded track.  Of course, the padding I did was on a floppy with double density beyond track zero which is not supported presently.  I will build a single density floppy with padded sectors on track zero and see how that works.

From: Keir Fraser 
Sent: Thursday, October 8, 2020 5:11 PM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

Hi yes it does. However an earlier DSK you sent also looked formatted correctly: BB184.dsk. But that one had track 0 padded out to same size as other tracks (0x2400 bytes). Whereas this new TEST.DSK has track 0 taking only 0x1400 bytes. So.... to pad or not to pad? Does it matter to FLEX utilities either way (can any of them handle the mixed density)?

keirf, 2020-10-08:
The DSK files will not yet work on FlashFloppy, padded or not. The question was whether FLEX tools cared. Apparently not. In which case no padding will be easier to get working with FlashFloppy probably.

keirf, 2020-10-09:
This in progress on branch `imgcfg`. Done some code rearranging, mainly now need to add the new `tracks=` option and also allow interleave and sector numbering to be continuous across sides of the disk.

mikewbrantley, 2020-10-09:
Thanks for your support and generosity in working on this.  If changing config for dsk type images, we could name this configuration something else like gti.

keirf, 2020-10-09:
Okay, here is a new firmware for you to try. This should work with images of the unpadded form. You can use the IMG.CFG file in the examples/ subfolder: Just copy it to the root of your USB stick. In this case your flex image must have a filename ending in `.flex.img` or `.flex.dsk`, in order to match the `[flex]` tag defined in IMG.CFG. Note the firmware version number is bumped to v3.100 so you can tell you're running the correct modified version.

Let me know how it goes!

[ff_378_1.zip](https://github.com/keirf/FlashFloppy/files/5355809/ff_378_1.zip)

mikewbrantley, 2020-10-09:
Thanks Keir, I am away from house right now.  Should be back in couple of hours.

mikewbrantley, 2020-10-09:
I installed firmware which shows v3.100 on my oled.  I placed two files formatted with no pads on track 0, just 20 sectors.  I named one of these TEST.FLEX.DSK and the other TEST2.FLEX.IMG.  I also placed a BLANK.DSK file formatted for double sided, single density and 80 tracks with no pads.  The two files ending in FLEX.DSK and FLEX.IMG both give an ERR 31.  The one file named BLANK.DSK does not give an error but when I try to format it, I get a disk read error afterwards.  I am sure this is because the img.cfg file does not contain configuration for that file. Attached is the IMG.CFG I am using.

From: Keir Fraser 
Sent: Friday, October 9, 2020 11:05 AM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

Okay, here is a new firmware for you to try. This should work with images of the unpadded form. You can use the IMG.CFG file in the examples/ subfolder: Just copy it to the root of your USB stick. In this case your flex image must have a filename ending in .flex.img or .flex.dsk, in order to match the [flex] tag defined in IMG.CFG. Note the firmware version number is bumped to v3.100 so you can tell you're running the correct modified version.

Let me know how it goes!

ff_378_1.zip

keirf, 2020-10-09:
You didn't attach your IMG.CFG but if it's the one I included in the firmware zip file, the problem is that the tag match is case sensitive, so `[flex]` won't match `TEST.FLEX.DSK`. If you want to use ALL CAPS you will need to edit IMG.CFG and change the line `[flex]` to `[FLEX]`.

keirf, 2020-10-09:
For bonus points see if you can make your own tagged config in IMG.CFG to handle your single-density image (eg. called BLANK.FLEXSS.DSK). As a hint, all cylinders of `[FLEXSS]` will have the same configuration as `[FLEX]`'s cylinder 0.

mikewbrantley, 2020-10-09:
Okay.  I did attach zip file.  Guess I should put it on github next time.  I will make changes to img.cfg and try again.

From: Keir Fraser 
Sent: Friday, October 9, 2020 2:51 PM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

You didn't attach your IMG.CFG but if it's the one I included in the firmware zip file, the problem is that the tag match is case sensitive, so [flex] won't match TEST.FLEX.DSK. If you want to use ALL CAPS you will need to edit IMG.CFG and change the line [flex] to [FLEX].

keirf, 2020-10-09:
Possibly the tag match should be case insensitive. After all FAT is not case sensitive. I'll have a think about that for the next build.

mikewbrantley, 2020-10-09:
I will try that in a few minutes.  I am now running a format on the test.flex.dsk and write went good.  Verify is going good also so far.

From: Keir Fraser 
Sent: Friday, October 9, 2020 2:53 PM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

For bonus points see if you can make your own tagged config in IMG.CFG to handle your single-density image (eg. called BLANK.FLEXSS.DSK). As a hint, all cylinders of [FLEXSS] will have the same configuration as [FLEX]'s cylinder 0.

mikewbrantley, 2020-10-09:
Format finished good.  I am now copying real floppy drive contents to flashfloppy.  So far, good.  Will work on IMG.CFG for BLANK.FLEXSS.DSK.

From: Mike Brantley 
Sent: Friday, October 9, 2020 3:08 PM
To: keirf/FlashFloppy 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

I will try that in a few minutes.  I am now running a format on the test.flex.dsk and write went good.  Verify is going good also so far.

From: Keir Fraser 
Sent: Friday, October 9, 2020 2:53 PM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

For bonus points see if you can make your own tagged config in IMG.CFG to handle your single-density image (eg. called BLANK.FLEXSS.DSK). As a hint, all cylinders of [FLEXSS] will have the same configuration as [FLEX]'s cylinder 0.

mikewbrantley, 2020-10-09:
Copy of complete real floppy to flashfloppy worked great.  I created a tagged config [FLEXSS] in the IMG.CFG and formatted the BLANK.FLEXSS.DSK with no problems.  I am copying all files from real floppy disk to flashfloppy now.  I am including my IMG.CFG zip file with this note.  By the way, BLANK.FLEXXSS.DSK is configured as 80 track, double density.  Thanks for everything Keir.  I don't know how to thank you but just know I appreciate everything you have done.  If I can ever help you in any way, please call on me.
[IMG.zip](https://github.com/keirf/FlashFloppy/files/5357197/IMG.zip)

keirf, 2020-10-09:
Ah looks great! I will be further enhancing this, probably over this weekend:
1. I will make the tags case insensitive
2. I will allow to tag by filesize in addition to (or instead of) nametag. This will allow you to use untagged image names, and the correct config picked out of IMG.CFG by filesize alone, if you want.
3. I will add example configs under examples/host/<host>/IMG.CFG. So in this case perhaps `examples/host/TSC_Flex/IMG.CFG` and we can add all the necessary configs for various image types/sizes there.

mikewbrantley, 2020-10-09:
Sounds great.

I will add example configs under examples/host//IMG.CFG. So in this case perhaps examples/host/TSC_Flex/IMG.CFG and we can add all the necessary configs for various image types/sizes there.

keirf, 2020-10-10:
I'll be back in touch in a while with another firmware to try!

keirf, 2020-10-10:
Okay, here it is. Version is bumped to v3.101.
1. TSC Flex now has an IMG.CFG included in the distribution: Simply copy examples/Host/TSC_Flex/IMG.CFG to the root of your USB stick. This should match your DSDD and SSDD images. Let me know if you want any more formats added.
2. You will see that matching is now by filesize rather than filename. You do not need to tag your image names with `flex` any longer.

[ff_378_2.zip](https://github.com/keirf/FlashFloppy/files/5358824/ff_378_2.zip)

mikewbrantley, 2020-10-10:
That worked great.  I even did one for 40 track, DSDD and it worked too.  Thank you again for all the time and effort you have spent on this project.  There will be many people in our Flex User Group thankful for this.

From: Keir Fraser 
Sent: Saturday, October 10, 2020 6:43 AM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

Okay, here it is. Version is bumped to v3.101.

  1.. TSC Flex now has an IMG.CFG included in the distribution: Simply copy examples/Host/TSC_Flex/IMG.CFG to the root of your USB stick. This should match your DSDD and SSDD images. Let me know if you want any more formats added. 
  2.. You will see that matching is now by filesize rather than filename. You do not need to tag your image names with flex any longer. 
ff_378_2.zip

keirf, 2020-10-10:
Okay, all done then! Any formats you'd like to add to the distributed TSC_Flex/IMG.CFG, just let me know. I'm happy to add them, especially if you've tested them and supply them for me simply to append to the file.

Likewise, if you need any extensions to the format (eg to deal with image-file padding) let me know and I'll consider it.

This support as it is will be in the next release (v3.20).

keirf, 2020-10-10:
By the way, here is the latest. Further code tweaks and I've also modified the IMG.CFG so the track format is closer to the HFE you provided earlier in this ticket. Please feel free to test (and extend IMG.CFG) at your convenience. Should work just the same as the previous, hopefully!

[ff_378_3.zip](https://github.com/keirf/FlashFloppy/files/5359590/ff_378_3.zip)

mikewbrantley, 2020-10-10:
This unzipped to v3.19.  Is that correct?  There is already a v3.19.

From: Keir Fraser 
Sent: Saturday, October 10, 2020 1:51 PM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

By the way, here is the latest. Further code tweaks and I've also modified the IMG.CFG so the track format is closer to the HFE you provided earlier in this ticket. Please feel free to test (and extend IMG.CFG) at your convenience. Should work just the same as the previous, hopefully!

ff_378_3.zip

keirf, 2020-10-10:
Yes I didn't bother to hack the version number on this one. Rest assured it is not the released v3.19

mikewbrantley, 2020-10-10:
Okay.  I will check it out and get back to you.

From: Keir Fraser 
Sent: Saturday, October 10, 2020 2:19 PM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

Yes I didn't bother to hack the version number on this one. Rest assured it is not the released v3.19

mikewbrantley, 2020-10-10:
I updated firmware and copied the IMG.CFG but I now get a 31 error.  I put the old IMG.CFG file back on the USB and it works okay.  Any ideas?

From: Keir Fraser 
Sent: Saturday, October 10, 2020 2:19 PM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

Yes I didn't bother to hack the version number on this one. Rest assured it is not the released v3.19

mikewbrantley, 2020-10-10:
I noticed on the new IMG.CFG, the tag has flex in it now......[flex::733184].   Old one did not have flex; just file size.

From: Mike Brantley 
Sent: Saturday, October 10, 2020 2:48 PM
To: keirf/FlashFloppy 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

I updated firmware and copied the IMG.CFG but I now get a 31 error.  I put the old IMG.CFG file back on the USB and it works okay.  Any ideas?

From: Keir Fraser 
Sent: Saturday, October 10, 2020 2:19 PM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

Yes I didn't bother to hack the version number on this one. Rest assured it is not the released v3.19

keirf, 2020-10-10:
You didn't copy the correct IMG.CFG. You want the one under `examples/Host/TSC_Flex`.

mikewbrantley, 2020-10-10:
You are correct.  Sorry, I just picked up the first one I saw.  Copied the right one all works great.  Thanks.

From: Keir Fraser 
Sent: Saturday, October 10, 2020 4:06 PM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

You didn't copy the correct IMG.CFG. You want the one under examples/Host/TSC_Flex.

mikewbrantley, 2020-10-13:
Hey Keir, I have run into a problem with Flex formatted disk on FlashFloppy when done by different hardware.  I asked someone to check out the FlashFloppy firmware on their system.  They are using SWTPC computer but it has a different Floppy Disk Controller and runs at 1 Mhz instead of 2 Mhz like mine.  I did slow my system to 1 Mhz and it still worked on FlashFloppy fine.  They can format an 80 track, ds, dd floppy on their physical drive but when trying to do the same on the FlashFloppy system, It fails with a disk read error on the verify.  Do you think the IMG.CFG file needs different settings?  Would you like for me to reopen #378 issue or create another one or just let you take a look?  If you want me to send you an HFE dump of his formatted floppy, let me know?  Thanks,
Mike

keirf, 2020-10-14:
The problem is probably write speed to the USB stick amazingly enough.

Perhaps confirm with the logfile firmware (https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive) or try another USB stick.

Some sticks are fussier about small writes than others. And especially because Flex format has sector interleave, so the small blocks aren't even written in-order into the IMG file. The best solution would be to buffer the track's writes and write out on head movements. This isn't implemented yet.

keirf, 2020-10-14:
The other thing to confirm is that this other machine's Flex really is exactly same format as yours. To do this we'd need to dump of one of his real formatted disks. Eg. using Greaseweazle. Then I could inspect and confirm one way or another.

Or he could do what you did and format an HFE image file? That would be okay too!

mikewbrantley, 2020-10-14:
I will check it out and get back to you.

mikewbrantley, 2020-10-14:
Okay.  By the way, I have his machine here with me.  I have tried several versions of Flex on his machine and I get the same results.  The format on the FlashFloppy writes through all 80 tracks and then returns to track 0 and tries to write something and fails on sector B and then sector C and then I stop it by resetting the machine, change FlashFloppy to a different image and then dump the log file that I am including as a zip.  I am also including a zip file containing an HFE image file of a good formatted disk from his machine.  I did change to a different USB stick and got same results.  Thanks for help.

From: Keir Fraser 
Sent: Wednesday, October 14, 2020 6:54 AM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

The other thing to confirm is that this other machine's Flex really is exactly same format as yours. To do this we'd need to dump of one of his real formatted disks. Eg. using Greaseweazle. Then I could inspect and confirm one way or another.

Or he could do what you did and format an HFE image file? That would be okay too!

mikewbrantley, 2020-10-14:
Sometimes, gmail blocks attachments so am sending this from different account.

keirf, 2020-10-14:
The attachments probably got stripped off by Github. It is easiest to log into github and attach a file in their issue tracker directly in their website.

mikewbrantley, 2020-10-14:
Okay.  By the way, I have his machine here with me.  I have tried several versions of Flex on his machine and I get the same results.  The format on the FlashFloppy writes through all 80 tracks and then returns to track 0 and tries to write something and fails on sector B and then sector C and then I stop it by resetting the machine, change FlashFloppy to a different image and then dump the log file that I am including as a zip.  I am also including a zip file containing an HFE image file of a good formatted disk from his machine.  I did change to a different USB stick and got same results.  Thanks for help.
[FFLOG.zip](https://github.com/keirf/FlashFloppy/files/5380125/FFLOG.zip)
[Floppy HFE formatted with diff FDC.zip](https://github.com/keirf/FlashFloppy/files/5380127/Floppy.HFE.formatted.with.diff.FDC.zip)

keirf, 2020-10-14:
Okay, there are some format differences. The order of sectors on cylinder 0 is different, which probably doesn't matter. All the other tracks are only 17 sectors, rather than 18: sector 1 is missing on head 0, and sector 22(!) is missing on head 1. But again I'm kind of surprised if it can't deal with extra sectors magically appearing on the track. It would probably just ignore them!

What I hadn't spotted before is that both heads of cylinder 0 have all sectors marked as "head 0". And both heads of all other cylinders have all sectors marked "head 1". Obviously your FDC doesn't care, as I don't replicate this and your system didn't mind. But perhaps his does.

I will see about making a patch to replicate this behaviour.

keirf, 2020-10-14:
Okay, try this. You will also need the IMG.CFG from `examples/Host/TSC_Flex/`, as that has been modified to force the head number.

[ff_378_4.zip](https://github.com/keirf/FlashFloppy/files/5380396/ff_378_4.zip)

mikewbrantley, 2020-10-14:
I’m sorry I did not catch that earlier.  Thanks for help.

From: Keir Fraser 
Sent: Wednesday, October 14, 2020 2:46 PM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

Okay, there are some format differences. The order of sectors on cylinder 0 is different, which probably doesn't matter. All the other tracks are only 17 sectors, rather than 18: sector 1 is missing on head 0, and sector 22(!) is missing on head 1. But again I'm kind of surprised if it can't deal with extra sectors magically appearing on the track. It would probably just ignore them!

What I hadn't spotted before is that both heads of cylinder 0 have all sectors marked as "head 0". And both heads of all other cylinders have all sectors marked "head 1". Obviously your FDC doesn't care, as I don't replicate this and your system didn't mind. But perhaps his does.

I will see about making a patch to replicate this behaviour.

keirf, 2020-10-14:
No problem, you now have a patched version, see above ^ :)

mikewbrantley, 2020-10-14:
Great!  I am away from house right now.  Be back in about an hour.  Will get right back to you.  Thanks again.

mikewbrantley, 2020-10-15:
Sorry, it took so long to get back to you.  This friends computer stopped working and I had to work on it.  I am sure you are sleeping by now but maybe you can get this in the morning.  The DSDD 80 track worked great.  It formatted and returned to track 0 and verified all 80 tracks and then wrote the boot loader.  Looks good.  I then tried the SSDD 80 track format.  It failed as before...formatted through track 79 and returned to track 0 and failed at sector B and beyond.  Thanks again for working on this.

keirf, 2020-10-15:
Thanks. Try this one. I have modified the firmware, and also you need the new `examples/Host/TSC_Flex/IMG.CFG' too.

[ff_378_5.zip](https://github.com/keirf/FlashFloppy/files/5382900/ff_378_5.zip)

Are you sure by the way that the computer was correctly formatting as single sided? Just a thought. As there'd be no sector B on cylinder 0 (sector B is on the other side!).

Also you'll want to verify these changes on your machine, so we know we're not breaking what we already got working.

mikewbrantley, 2020-10-15:
That worked good for both ssdd and dsdd.  I will get my computer back out and test changes on it.  Will get back a little later.  Thanks again for everything.

From: Keir Fraser 
Sent: Thursday, October 15, 2020 2:47 AM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

Thanks. Try this one. I have modified the firmware, and also you need the new `examples/Host/TSC_Flex/IMG.CFG' too.

ff_378_5.zip

Are you sure by the way that the computer was correctly formatting as single sided? Just a thought. As there'd be no sector B on cylinder 0 (sector B is on the other side!).

Also you'll want to verify these changes on your machine, so we know we're not breaking what we already got working.

mikewbrantley, 2020-10-15:
Excellent work!  It still works on my machine also.  I am really impressed with your skills.  Thank you so much for all you time and effort in making FlashFloppy work with TSC Flex software and different hardware.  Please call on me if I can ever be any help to you.
Mike

From: Mike Brantley 
Sent: Thursday, October 15, 2020 6:35 AM
To: keirf/FlashFloppy ; keirf/FlashFloppy 
Cc: Mention ; Brantley Mike 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

That worked good for both ssdd and dsdd.  I will get my computer back out and test changes on it.  Will get back a little later.  Thanks again for everything.

From: Keir Fraser 
Sent: Thursday, October 15, 2020 2:47 AM
To: keirf/FlashFloppy 
Cc: mikewbrantley ; Mention 
Subject: Re: [keirf/FlashFloppy] Per-track format configuration in IMG.CFG (#378)

Thanks. Try this one. I have modified the firmware, and also you need the new `examples/Host/TSC_Flex/IMG.CFG' too.

ff_378_5.zip

Are you sure by the way that the computer was correctly formatting as single sided? Just a thought. As there'd be no sector B on cylinder 0 (sector B is on the other side!).

Also you'll want to verify these changes on your machine, so we know we're not breaking what we already got working.

keirf, 2020-10-15:
That's great, and thanks for your testing! I am going to do a v3.20 release now.

closed 2020-10-15 by keirf