keirf/flashfloppy #379: [Enhancement] Two button navigation emulation of Gotek defaults

josederas, 2020-10-09:
Apologies if I somewhat missed this.  But for those of us that don't want the rotary encoder, it would be beneficial to emulate the two button behaviour of Gotek:  Button 1, for single digits, Button 2 for Tens, and Button 1+2 for Hundreds.  

Looking at the FF.CFG it doesn't appear that is possible today?

keirf, 2020-10-09:
No. It does accelerate if you hold a button, though?

keirf, 2020-10-09:
How does it work anyway? Does image navigation always start from 000 and then you select the image number from scratch with buttons?

josederas, 2020-10-09:
It will remember the last disk.  But for example, assuming you are at 000, and you need disk 348:

1. Press Button 1+2 ...... 3 times
2. Press Button 2 ......... 4 times
3. Press Button 1 ......... 8 times

is that better than accelerating buttons 1 and 2 to reach that?  I don't know... Thoughts?

keirf, 2020-10-10:
Okay, here's a firmware to try. You need `twobutton-action = htu` in FF.CFG.
There are some behjaviours here which may need to be tweaked:
1. Should left and right buttons be swapped? Currently the left button does Units, right does Tens. That's opposite way round to the digits affected on the display. What do you think?
2. When a digit wraps 9->0, it currently carries into the next column (eg. 179 -> 180 not 170). Is that good?
3. When wrapping past the maximum available image, I simply reset the count to 000. Perhaps just the column being adjusted should be reset to 0? Not sure here. This does make it easier to get back to 000...

[ff_379_1.zip](https://github.com/keirf/FlashFloppy/files/5358670/ff_379_1.zip)

josederas, 2020-10-10:
Thank you!!!  I've tried it and works.  To your questions:

1. Yes, L/R buttons should be swapped, it's more natural.  I think it was my mistake in identifying the buttons as 1 and 2 above.  
2. No, I think it should just rollover.  I'm not sure what Gotek does though, but I would think that rolling over is better.
3. I think is dependent on behavior of 2.  If 2 increments the next column then I agree at the end it should go to 000.  But otherwise it wraps.

I think item 3 is the most problematic... how to get to 000 easily if you have lots of images.  As I only use dozens I have not come across this issue.

keirf, 2020-10-10:
I will post a revised version, probably tomorrow rather than tonight.

keirf, 2020-10-10:
Here is the revised firmware.
1. L/R buttons are now swapped.
2. Digits now roll over, no carry into next column.
3. On overflow, the incremented digit is reset to 0. To reset all digits to 0, hold both buttons for 1sec (display changes immediately after 1sec).

See what you think.

[ff_379_2.zip](https://github.com/keirf/FlashFloppy/files/5359939/ff_379_2.zip)

josederas, 2020-10-11:
Sorry for the delay, tested and works perfect.  I think the 1-sec. return to 000 works well, good idea.  This will be my default configuration, as I don't want the encoder in my particular deployment.

keirf, 2020-10-15:
Released in v3.20

closed 2020-10-15 by keirf