keirf/flashfloppy #380: [Enhancement] Renaming files inside Gotek drive with encoder - Eject Menu

ArtyGZ, 2020-10-14:
Hi! First time suggesting something on Github. Love the project, it is amazing what has been done so far!

So here is my question: Is it a realistic suggestion to add a rename function inside the eject menu?
I do realize that this is pretty much impossible for users who do not have a rotary encoder, or at least very impractical.  

Also this question was asked a while back, but now that we have the Eject-menu and after reading the code (very quickly I admit), I think it's definitely feasable :)

Here's how I imagine this:

1 - You select a file and get to the eject menu 
2 - Get to the rename function and select it 
3 - You now have a cursor displayed on the first character of the current name
4 - You can change the selected character by turning the encoder 
5 - You can move the cursor with the two tactile switches 
6 - Once you are satisfied with the changes, you click the rotary encoder and the changes are saved.

Basically that's how renaming works on some old Akai samplers.

This would be super useful for users of old samplers for simply naming files or organizing sounds, sequences and projects.
Removing the USB drive and getting to a computer etc... is sometimes a bit annoying. Especially with more "Portable" devices when you are on the go (Akai MPCs for example) and DAW-less setups.

I do hope this is the appropriate place to make such a suggestion, like I said, I absolutely love the project and I hope this is a welcome enhancement Idea :)

Thanks!

keirf, 2020-10-14:
Yes this could certainly be done. Actually buttons and rotary are somewhat hard to distinguish in the code right now. Probably I would use rotary click to move to next. Hold rotary and left/right to move character.

ArtyGZ, 2020-10-14:
That's actually interesting, I did not really look into the way inputs work in the code. But hey, that's a compromise I could absolutely live with. Any solution even if really basic would be amazing. With cloning already working this could make the emulator 10x more useful for many :)

I'm definitely curious as to how all of this works aswell, I used to write C a few years ago. I'll look into the code a bit more later. Thanks again!

keirf, 2020-10-14:
By the way next release Clone is becoming Cut/Copy/Paste. Essentially like in Windows you can Cut or Copy an image, and then Paste it anywhere (same folder, or a different folder). Cut will delete the original after Paste.

Well I might only get Copy/Paste done for next release. Cut will get done though.

ArtyGZ, 2020-10-14:
That's awesome! Thanks for the info and these cool updates :)
I couldn't live without the cloning function, It's super useful for vintage samplers.

keirf, 2020-10-14:
Do you think Clone should stay alongside Copy/Paste? It's not much more complex to do Clone. Select Copy, then move and select Paste as many times as you want...

ArtyGZ, 2020-10-14:
Indeed, If we add copy/paste/cut, the clone function should go as it's just redundant. That way we could paste the copied file elsewhere if needed I guess.

Having clone along with copy/paste might be too much, that's my thought :)

sckramer, 2022-10-21:
Can the LED version do clone? Is it stashed somewhere, or could it be added somehow while it's in eject mode?